ZerBea/hcxdumptool #305: Code optimization

kimocoder, 2023-04-18:
hcxdumptool.c: note: in included file (through /usr/include/netinet/in.h, /usr/include/arpa/inet.h):
/usr/include/sys/socket.h:80:69: warning: attribute __transparent_union__
/usr/include/sys/socket.h:84:75: warning: attribute __transparent_union__
hcxdumptool.c: note: in included file:
/usr/include/net/if.h:196:5: warning: unknown attribute '__access__'
hcxdumptool.c: note: in included file (through /usr/include/bits/sigstksz.h, /usr/include/signal.h):
/usr/include/unistd.h:372:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:379:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:391:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:400:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:424:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:429:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:547:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:645:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:712:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:805:6: warning: unknown attribute '__access__'
/usr/include/unistd.h:841:6: warning: unknown attribute '__access__'
/usr/include/unistd.h:854:6: warning: unknown attribute '__access__'
/usr/include/unistd.h:890:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:912:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:920:36: warning: unknown attribute '__access__'
/usr/include/unistd.h:932:6: warning: unknown attribute '__access__'
/usr/include/unistd.h:934:36: warning: unknown attribute '__access__'
/usr/include/unistd.h:1170:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:1171:5: warning: unknown attribute '__access__'
/usr/include/unistd.h:1199:5: warning: unknown attribute '__access__'
hcxdumptool.c: note: in included file:
/usr/include/stdio.h:593:12: warning: unknown attribute '__access__'
/usr/include/stdio.h:617:5: warning: unknown attribute '__access__'
/usr/include/stdio.h:838:3: warning: unknown attribute '__access__'
/usr/include/stdio.h:845:3: warning: unknown attribute '__access__'
hcxdumptool.c: note: in included file:
/usr/include/stdlib.h:956:5: warning: unknown attribute '__access__'
/usr/include/stdlib.h:961:3: warning: unknown attribute '__access__'
/usr/include/stdlib.h:962:3: warning: unknown attribute '__access__'
/usr/include/stdlib.h:1016:30: warning: unknown attribute '__access__'
hcxdumptool.c: note: in included file:
/usr/include/string.h:56:32: warning: unknown attribute '__access__'
/usr/include/string.h:135:7: warning: unknown attribute '__access__'
/usr/include/string.h:168:29: warning: unknown attribute '__access__'
/usr/include/string.h:181:6: warning: unknown attribute '__access__'
/usr/include/string.h:392:5: warning: unknown attribute '__access__'
/usr/include/string.h:393:5: warning: unknown attribute '__access__'
/usr/include/string.h:445:37: warning: unknown attribute '__access__'
/usr/include/string.h:467:5: warning: unknown attribute '__access__'
/usr/include/string.h:514:5: warning: unknown attribute '__access__'
hcxdumptool.c: note: in included file:
/usr/include/sys/epoll.h:126:9: warning: unknown attribute '__access__'
/usr/include/sys/epoll.h:137:9: warning: unknown attribute '__access__'
/usr/include/sys/epoll.h:147:9: warning: unknown attribute '__access__'
hcxdumptool.c:4027:28: warning: constant 0x7ffffffffff is so big it is long
hcxdumptool.c:864:44: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:3761:20: warning: incorrect type in assignment (different base types)
hcxdumptool.c:3761:20:    expected restricted __be16 static [addressable] [assigned] [usertype] sll_protocol
hcxdumptool.c:3761:20:    got unsigned short
hcxdumptool.c:3809:20: warning: incorrect type in assignment (different base types)
hcxdumptool.c:3809:20:    expected restricted __be16 static [addressable] [assigned] [usertype] sll_protocol
hcxdumptool.c:3809:20:    got unsigned short
hcxdumptool.c:4365:20: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4314:55: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4467:47: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4467:60: error: undefined identifier 'VERSION_YEAR'

kimocoder, 2023-04-18:
@ZerBea We wanna sort some of this too.

ZerBea, 2023-04-18:
Fixed:
hcxdumptool.c:4027:28: warning: constant 0x7ffffffffff is so big it is long

This values are taken from Make file directly
```
hcxdumptool.c:864:44: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4365:20: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4314:55: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4467:47: error: undefined identifier 'VERSION_TAG'
hcxdumptool.c:4467:60: error: undefined identifier 'VERSION_YEAR'
```
If production flag is 0, this values are taken from git
If production flag is 1, this values are taken from Makefile:
```
PRODUCTION		:= 0
PRODUCTION_VERSION	:= 6.2.9
PRODUCTION_YEAR		:= 2023
```

I can't reproduce the remaining warnings (e.g.:  expected restricted __be16 static [addressable] [assigned] [usertype] sll_protocol
```
The sockaddr_ll structure is a device-independent physical-layer  address.
           struct sockaddr_ll {
               unsigned short sll_family;   /* Always AF_PACKET */
               unsigned short sll_protocol; /* Physical-layer protocol */
               int            sll_ifindex;  /* Interface number */
               unsigned short sll_hatype;   /* ARP hardware type */
               unsigned char  sll_pkttype;  /* Packet type */
               unsigned char  sll_halen;    /* Length of address */
               unsigned char  sll_addr[8];  /* Physical-layer address */
           };
```
On Linux it is It is definitely not __be16 static:

Also I can't  reproduce this warning:  warning: unknown attribute 'access'?  Attribute 'access' isn't used in hcxdumptool.

Operating system?
Compiler?

Looks like your OS is missing some includes.

kimocoder, 2023-04-18:
This is Kali Linux on x86_64 latest kernel 👍

ZerBea, 2023-04-18:
Looks like the includes are completely broken on KALI (again).

ZerBea, 2023-04-18:
I can't reproduce it, neither on Arch, nor on Debian nor on Raspbian.

kimocoder, 2023-04-18:
Few fixed.

How to reproduce or look for sparse? Debian; apt install sparse

then;

sparse -Wsparse-all hcxdumptool.c (or C in general)

kimocoder, 2023-04-18:
Sorry, Im running a custom kernel. But look above for sparse warnings. :+1:

closed 2023-04-18 by kimocoder

ZerBea, 2023-04-18:
Looks like sparse generate warnings of includes, too.

BTW:
$ sparse -Wsparse-all hcxdumptool.c (or C in general)
doesn't work on Arch:
$ sparse -Wsparse-all hcxdumptool.c
hcxdumptool.c: note: in included file (through /usr/include/sys/socket.h, /usr/include/netinet/in.h, /usr/include/arpa/inet.h):
/usr/include/bits/types/struct_iovec.h:23:11: error: unable to open 'stddef.h'

ZerBea, 2023-04-18:
I have no idea to fix e.g. this:
`/usr/include/sys/epoll.h:126:9: warning: unknown attribute 'access'`

/usr/include/sys/epoll.h:126:
```
extern int epoll_wait (int __epfd, struct epoll_event *__events,
		       int __maxevents, int __timeout)
	__attr_access ((__write_only__, 2, 3));
```

kimocoder, 2023-04-18:
Dont worry, as long as we smash some, we good.

ZerBea, 2023-04-18:
But I really want to know why sparse doesn't recognize "__attr_access"
added:
CFLAGS		+= -Wattributes

and still everything is fine:
$ make
cc -O3 -Wall -Wextra -std=gnu99 -Wattributes   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.2.9-110-gc855777\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT

ZerBea, 2023-04-18:
BTW:
On Raspbian and Debian Arm I have not installed git.

This is a modified make file:
```
VERSION_TAG		:= 6.2.9-93-g4e0a783
VERSION_YEAR		:= 2023

PREFIX		?= /usr
BINDIR		= $(DESTDIR)$(PREFIX)/bin

HOSTOS		:= $(shell uname -s)

CC		?= gcc
CFLAGS		?= -O3 -Wall -Wextra
CFLAGS		+= -std=gnu99
#CFLAGS		+= -ggdb -fsanitize=address
DEFS		= -DVERSION_TAG=\"$(VERSION_TAG)\" -DVERSION_YEAR=\"$(VERSION_YEAR)\"
#DEFS		+= -DSTATUSOUT -DNMEAOUT

INSTALL		?= install
INSTFLAGS	=

ifeq ($(HOSTOS), Linux)
INSTFLAGS += -D
endif

TOOLS=hcxdumptool

.PHONY: all build install clean uninstall

all: build

build: $(TOOLS)

# $1: tool name
define tool-build
$(1)_src ?= $(1).c
$(1)_libs ?=
$(1)_cflags ?=

$(1): $$($(1)_src)
	$$(CC) $$(CFLAGS) $$($(1)_cflags) $$(CPPFLAGS) -o $$@ $$($(1)_src) $$(DEFS)

.deps/$(1).d: $(1)

.PHONY: $(1).install
$(1).install: $(1)
	$$(INSTALL) $$(INSTFLAGS) -m 0755 $(1) $$(BINDIR)/$(1)

.PHONY: $(1).clean
$(1).clean:
	rm -f .deps/$(1).d
	rm -f $(1)

.PHONY: $(1).uninstall
$(1).uninstall:
	rm -rf $$(BINDIR)/$(1)

endef

$(foreach tool,$(TOOLS),$(eval $(call tool-build,$(tool))))

install: $(patsubst %,%.install,$(TOOLS))

clean: $(patsubst %,%.clean,$(TOOLS))
	rm -rf .deps
	rm -f *.o *~

uninstall: $(patsubst %,%.uninstall,$(TOOLS))

-include .deps/*.d

```

ZerBea, 2023-04-18:
Funny thing that sparse:

```
#include <stddef.h>
#include <unistd.h>
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>

int main()
{

return EXIT_SUCCESS;
}
```

$ gcc sparsetest.c -o sparsetest
$

$ sparse -Wsparse-all sparsetest.c
sparsetest.c:1:11: error: unable to open 'stddef.h'

```
#include <unistd.h>
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>

int main()
{

return EXIT_SUCCESS;
}
```

$ gcc sparsetest.c -o sparsetest
$

$ sparse -Wsparse-all sparsetest.c
sparsetest.c: note: in included file:
/usr/include/unistd.h:226:11: error: unable to open 'stddef.h'

ZerBea, 2023-04-18:
And it looks like this problem exists since 2014:
https://bugzilla.redhat.com/show_bug.cgi?id=1114755

"I'm afraid that this is just a fundamental limitation in the way sparse currently works. It defaults to looking for gcc headers in the location given by -DGCC_BASE. Once gcc is upgraded, then that no longer works by default since the headers will have moved. The reason "upstream sparse" works for you is because you built it with gcc that's currently installed. If you changed your gcc version after building it, you'd hit the same problem."