ZerBea/hcxdumptool #306: I get error

strunz983, 2023-04-20:
Hello i get error message 
My OS is
Linux kali 6.1.0-kali7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.20-2kali1 (2023-04-18) x86_64 GNU/Linux

The error is when i type make and make install
make
cc -O3 -Wall -Wextra -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.2.9-116-gb30db70\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
hcxdumptool.c: In function ‘writeshb’:
hcxdumptool.c:857:21: error: ‘SHB_SYSINFO_LEN’ undeclared (first use in this function)
  857 | static char sysinfo[SHB_SYSINFO_LEN];
      |                     ^~~~~~~~~~~~~~~
hcxdumptool.c:857:21: note: each undeclared identifier is reported only once for each function it appears in
hcxdumptool.c:857:13: warning: unused variable ‘sysinfo’ [-Wunused-variable]
  857 | static char sysinfo[SHB_SYSINFO_LEN];
      |             ^~~~~~~
make: *** [Makefile:66: hcxdumptool] Error 1

make install                                                                                                                                                      2 ⨯
cc -O3 -Wall -Wextra -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.2.9-116-gb30db70\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
hcxdumptool.c: In function ‘writeshb’:
hcxdumptool.c:857:21: error: ‘SHB_SYSINFO_LEN’ undeclared (first use in this function)
  857 | static char sysinfo[SHB_SYSINFO_LEN];
      |                     ^~~~~~~~~~~~~~~
hcxdumptool.c:857:21: note: each undeclared identifier is reported only once for each function it appears in
hcxdumptool.c:857:13: warning: unused variable ‘sysinfo’ [-Wunused-variable]
  857 | static char sysinfo[SHB_SYSINFO_LEN];
      |             ^~~~~~~
make: *** [Makefile:66: hcxdumptool] Error 1

Thank you :)

ZerBea, 2023-04-20:
include/pcapng.h was not updated
Should be fixed, now.
Please try latest commit.

closed 2023-04-20 by ZerBea

strunz983, 2023-04-20:
OK Thank you :))))

ZerBea, 2023-04-20:
You're welcome.
Thanks for reporting this bug.

The code changes from the old WIRELESS EXTENSION dinosaur to the modern NL80211 version are massive.
Nearly everything is refactored and hcxdumptool is trimmed for effectiveness and speed, now.
Everything that made hcxdumptool slow has been removed. Most of the dependencies have been removed, too and the remaining ones are part of every Linux distribution.