ZerBea/hcxdumptool #308: Make fail

kocsaga, 2023-04-22:
Hello ZerBea! I tried to build the latest build on my raspberry pi zero(latest kernel 5.15.56+). I have build last year without a problem. Now I get this after try to make:
`cc -O3 -Wall -Wextra -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.2                                                                                      .9-122-gdb85452\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
hcxdumptool.c: In function ‘writegpwpl’:
hcxdumptool.c:667:4: warning: implicit declaration of function ‘write’; did you me                                                                                      an ‘fwrite’? [-Wimplicit-function-declaration]
 if(write(fd_hcxpos, gprmc, gprmclen) != gprmclen) errorcount++;
    ^~~~~
    fwrite
hcxdumptool.c: In function ‘writeshb’:
hcxdumptool.c:857:23: error: storage size of ‘unameData’ isn’t known
 static struct utsname unameData;
                       ^~~~~~~~~
hcxdumptool.c:869:4: warning: implicit declaration of function ‘uname’; did you me                                                                                      an ‘rename’? [-Wimplicit-function-declaration]
 if(uname(&unameData) == 0)
    ^~~~~
    rename
hcxdumptool.c:857:23: warning: unused variable ‘unameData’ [-Wunused-variable]
 static struct utsname unameData;
                       ^~~~~~~~~
hcxdumptool.c: In function ‘open_pcapng’:
hcxdumptool.c:960:8: warning: implicit declaration of function ‘stat’; did you mea                                                                                      n ‘strcat’? [-Wimplicit-function-declaration]
  while(stat(pcapngname, &statinfo) == 0)
        ^~~~
        strcat
hcxdumptool.c:968:1: warning: implicit declaration of function ‘umask’ [-Wimplicit                                                                                      -function-declaration]
 umask(0);
 ^~~~~
hcxdumptool.c: In function ‘process_nmea0183’:
hcxdumptool.c:2407:15: warning: implicit declaration of function ‘read’; did you m                                                                                      ean ‘fread’? [-Wimplicit-function-declaration]
 if((nmealen = read(fd_gps, nmeabuffer, NMEA_SIZE)) < NMEA_MIN)
               ^~~~
               fread
hcxdumptool.c: In function ‘nl_get_interfacecapabilities’:
hcxdumptool.c:3097:17: warning: implicit declaration of function ‘readlink’; did y                                                                                      ou mean ‘read_bpf’? [-Wimplicit-function-declaration]
     if((dnlen = readlink(driverfmt, driverlink, 64)) > 0)
                 ^~~~~~~~
                 read_bpf
hcxdumptool.c: In function ‘open_device_nmea0183’:
hcxdumptool.c:3883:23: error: storage size of ‘tty’ isn’t known
 static struct termios tty;
                       ^~~
hcxdumptool.c:3887:4: warning: implicit declaration of function ‘tcgetattr’ [-Wimp                                                                                      licit-function-declaration]
 if(tcgetattr(fd_gps, &tty) < 0) return false;
    ^~~~~~~~~
hcxdumptool.c:3888:17: error: ‘PARENB’ undeclared (first use in this function)
 tty.c_cflag &= ~PARENB; // Clear parity bit, disabling parity (most common)
                 ^~~~~~
hcxdumptool.c:3888:17: note: each undeclared identifier is reported only once for                                                                                       each function it appears in
hcxdumptool.c:3889:17: error: ‘CSTOPB’ undeclared (first use in this function); di                                                                                      d you mean ‘CSTOP’?
 tty.c_cflag &= ~CSTOPB; // Clear stop field, only one stop bit used in communicat                                                                                      ion (most common)
                 ^~~~~~
                 CSTOP
hcxdumptool.c:3890:17: error: ‘CSIZE’ undeclared (first use in this function); did                                                                                       you mean ‘CB_SIZE’?
 tty.c_cflag &= ~CSIZE; // Clear all bits that set the data size
                 ^~~~~
                 CB_SIZE
hcxdumptool.c:3891:16: error: ‘CS8’ undeclared (first use in this function)
 tty.c_cflag |= CS8; // 8 bits per byte (most common)
                ^~~
hcxdumptool.c:3892:17: error: ‘CRTSCTS’ undeclared (first use in this function)
 tty.c_cflag &= ~CRTSCTS; // Disable RTS/CTS hardware flow control (most common)
                 ^~~~~~~
hcxdumptool.c:3893:16: error: ‘CREAD’ undeclared (first use in this function)
 tty.c_cflag |= CREAD | CLOCAL; // Turn on READ & ignore ctrl lines (CLOCAL = 1)
                ^~~~~
hcxdumptool.c:3893:24: error: ‘CLOCAL’ undeclared (first use in this function); di                                                                                      d you mean ‘AF_LOCAL’?
 tty.c_cflag |= CREAD | CLOCAL; // Turn on READ & ignore ctrl lines (CLOCAL = 1)
                        ^~~~~~
                        AF_LOCAL
hcxdumptool.c:3894:17: error: ‘ICANON’ undeclared (first use in this function)
 tty.c_lflag &= ~ICANON;
[... 78 lines cut ...]
hcxdumptool.c:3907:23: error: ‘TCSANOW’ undeclared (first use in this function); d                                                                                      id you mean ‘TCSETSW’?
 if (tcsetattr(fd_gps, TCSANOW, &tty) < 0) return false;
                       ^~~~~~~
                       TCSETSW
hcxdumptool.c:3883:23: warning: unused variable ‘tty’ [-Wunused-variable]
 static struct termios tty;
                       ^~~
hcxdumptool.c: In function ‘close_sockets’:
hcxdumptool.c:3916:25: warning: implicit declaration of function ‘close’; did you                                                                                       mean ‘pclose’? [-Wimplicit-function-declaration]
 if(fd_socket_unix != 0) close(fd_socket_unix);
                         ^~~~~
                         pclose
hcxdumptool.c: In function ‘open_socket_rt’:
hcxdumptool.c:3946:16: warning: implicit declaration of function ‘getpid’; did you                                                                                       mean ‘getpt’? [-Wimplicit-function-declaration]
 saddr.nl_pid = getpid();
                ^~~~~~
                getpt
hcxdumptool.c: In function ‘main’:
hcxdumptool.c:4884:4: warning: implicit declaration of function ‘getuid’; did you                                                                                       mean ‘getpt’? [-Wimplicit-function-declaration]
 if(getuid() != 0)
    ^~~~~~
    getpt
make: *** [Makefile:66: hcxdumptool] Error 1
`
What I am missing?

ZerBea, 2023-04-23:
Look like the development files (includes / headers) are not installed. Even the header files for such basic functions like write and read are missing on your system. You need to install them.
Than everything compiles fine:
```
$ git clone https://github.com/ZerBea/hcxdumptool
Cloning into 'hcxdumptool'...
remote: Enumerating objects: 4802, done.
remote: Counting objects: 100% (337/337), done.
remote: Compressing objects: 100% (139/139), done.
remote: Total 4802 (delta 204), reused 298 (delta 198), pack-reused 4465
Receiving objects: 100% (4802/4802), 1.52 MiB | 1.97 MiB/s, done.
Resolving deltas: 100% (3232/3232), done.

$ cd hcxdumptool

$ make
cc -O3 -Wall -Wextra -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.2.9-122-gdb85452\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT

$ sudo make install
install -D -m 0755 hcxdumptool /usr/bin/hcxdumptool
```

kocsaga, 2023-04-23:
Thanks for the answer! Updated the build-essential to the latest, but the error is the same. Tried to reinstall it is still same. Reverted back to an older SD image, still same problem.
Tried to build the last one, that I used (version 6.2.5) and it does the make without problem.
It is very strange. Do you have any ideas?

ZerBea, 2023-04-23:
Indeed, very strange. Looks like a header file is missing on your system.
But the standard includes have not changed on all versions of hcxdumptool.

Please comment output of
$ uname -r

$ gcc --version

ZerBea, 2023-04-23:
I changed some includes.
Please try again on latest git head.

kocsaga, 2023-04-23:
Nice Job! Make works again.
Thank you!

closed 2023-04-23 by kocsaga

ZerBea, 2023-04-23:
No problem, you're welcome.

But unfortunately I don't know why it is working, now.

Can you please comment output of
$ uname -r
and
$ gcc --version

Maybe we can figure out why "unistd.h" wasn't included by other headers.

kocsaga, 2023-04-23:
Updated the last comment just now :)

ZerBea, 2023-04-23:
Ok, Thanks.
Now I know what happened.
I tested the version only on gcc 11 and 12. It looks like 8.3 have a different behavior regarding some standard includes.

ZerBea, 2023-04-24:
You may have noticed that I changed the headline.
According to GCC release overview
https://gcc.gnu.org/releases.html

GCC 8 is deprecated and not longer maintained:
https://gcc.gnu.org/gcc-8/

GCC 8.3 reached EOL on Feb 22, 2019.