ZerBea/hcxdumptool #313: kernel v6.3 issue?

kimocoder, 2023-05-03:
I moved my RPI4B to kernel 6.3 yesterday and I got this, and updated Kali on laptop to 6.3 today and the same occours.

[image]

ZerBea, 2023-05-03:
Looks like the driver does not transmit frames coming from hcxdumptool.

Which driver?
$ hcxdumptool -L
$ hcxdumptool -I interfacename

Arch is still on 6.2.13-arch1-1, but 6.3 is already on testing:
https://archlinux.org/packages/testing/x86_64/linux/
I expect that kernel 6.3 will be distributed, soon. Than I can do further going tests.

kimocoder, 2023-05-03:
This is the mt76x0u. MT7601u. Looks like a change between kernel 6.1 and 6.3.

  CHA    LAST   R M1C M3A PID A    MAC-AP    ESSID (last seen on top)        SCAN-FREQUENCY:   2437
----------------------------------------------------------------------------------------------------
 [  6] 08:09:31               + a0e4cb77f1cd Telenor1401rar
 [  6] 08:09:31               + 743aefcfba4f Telenor1802sel
 [  6] 08:09:31               + 743aefcfb76f Telenor8210lun
 [  6] 08:09:31               + 342109223f40 AirLink223f40
 [  6] 08:09:31               + ac220b31c4d8 Inges ruter
 [  6] 08:09:31               + 44f034b2d75f IkkeDinWifi
 [  6] 08:09:31               + 3421098352e9 Jensen
 [  6] 08:09:31               + 3421092f18dc dlink-824C_2.4Gre
 [  6] 08:09:31               + 7e45583d2e67 
 [  6] 08:09:31               + 18a6f7225b9c raptus
 [  6] 08:09:31               + 743aefcfa55f Telenor9229nam
 [  6] 08:09:31               + 4494fc984456 NETGEAR01
 [  6] 08:09:31               + 3421098352e1 Jensen
 [  6] 08:09:31               + 743aefcf895f Telenor7822fyr
 [  6] 08:09:31               + 743aefcfb74f Telenor9963sjy
 [  6] 08:09:31                 fa8fca7075c5 NestMini9073.ynm,
 [  6] 08:09:30               + 342109763e19 Hogwarts WizFi
 [  6] 08:09:30               + 743aefceb53f RYE
 [  6] 08:09:30               + c0a0bbfe824c dlink-824C
 [  6] 08:09:30               + 743aefcf2d8f Telenor2988ost

   LAST   M2R MAC-AP-ROGUE  MAC-CLIENT  ESSID (last seen on top)
----------------------------------------------------------------------------------------------------

101 errors during runtime

exit on error

bye-bye

ZerBea, 2023-05-03:
That sounds not good. R column is empty. That means packet injection is not working.

kimocoder, 2023-05-03:
But it works in all other tools, did tests on mon/inject.

However, on the Nothing Phone 1 on 5.4 🔢 

[image]

ZerBea, 2023-05-03:
6.2.13-arch1-1 is still ok, too. Frame injection and active monitor mode is working as expected: 
```
ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
ID 7392:7710 Edimax Technology Co., Ltd Edimax Wi-Fi
```
```
$ hcxdumptool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1   4 0c9d92b476c1 b025aaa0b8fc * wlp39s0f3u1u1u4  mt76x0u (NETLINK & WIRELESS EXTENSIONS)
  2   5 74da38eb360f b025aa1b3e8d * wlp39s0f3u1u1u1  mt7601u (NETLINK & WIRELESS EXTENSIONS)

* active monitor mode available
+ monitor mode available
- no monitor mode available
```

I'll add some debug code if kernel 6.3 is distributed on Arch package management system.

ZerBea, 2023-05-03:
There are huge code changes from kernel 6.2 to 6.3 on MediaTek drivers.

kimocoder, 2023-05-03:
Yes I know, thats why I rolled it. and rtw88 and rtl8xxxu too :+1:

kimocoder, 2023-05-03:
Better alignement:)

[image]

kimocoder, 2023-05-03:
[13730.254922] brcmfmac mmc1:0001:1 wlan0: entered promiscuous mode
[13744.281236] brcmfmac mmc1:0001:1 wlan0: left promiscuous mode
[16405.097326] usbcore: deregistering interface driver brcmfmac
[16411.075455] brcmfmac: F1 signature read @0x18000000=0x15264345
[16411.082289] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[16411.083913] usbcore: registered new interface driver brcmfmac
[16411.400265] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[16411.403756] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May  3 2023 00:35:10 version 7.45.206 (nexmon.org: 2.2.2-407-g3a67-8) FWID 01-88ee44ea
[16411.531198] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

Nexmon powered brcmfmac :+1:

ZerBea, 2023-05-03:
This
`5 12 dca63209cc20 dca63209cc20 + wlan0 brcmfmac (NETLINK & WIRELESS EXTENSIONS)`
is new for me, because the default driver doesn't work:
https://github.com/ZerBea/hcxdumptool/issues/309

ZerBea, 2023-05-03:
During init, hcxdumptool:
set interface down via RTNETLINK
set MONITOR MODE via NL80211
set interface up via RTNETLINK
set CHANNEL and BANDWIDTH via NL80211

I'll add some debug code.....

ZerBea, 2023-05-03:
BTW:
This is a show stopper that prevent release of hcxdumptool 6.3.0 which was planed together with release of Linux kernel 6.3.

ZerBea, 2023-05-04:
Right now, update is running:
gcc 12.2.1-2 -> 13.1.1-1
linux 6.2.13.arch1-1 -> 6.3.1.arch1-1

ZerBea, 2023-05-04:
First test passed:
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/13.1.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,objc,obj-c++ --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.1.1 20230429 (GCC) 
```

```
$ make
cc -O3 -Wall -Wextra -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.2.9-162-gcd266fb\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
```

```
$ sudo make install
install -D -m 0755 hcxdumptool /usr/bin/hcxdumptool
```

```
$ hcxdumptool -v
hcxdumptool 6.2.9-162-gcd266fb (C) 2023 ZeroBeat
```

hcxdumptool (and hcxtools, too) is ready for gcc 13.x.x

ZerBea, 2023-05-04:
mt76 test passed:

```
$ uname -r
6.3.1-arch1-1
```

```
$ lsub
ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
```

```
$ hcxdumptool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1   4 00e0614861dd 00e0614861dd * wlp39s0f3u1u1u2  mt7601u (NETLINK & WIRELESS EXTENSIONS)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```

```
$ hcxdumptool -I wlp39s0f3u1u1u2

Requesting interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1   4 00e0614861dd 00e0614861dd * wlp39s0f3u1u1u2  mt7601u (NETLINK & WIRELESS EXTENSIONS)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] disabled

bye-bye
```

```
$ hcxpcapngtool -o test.22000 *.pcapng
hcxpcapngtool 6.2.9-73-g89f7d50 reading from 20230504073648-wlp39s0f3u1u1u2.pcapng...

summary capture file
--------------------
file name................................: 20230504073648-wlp39s0f3u1u1u2.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.3.1-arch1-1
application..............................: hcxdumptool 6.2.9-162-gcd266fb
interface name...........................: wlp39s0f3u1u1u2
interface vendor.........................: 00e061
openSSL version..........................: 1.0
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 980ee479adc7 (incremented on every new client)
MAC CLIENT...............................: e0cb1da32ad6
REPLAYCOUNT..............................: 65532
[... 72 lines cut ...]

0e9c49c1ffcb0526cc119a88b5b84c52:0896d798e19e:7e8ab7b1d211:AP_7272:12345678
                                                          
Session..........: hashcat
Status...........: Cracked
Hash.Mode........: 22000 (WPA-PBKDF2-PMKID+EAPOL)
Hash.Target......: test.22000
Time.Started.....: Thu May  4 07:40:39 2023 (0 secs)
Time.Estimated...: Thu May  4 07:40:39 2023 (0 secs)
Kernel.Feature...: Pure Kernel
Guess.Mask.......: 12345678 [8]
Guess.Queue......: 1/1 (100.00%)
Speed.#1.........:       28 H/s (1.43ms) @ Accel:8 Loops:256 Thr:512 Vec:1
Recovered........: 1/1 (100.00%) Digests (total), 1/1 (100.00%) Digests (new)
Progress.........: 1/1 (100.00%)
Rejected.........: 0/1 (0.00%)
Restore.Point....: 0/1 (0.00%)
Restore.Sub.#1...: Salt:0 Amplifier:0-1 Iteration:0-1
Candidate.Engine.: Device Generator
Candidates.#1....: 12345678 -> 12345678
Hardware.Mon.#1..: Temp: 57c Fan: 32% Util: 96% Core:1860MHz Mem:5005MHz Bus:16

Started: Thu May  4 07:40:37 2023
Stopped: Thu May  4 07:40:40 2023
```

ZerBea, 2023-05-04:
[image]

ZerBea, 2023-05-04:
Neither ERRORs nor warnings. Everything is working as expected.

Test targets (AP and and CLIENT) attacked successfully. 

BTW:
That include hcxtools (compiled by gcc 13.1.1) and hashcat (compiled by gcc 13.1.1), too.

ZerBea, 2023-05-04:
By latest commit I fixed CLIENT display (E and 2) output are swapped, now).

ZerBea, 2023-05-04:
BTW:
Last test showed another lesson learned for me:
Test target AP, test target CLIENT and attack device are too close together (< 1m). That caused that the interfaces interfered with each other!!! My spectrum analyzer showed really ugly signals to me.

ZerBea, 2023-05-04:
Mainly hcxdumptool is designed to run on a headless Raspberry Pi (status out can be disabled by Makefile, completely).
For this reason, internal error detection is merciless.
In order to be able to react to this ERRORs (mandatory on headless operation), there are several command line options.

j-v-bernstein, 2023-05-04:
> Better alignement:)
> 
> [image]

I hope this is not completely OT, but I wanted @kimocoder ask what kind keyboard app resp font he uses!?

ZerBea, 2023-05-04:
Does the terminal understand system("clear")?

kimocoder, 2023-05-04:
Its a full Debian chroot running on top of Android, so everything works. Past 5 years we even got control of more hardware through chroot 👍

ZerBea, 2023-05-04:
I still can't reproduce the ERROR. Running kernel 6.3.1 all tests passed successfully.

kimocoder, 2023-05-04:
I certain it is related to indents prior to terminal sizing, because end result is now working for me too.

closed 2023-05-04 by kimocoder

ZerBea, 2023-05-04:
Thanks. Glad to hear that the new size is working for you, now.

j-v-bernstein, 2023-05-08:
> Standard NetHunter Terminal settings, but we switch to a new Terninal in the next release cycle using same font as Kali.

I'm well-known with the terminal.  ;-)

My interest consist more of the keyboard apk you use resp. the theme..?!