ZerBea/hcxdumptool #315: problem with option - sudo hcxdumptool --do_rcascan-rc -i wlan0

visionary22, 2023-05-03:
(kali㉿kali)-[~]
$ sudo hcxdumptool --do_rcascan-rc -i wlan0
**hcxdumptool: unrecognized option '--do_rcascan-rc'**
hcxdumptool 6.2.9-152-g50ecce3 (C) 2023 by ZeroBeat
usage: hcxdumptool -h for help

_it always says unrecognized option and I don't know how to proceed. I'm a newbie, so I apologize that I may be asking an easy question for you, but I'm stuck here. I also tried the "--do_rcascan" command but still the same problem.

Thank you very much for every comment, I will be very much appreciated.

Have a nice day and greetings from the Czech Republic :D_

ZerBea, 2023-05-03:
Starting on 07.04.2023, hcxdumptool is completely refactored.
Old code has been removed or replaced by new code. That include most of the options, too. 
A good starting point is to read changelog, README.md and hcxdumptool --help.

Th function rca_scan was completely replaced by:
```
--rcascan=<character>          : do (R)adio (C)hannel (A)ssignment scan
                                  default = passive scan
                                  a = active scan
                                  p = passive scan
```
Now you can decide to run an active scan or a passive scan.

visionary22, 2023-05-03:
Oh, thank you so much for your helpfulness and explanation :) have a great day! I have about 4 more questions and stuckness, but I'll have to go through that myself, I think I'll learn the most from it. :) thank you so much again! Have a great day!

ZerBea, 2023-05-03:
You're welcome.

The simplest way to use hcxdumptool:

stop all services that take access to the interface:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service

start hcxdumptool:
$ hcxdumptool

Take a look at what hcxdumptool does.
Than take a look at the options to find out which options make sense for you.
Now run hcxdumptool with your options and see what changed.

I'll close this report, because it is not a bug,  but feel free to ask your questions.

closed 2023-05-03 by ZerBea