ZerBea/hcxdumptool #320: Errors when make on Debian Bullseye

mike89klein, 2023-05-11:
Hi. Install all dependies, but when run command "make" get errors: 
cc -O3 -Wall -Wextra -std=gnu99    -MMD -MF .deps/hcxpcapngtool.d -o hcxpcapngtool hcxpcapngtool.c -lssl -lcrypto -lz  -DVERSION_TAG=\"6.3.0-22-g9f5885c\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
hcxpcapngtool.c:86:8: error: unknown type name 'EVP_MAC'
 static EVP_MAC *hmac;
        ^~~~~~~
hcxpcapngtool.c:87:8: error: unknown type name 'EVP_MAC'
 static EVP_MAC *cmac;
        ^~~~~~~
hcxpcapngtool.c:88:8: error: unknown type name 'EVP_MAC_CTX'
 static EVP_MAC_CTX *ctxhmac;
        ^~~~~~~~~~~
hcxpcapngtool.c:89:8: error: unknown type name 'EVP_MAC_CTX'
 static EVP_MAC_CTX *ctxcmac;
        ^~~~~~~~~~~
hcxpcapngtool.c:90:8: error: unknown type name 'OSSL_PARAM'
 static OSSL_PARAM paramsmd5[3];
        ^~~~~~~~~~
hcxpcapngtool.c:91:8: error: unknown type name 'OSSL_PARAM'
 static OSSL_PARAM paramssha1[3];
        ^~~~~~~~~~
hcxpcapngtool.c:92:8: error: unknown type name 'OSSL_PARAM'
 static OSSL_PARAM paramssha256[3];
        ^~~~~~~~~~
hcxpcapngtool.c:93:8: error: unknown type name 'OSSL_PARAM'
 static OSSL_PARAM paramsaes128[3];
        ^~~~~~~~~~
hcxpcapngtool.c: In function 'testpmkid':
hcxpcapngtool.c:1956:5: warning: implicit declaration of function 'EVP_MAC_init'; did you mean 'EVP_MD_CTX_init'? [-Wimplicit-function-declaration]
 if(!EVP_MAC_init(ctxhmac, testpmk, 32, paramssha1)) return false;
     ^~~~~~~~~~~~
     EVP_MD_CTX_init
hcxpcapngtool.c:1957:5: warning: implicit declaration of function 'EVP_MAC_update'; did you mean 'HMAC_Update'? [-Wimplicit-function-declaration]
 if(!EVP_MAC_update(ctxhmac, pmkidcalc, 20)) return false;
     ^~~~~~~~~~~~~~
     HMAC_Update
hcxpcapngtool.c:1958:5: warning: implicit declaration of function 'EVP_MAC_final'; did you mean 'HMAC_Final'? [-Wimplicit-function-declaration]
 if(!EVP_MAC_final(ctxhmac, pmkidcalc, NULL, 20)) return false;
     ^~~~~~~~~~~~~
     HMAC_Final
hcxpcapngtool.c: In function 'evpdeinitwpa':
hcxpcapngtool.c:5688:2: warning: implicit declaration of function 'EVP_MAC_CTX_free'; did you mean 'EVP_MD_CTX_free'? [-Wimplicit-function-declaration]
  EVP_MAC_CTX_free(ctxhmac);
  ^~~~~~~~~~~~~~~~
  EVP_MD_CTX_free
hcxpcapngtool.c:5689:2: warning: implicit declaration of function 'EVP_MAC_free'; did you mean 'EVP_PKEY_free'? [-Wimplicit-function-declaration]
  EVP_MAC_free(hmac);
  ^~~~~~~~~~~~
  EVP_PKEY_free
hcxpcapngtool.c: In function 'evpinitwpa':
hcxpcapngtool.c:5720:8: warning: implicit declaration of function 'EVP_MAC_fetch'; did you mean 'EVP_PBE_get'? [-Wimplicit-function-declaration]
 hmac = EVP_MAC_fetch(NULL, "hmac", NULL);
        ^~~~~~~~~~~~~
        EVP_PBE_get
hcxpcapngtool.c:5720:6: warning: assignment to 'int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 hmac = EVP_MAC_fetch(NULL, "hmac", NULL);
      ^
hcxpcapngtool.c:5722:6: warning: assignment to 'int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 cmac = EVP_MAC_fetch(NULL, "cmac", NULL);
      ^
hcxpcapngtool.c:5725:16: warning: implicit declaration of function 'OSSL_PARAM_construct_utf8_string' [-Wimplicit-function-declaration]
 paramsmd5[0] = OSSL_PARAM_construct_utf8_string("digest", "md5", 0);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hcxpcapngtool.c:5726:16: warning: implicit declaration of function 'OSSL_PARAM_construct_end' [-Wimplicit-function-declaration]
 paramsmd5[1] = OSSL_PARAM_construct_end();
                ^~~~~~~~~~~~~~~~~~~~~~~~
hcxpcapngtool.c:5737:11: warning: implicit declaration of function 'EVP_MAC_CTX_new'; did you mean 'EVP_MD_CTX_new'? [-Wimplicit-function-declaration]
 ctxhmac = EVP_MAC_CTX_new(hmac);
           ^~~~~~~~~~~~~~~
           EVP_MD_CTX_new
hcxpcapngtool.c:5737:9: warning: assignment to 'int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 ctxhmac = EVP_MAC_CTX_new(hmac);
         ^
hcxpcapngtool.c:5739:9: warning: assignment to 'int *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 ctxcmac = EVP_MAC_CTX_new(cmac);
         ^
make: *** No rule to make target 'hcxpcapngtool', needed by 'build'.  Stop.

Platform:Radxa Zero
Kernel: 5.10.69-12-amlogic

ZerBea, 2023-05-11:
I closed this issue for 2 reasons:
1) this is hcxdumptool bug tracker, but you report a problem with hcxtools 
2) not solved or missing dependencies - in your case: you are running an old version of OpenSSL. You need version >= 3.0

https://packages.debian.org/sid/openssl

from README.md (hcxtools) section Requirements:
```
Requirements
--------------
* libopenssl (>= 3.0) and openssl-dev installed
```

Please read more here:
https://github.com/ZerBea/hcxtools/issues/275
https://github.com/ZerBea/hcxtools/issues/249
https://github.com/ZerBea/hcxtools/issues/233
https://github.com/ZerBea/hcxtools/issues/229
https://github.com/ZerBea/hcxtools/issues/228

closed 2023-05-11 by ZerBea

ZerBea, 2023-05-11:
You mentioned that you are running bullseye.
This is the official package of hcxtools, available via Debian package management system:
https://packages.debian.org/bullseye/net/hcxtools

If you would like to compile a newer version, it is mandatory to meet the conditions as mentioned in README.md Requirements section.

ZerBea, 2023-05-12:
Just merged a PR to hcxtools and compiled the entire repository (a good test to see that everything is still working fine).

When the dependencies are met as described in README.md:
```
$ pacman -Q | grep openssl
openssl 3.0.8-1

$ pacman -Q | grep gcc
gcc 13.1.1-1
gcc-libs 13.1.1-1
```

everything compiles as expected:
```
$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99    -MMD -MF .deps/hcxpcapngtool.d -o hcxpcapngtool hcxpcapngtool.c -lssl -lcrypto  -lz   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99    -MMD -MF .deps/hcxhashtool.d -o hcxhashtool hcxhashtool.c -lssl -lcrypto  -lcurl   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpsktool.d -o hcxpsktool hcxpsktool.c -lssl -lcrypto   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpmktool.d -o hcxpmktool hcxpmktool.c -lssl -lcrypto   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxeiutool.d -o hcxeiutool hcxeiutool.c   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxwltool.d -o hcxwltool hcxwltool.c   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxhash2cap.d -o hcxhash2cap hcxhash2cap.c   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99    -MMD -MF .deps/wlancap2wpasec.d -o wlancap2wpasec wlancap2wpasec.c -lssl -lcrypto  -lcurl   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
cc -O3 -Wall -Wextra -std=gnu99    -MMD -MF .deps/whoismac.d -o whoismac whoismac.c -lssl -lcrypto  -lcurl   -DVERSION_TAG=\"6.3.0-24-gd01bf32\" -DVERSION_YEAR=\"2023\" -DWANTZLIB
```
and install as expected: 
```
$ sudo make install
install -D -m 0755 hcxpcapngtool /usr/bin/hcxpcapngtool
install -D -m 0755 hcxhashtool /usr/bin/hcxhashtool
install -D -m 0755 hcxpsktool /usr/bin/hcxpsktool
install -D -m 0755 hcxpmktool /usr/bin/hcxpmktool
install -D -m 0755 hcxeiutool /usr/bin/hcxeiutool
install -D -m 0755 hcxwltool /usr/bin/hcxwltool
install -D -m 0755 hcxhash2cap /usr/bin/hcxhash2cap
install -D -m 0755 wlancap2wpasec /usr/bin/wlancap2wpasec
install -D -m 0755 whoismac /usr/bin/whoismac
```