ZerBea/hcxdumptool #321: Option "disable_beacon" does not stop beacon.

Apexwin, 2023-05-13:
Hi, when using option "disable_beacon", i steel see fake access points created by hcxdumptool, please fix.
hcxdumptool 6.3.0-9-g12713a6

ZerBea, 2023-05-13:
The option "disable_beacon" only stops beaconing and hcxdumptool will not transmit beacons.
That is explained in --help:
`-disable_beacon               : do not transmit BEACON frames`
If you run Wireshark in parallel, you will not see a BEACON transmitted by hcxdumtool.

If a CLIENT transmit a PROBEREQUEST, hcxdumptool will respond to the CLIENT.
If a CLIENT want to connect to hcxdumptool, hcxdumptool will allow this.

By latest commit, 
https://github.com/ZerBea/hcxdumptool/commit/ba034ae12fdd93f88548d3e4b7711ad8218271aa
hcxdumptool allow to set "attemptclientmax to 0" (--attemptclientmax=0).

Running:
`sudo hcxdumptool --attemptclientmax=0 --disable_beacon`
hcxdumptool shouldn't respond any longer to a CLIENT.

ZerBea, 2023-05-13:
You may have noticed that I changed the head line. If you set "disable_beacon", hcxdumptool will not send BEACONs.

ZerBea, 2023-05-13:
Please notice:
The last commits will not(!) change the behavior if the CLIENT display. You will still see CLIENTs there.
But you shouldn't see an M2 "+" there.

ZerBea, 2023-05-13:
Please also notice:
hcxdumptool is focused on CLIENT attacks (in contrast to other tools that are focused on AP attacks).
If you disable CLIENT attacks completely, hcxdumptool will lose 90 percent of its capabilities!

Apexwin, 2023-05-15:
Thank you, for help.

closed 2023-05-15 by Apexwin

ZerBea, 2023-05-15:
You're welcome.

Please notice:
On every case and regardless of the options you choose, hcxdumptool receive everything and store useful frames to the pcapng file. If hcxdumptool receive a handshake it will store it. The options only control the transmit behavior of hcxdumptool.

The only way to prevent this is to set a Berkeley Packet Filter. Doing this, you can control hcxdumtools entire behavior regarding content, reception and transmission of frames, That is a thousand times faster than doing it in user space.

If you set attemptclientmax=0 you'll also notice that the bottom screen doesn't show all ESSIDs. The ESSID column is mostly empty, because hcxdumptool doesn't request this information from CLIENTs any longer.
In case of missing frames, hcxdumptool no longer request them. You have to expect a packet loss.

If you set attempapmax=0, hcxdumptool disable:
```
deauthentication     : do not transmit DEAUTHENTICATION/DISASSOCIATION frames
proberequest         : do not transmit PROBEREQUEST frames
association          : do not AUTHENTICATE/ASSOCIATE
reassociation        : do not REASSOCIATE a CLIENT
```