ZerBea/hcxdumptool #322: High radiotap errors on openwrt

rubenkr, 2023-05-15:
Hi ZerBea,

i have an issue I am unable to solve. Following background:

Unifi Loco Nanostation2 with OpenWRT

`root@OpenWrt:/tmp# uname -a
Linux OpenWrt 5.10.176 #0 Thu Apr 27 20:28:15 2023 mips GNU/Linux
`

```
root@OpenWrt:/tmp# hcxdumptool -v
hcxdumptool 6.2.4 (C) 2021 ZeroBeat

```

`root@OpenWrt:/tmp# hcxdumptool -I
wlan interfaces:
6872515019a0 wlan0 (ath9k)
`

When running rca-scan for 30seconds it stops with issue:
`terminating...
100 radiotap errors encountered
`

```
root@OpenWrt:/tmp# hcxdumptool -i wlan0 --check_driver
initialization of hcxdumptool 6.2.4...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

```

`root@OpenWrt:/tmp# hcxdumptool -i wlan0 --check_injection
initialization of hcxdumptool 6.2.4...
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,14
packet injection is working on 2.4GHz!
injection ratio: 18% (BEACON: 580 PROBERESPONSE: 108)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 52% (NETWORK: 44 PROBERESPONSE: 23)
your antenna ratio is good
7 radiotap errors encountered during the test

terminating...
7 radiotap errors encountered
`
I stopped wpa_supplicant, firewall, dnsmasq and dhcpcd server already.

What can I do about it except raising the error threshold which seems like a too simple way.

Thank you for your ideas.

ZerBea, 2023-05-16:
Everything < 10 ERRORS is  completely normal. This is garbage inside the receive buffer when hcxdumptool starts.

But your hcxdumptool (6.2.4) is outdated. I recommend to update to the latest git head. Starting with v6.3.0 I removed WIRELESS EXTENSIONS in favor of NL80211:
https://hashcat.net/forum/thread-11349-post-58207.html#pid58207
Nearly all options changed and a separate driver test / injection test is no longer necessary.

rubenkr, 2023-05-16:
Thanks for the fast reply. I installed it via opkg packet manager. I will dig into how to compile it on openwrt as the device itself has not much freespace.

However when I run hcxdumptool over night I get over 1.000 errors. SO as you say probably the driver issue. If you confirm I will simply close the issue with that information. Thanks and regards

ZerBea, 2023-05-16:
That (> 1.000 ERRORs) is a way too much and it point to a serious driver problem.
Especially the the combination of WEXT and Realtek driver lead to this problems, as well as pspoll().
Latest hcxdumptool doesn't call pspoll() any longer. Instead it use epoll(). It use NL80211 instead of WEXT. All this increase the driver compatibility and reduce problems.

BTW:
Many things have changed since kernel 5.10 and the master is (as of today) on 5.15:
https://openwrt.org/docs/techref/targets/kernelversions

closed 2023-05-16 by rubenkr

ZerBea, 2023-05-16:
You can give hcxdumptool 6.3.0 (NL90211 version a try). It should work on kernel 5.10 (as long NL80211 is in use).
There are no additional dependencies if you compile 6.3.0.

rubenkr, 2023-05-17:
thank you its works much better. Tanks so much for your dedication which give this project truly the most unique character of open source projects <3