ZerBea/hcxdumptool #324: Doesn't compile

alexvillalb159, 2023-05-17:
I tried to compile it on the Trisquel Linux distribution and it gives the following error:

cc -O3 -Wall -Wextra -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.0-24-g5cb94f3\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
hcxdumptool.c: In function ‘channel_to_frequency’:
hcxdumptool.c:658:7: error: ‘NL80211_BAND_S1GHZ’ undeclared (first use in this function); did you mean ‘NL80211_BAND_6GHZ’?
  658 |  case NL80211_BAND_S1GHZ:
      |       ^~~~~~~~~~~~~~~~~~
      |       NL80211_BAND_6GHZ
hcxdumptool.c:658:7: note: each undeclared identifier is reported only once for each function it appears in
hcxdumptool.c: In function ‘rt_get_interfacelist’:
hcxdumptool.c:3569:24: error: ‘IFLA_PERM_ADDRESS’ undeclared (first use in this function); did you mean ‘IFLA_ADDRESS’?
 3569 |    if(rta->rta_type == IFLA_PERM_ADDRESS)
      |                        ^~~~~~~~~~~~~~~~~
      |                        IFLA_ADDRESS
hcxdumptool.c: In function ‘set_interface’:
hcxdumptool.c:3754:68: error: ‘NL80211_BAND_S1GHZ’ undeclared (first use in this function); did you mean ‘NL80211_BAND_6GHZ’?
 3754 |    else if(userband[0] == 'e') ufreq = channel_to_frequency(uband, NL80211_BAND_S1GHZ);
      |                                                                    ^~~~~~~~~~~~~~~~~~
      |                                                                    NL80211_BAND_6GHZ
make: *** [Makefile:66: hcxdumptool] Error 1

ZerBea, 2023-05-18:
Please comment output of 
```
$ uname -r
$ gcc --version
```

Does Trisquel Linux meet the requirements as mentioned in README.md?
```
* operating system: Linux Kernel >= 6.1
* gcc >= 12 recommended (deprecated versions are not supported: https://gcc.gnu.org/)
```

Linux kernel >= 6.1 is recommended (due to several driver updates), but  Linux kernels >= 5.15 (Trisquel Aramo) should work too.

If the requirements meet, "IFLA_PERM_ADDRESS" is defined here:
https://github.com/torvalds/linux/blob/master/include/uapi/linux/if_link.h
and  "NL80211_BAND_S1GHZ" is defined here:
https://github.com/torvalds/linux/blob/master/include/uapi/linux/nl80211.h#L5288

ZerBea, 2023-05-18:
Just downloaded the linux-libre kernel 6.1.29 from here:
https://linux-libre.fsfla.org/pub/linux-libre/releases/6.1.29-gnu/
All definitions are present - hcxdumptool should compile:
```
$ grep -r NL80211_BAND_S1GHZ
uapi/linux/nl80211.h: * @NL80211_BAND_S1GHZ: around 900MHz, supported by S1G PHYs
uapi/linux/nl80211.h:	NL80211_BAND_S1GHZ,
net/cfg80211.h: * Only allowed for band NL80211_BAND_S1GHZ

$ grep -r IFLA_PERM_ADDRESS
uapi/linux/if_link.h:	IFLA_PERM_ADDRESS,
```

I also downloaded the Linux-libre kernel 5.4.243. As expected, the definitions are not present.
```
$ grep -r IFLA_PERM_ADDRESS
$ grep -r NL80211_BAND_S1GHZ
$
```
As a result, hcxdumptool will not compile running Trisquel Linux Nabia. 
An information about the kernel versions is here:
https://en.wikipedia.org/wiki/Trisquel

ZerBea, 2023-05-21:
I think we can close this issue report, because the requirements (see README.md) are not met.
But you can still ask your questions here.

closed 2023-05-21 by ZerBea

ZerBea, 2023-05-23:
No problem.

Unfortunately, gcc 9.4.0 is outdated, too:
https://gcc.gnu.org/gcc-9/

The Linux Developers decided not to port this (nice)  features (IFLA_PERM_ADDRESS, NL80211_BAND_S1GHZ) back to kernel 5.4 and I use Arch Linux, a rolling release distribution, which is always on the latest kernel and  latest software.
For me, the maintenance effort (to port hcxdumptool/hcxtools back to this old kernels) is far too great, e.g. porting IFLA_PERM_ADDRESS back to kernel 5.4 would mean that I have to go back to ethtool and ioctl() system calls.

alexvillalb159, 2023-05-25:
> No problem.
>
> Unfortunately, gcc 9.4.0 is outdated, too: https://gcc.gnu.org/gcc-9/
>
> The Linux Developers decided not to port this (nice) features (IFLA_PERM_ADDRESS, NL80211_BAND_S1GHZ) back to [...]

Dear Friend!:

But shouldn't the portability of hcxdumptool take precedence over the latest updates? Based on what you tell me, I'm afraid that hcxdumptool only runs on Arch Linux. That doesn't seem very fair. I guess the only option would be to use the old versions of hcxdumptool

ZerBea, 2023-05-25:
I think that's a misunderstanding.  The machine I use to code hcxdumptool is based on Arch Linux. That is correct.
But the requirements do not depend on the distribution (Arch, Raspbian, Debian, Ubuntu). They highly depend on the Linux kernel. In other words, hcxumptool runs on all Linux distributions if the requirements (Linux kernel > 5.10) met.

An example running hcxpcapngtool on a dump file recorded by a Raspberry Pi (running latest Raspbian):
```
file name................................: 20230509232912-wlan0.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.1.21+
application..............................: hcxdumptool 6.3.0-8-gc13470a
interface name...........................: wlan0
interface vendor.........................: fc3497
openSSL version..........................: 1.0
```

At least Linux kernel 5.10 (or greater) should be available on all distributions

According to this:
https://en.wikipedia.org/wiki/Trisquel
there is a new version of Trisquel Linux called Aramo v 11.0 LTS.
It is based on kernel Linux-libre 5.15  and latest hcxdumptool will work on it.

If you decide to stay on Nabia, the latest version of hcxdumptool is v6.2.9.
The versions, provided by Trisquel Linux are much older:
hcxdumptool (5.1.7-1):
https://packages.trisquel.org/nabia/hcxdumptool
hcxdumptool (6.2.5-2):
https://packages.trisquel.org/aramo/hcxdumptool

BTW:
It is not my decision to retire WIRELESS EXTENSIONS:
```
Since kernel 6.3 Kernel developer decided to print a dmesg warning if tools are detected that use deprecated WEXT:
[ 2770.939021] warning: `hcxdumptool' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
```
https://github.com/ZerBea/hcxdumptool/commit/9201a5781d9d76569c7a73e21c677ca4df7e8d44

To get the permanent MAC address of an interface, IFLA_PERM_ADDRESS is the NETLINK replacement of ioctl(SIOCETHTOOL) and ethtool is not longer installed on my system:
```
$ sudo pacman -Rns ethtool
error: target not found: ethtool
``` 
As a result, I can't maintain that old code any longer.

ZerBea, 2023-05-25:
But let me think about it to disable the new features if an old kernel is detected. This might work.

ZerBea, 2023-05-25:
Please try latest commit. Now hcxdumptool should compile on Linux kernel < 5.10.

Please notice that band e (S1GHZ) is disabled and hcxdumptool does not get the real HW MAC address because kernel < 5.10 do not provide this.

This is all I can do.

alexvillalb159, 2023-05-26:
> 

Dear Friend!:

I have compiled the almost latest version of linux kernel 6.3.3, on Trisquel GNU/Linux Nabia (10.0.1). I built the linux headers from the new kernel source code and finally compiled hcxdumptool with gcc version 9.4.0 which is available in Trisquel. All good.

However, compiling the linux kernel on the computer I use was hard, because it is quite old. I have to use cpupower-gui so it doesn't blow up, which however slows it down even more. Compiler ate all 4 Gb of memory compiling the kernel. I had to make more swap space available on disk so that it could finally finish compiling. And more things happened but I don't think it's worth commenting on them.

Is it worth having to compile a kernel to install hcxdumptool? I frankly doubt it. But you as a developer have the last word and I respect that. Thank you.

alexvillalb159, 2023-05-26:
> Please try latest commit. Now hcxdumptool should compile on Linux kernel < 5.10.
> 
> Please notice that band e (S1GHZ) is disabled and hcxdumptool does not get the real HW MAC address because kernel < 5.10 do not provide this.
> 
> This is all I can do.

Thanks for the update !. But as I told you, I already compiled the kernel. Let's say that I like challenges and I was able to enjoy the process a bit too... although it was hard.

ZerBea, 2023-05-26:
Changes are good (mostly) and kernel 6.3 offer a lot of them (new/updated drivers).