ZerBea/hcxdumptool #325: Tool not working with rtl8192eu driver

jalalbello, 2023-05-21:
## My issue is that the tool is not working, even though my adapter supports packet injection:
```
└─$ sudo aireplay-ng -9 wlan0

10:29:25  Trying broadcast probe requests...
10:29:25  Injection is working!
```

## Here is the output of the error
```
sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15

initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
driver doesn't support ioctl() SIOCSIWFREQ
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored
```

ZerBea, 2023-05-21:
hcxdumptool 6.2.6 is outdated.
Latest release is 6.3.0:
https://github.com/ZerBea/hcxdumptool/releases/tag/6.3.0

Please update either to v6.3.0 or to latest git head:
```
$ hcxdumptool -v
hcxdumptool 6.3.0-25-g27d99e6 (C) 2023 ZeroBeat

$ hcxdumptool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  2   5 08beac311ace 08beac311ace + wlp39s0f3u1u1u2  rtl8xxxu (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
[zerobeat@tux1 ~]$ hcxdumptool -I wlp39s0f3u1u1u2

Requesting interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  2   5 08beac311ace 08beac311ace + wlp39s0f3u1u1u2  rtl8xxxu (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] disabled

bye-bye
```

ZerBea, 2023-05-21:
Explanation:
hcxdumptool <= 6.2.9 use WIRELESS EXTENSIONS.
Realtek drivers are know that the do not like WIRELESS EXTENSIONS. Most of them need NETLINK.

hcxdumptool >= 6.3.0 use NETLINK instead of WIRELESS EXTENSIONS.

ZerBea, 2023-05-21:
Your driver (rtl8192eu) is outdated, too.
It is replaced by rtl8xxxu:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtl8xxxu?h=v6.3.3

See README.md for new REQUIREMENTS, e.g.:
```
* operating system: Linux (recommended: kernel >= 6.3, mandatory: kernel >= 5.10)
* gcc >= 12 recommended (deprecated versions are not supported: https://gcc.gnu.org/)
```

ZerBea, 2023-05-21:
Closed, because:
- Realtek incompatibility was fixed by v6.3.0.
- a new Realtek driver (rtl8xxxu) is available
but you can still ask your questions here.

closed 2023-05-21 by ZerBea

fa1rid, 2023-05-26:
@ZerBea What do you recommend USB WiFi nowadays with strong range and fast response? 
I see available online for example:
(TP-Link Archer T9UH) RTL8814AU
(TP-Link Archer T3U Plus) RTL8812BU
Any model you know is good?

Also it seems like now we can use Realtek chipsets without problem as you added support for NETLINK, right?
So any preference you prefer between Realtek and MediaTek?

Thanks!

ZerBea, 2023-05-27:
Correct, if a driver support NL80211 and provide full(!) monitor mode and full packet injection, hcxdumptool will work.
Please notice if a driver support NL80211 it doesn't mean that it support monitor mode or packet injection, too:
```
wifi: ath12k: driver for Qualcomm Wi-Fi 7 devices
Monitor mode is not (yet) supported.
```
https://lwn.net/Articles/915152/

I still prefer MediaTek chipsets, because the driver is part of the stock Linux kernel and it work out of the box.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/mediatek?h=v6.3.4

That include active monitor mode.
There is one exception: mt76x1u: active monitor mode is not working as expected
https://bugzilla.kernel.org/show_bug.cgi?id=217465
but there is a workaround to deactivate this feature since this commit:
https://github.com/ZerBea/hcxdumptool/commit/800b67c600472642648f1e868fd115b16acb314f

Neither RTL8814AU nor RTL8812BU are part of the stock Linux kernel
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek?h=v6.3.4
and you have to wrestle with third party drivers on every kernel update!

Regarding the excellent maintenance of the mt76 driver, the fact that it is part of the Linux kernel and provide active monitor mode, I recommend this one: AWUS036ACHM
https://www.alfa.com.tw/products/awus036achm?variant=39477247082568

fa1rid, 2023-05-28:
Okay Thanks. What else do you recommend other than the Alfa you mentioned.

By the way, I have Netgear A6210 (MT7612U) and I find it slow in scanning and jumping between channels, others I have are faster, any idea why?

Also I was wondering if the implementation of different brands differs while they are using the same chip, for performance, range, etc.?

ZerBea, 2023-05-28:
On version 6.3.0 default scan interval is one second. It increase by one second on every new AUTHENTICATION.
This behavior can be controlled by -t, -attemptclientmax and -attemptapmax.

On every case, it shouldbe a device with an external antenna connector

Since kernel 6.3 rtl8xxxu driver is working fine and you can use this cheap devices e.g.:
```
$ lsusb
Bus 001 Device 008: ID 2357:010c TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]
```

working out of the box:
```
$ hcxdumptool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1   4 503eaad5f1ac 503eaad5f1ac + wlp22s0f0u3      rtl8xxxu (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```

ZerBea, 2023-05-28:
Ralink (now MediaTek) devices are fine, too.