ZerBea/hcxdumptool #326: atheros ar9271 strange issue

bibi05, 2023-05-26:
Hi there! i red lot of issues but could not find any that would suit my case.. thus i am opening this one. Hope you will be ok to help me sort this out pls :) 
sudo hcxdumptool --do_rcascan -i wlan1mon  command issues the following info that i don't know how to resolve

initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
warning possible interfere: NetworkManager is running with pid 677

warning possible interfere: wpa_supplicant is running with pid 861

failed to detect wlan interface: interface possible not comected
driver doesn't support CFG80211
kernel possible complied without Wireless Extensions (expected: CONFIG_CFG80211_WEXT=y and CONFIG_CFG80211_WEXT_EXPORT=y): No such device
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored
failed to get interface information: No such device
failed to set interface down: No such device
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device

Otherwise other commands seems to work just fine.. 
thx in advance for your kind help

ZerBea, 2023-05-27:
hcxdumptool 6.2.6 is outdated. Please update to hcxdumptool 6.3.0 or to latest git version.

It is important to stop all(!) services that take access to the wlan interface. 
You received a warning that these 2 services are still running:
```
warning possible interfere: NetworkManager is running with pid 677
warning possible interfere: wpa_supplicant is running with pid 861
```

You have to stop them before you start hcxdumtool as mentioned in --help:
```
$ hcxdumptool --help
hcxdumptool 6.3.0-28-gc27e429  (C) 2023 ZeroBeat
usage: hcxdumptool <options>
        first stop all services that take access to the interface, e.g.:
        $ sudo systemctl stop NetworkManager.service
        $ sudo systemctl stop wpa_supplicant.service
        then run hcxdumptool
        press ctrl+c to terminate
        press GPIO button to terminate
        hardware modification is necessary, read more:
        https://github.com/ZerBea/hcxdumptool/tree/master/docs
        stop all services (e.g.: wpa_supplicant.service, NetworkManager.service) that take access to the interface
        do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
        do not use logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
        do not use virtual machines or emulators
        do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)
        do not use tools to change MAC (like macchanger)
        do not merge (pcapng) dump files, because this destroys assigned hash values!
```

After that, hcxdumptool should run fine:
```
$ lsusb
Bus 005 Device 006: ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n

$ hcxdumptool -v
hcxdumptool 6.3.0-28-gc27e429 (C) 2023 ZeroBeat

$ hcxdumptool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 00c0cac375a3 00c0cac375a3 + wlp39s0f3u1u4    ath9k_htc (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye

$ hcxdumptool -I wlp39s0f3u1u4

Requesting interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 00c0cac375a3 00c0cac375a3 + wlp39s0f3u1u4    ath9k_htc (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: 98

  2412 [  1] 30.0 dBm	  2417 [  2] 30.0 dBm	  2422 [  3] 30.0 dBm	  2427 [  4] 30.0 dBm
  2432 [  5] 30.0 dBm	  2437 [  6] 30.0 dBm	  2442 [  7] 30.0 dBm	  2447 [  8] 30.0 dBm
  2452 [  9] 30.0 dBm	  2457 [ 10] 30.0 dBm	  2462 [ 11] 30.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled	  2484 [ 14] disabled

bye-bye
```

Closed this report, because it is related to an outdated version of hcxdumptool, but you can still ask your questions here.

closed 2023-05-27 by ZerBea

bibi05, 2023-05-27:
Thank you so much for your prompt feedback! i am really newbie and start to learn.. i don't exactly know how to update to latest version (shame)) could you pls let me know the git command so that i can learn at same time:)

ZerBea, 2023-05-27:
First uninstall hcxdumptool 6.2.6 (make uninstall)
or locate hcxdumptool and hcxpioff and rm them
Depending on your installation they are located in usr/bin or /usr/local/bin

If everything is ok you should get a Command not found message 

Than remove entire hcxdumptool directory

Now run
$ git clone https://github.com/ZerBea/hcxdumptool
$ cd hcxdumptool
$ make
$ sudo make install

$ hcxdumptool -v
should show you the actual version.