ZerBea/hcxdumptool #328: Driver is busy, chipset mt7921u

acekicker47, 2023-06-30:
─# hcxdumptool -i wlan0mon -o Wi-Fi_PMKID.pcapng --disable_deauthentication --disable_client_attacks --enable_status=3
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlan0mon
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 3 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: e0e1a938a3d3 (not used for the attack)
INTERFACE VIRTUAL MAC.....: e0e1a938a3d3 (not used for the attack)
DRIVER....................: mt7921u
DRIVER VERSION............: 6.1.0-kali9-amd64
DRIVER FIRMWARE VERSION...: ____010000-20230331110939
openSSL version...........: 1.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 6096200c88f7 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 6096200c88f8 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 6096200c88f9 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: e804102a355d
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62176
ANONCE....................: 92b17d3dd7a38c0ad4ae90fea10d64e622edb14dfd100123d3687089cfead32a
SNONCE....................: 64771a2d5285901939139ea468593e761820aa8af00a9a555165671216b3d1f6

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
01:57:55 2412/1 driver is busy: failed to transmit acknowledgement
01:57:56 2417/2 driver is busy: failed to transmit acknowledgement
01:57:57 2417/2 driver is busy: failed to transmit acknowledgement
01:57:57 2417/2 driver is busy: failed to transmit acknowledgement
01:57:58 2417/2 driver is busy: failed to transmit acknowledgement
01:57:58 2417/2 driver is busy: failed to transmit acknowledgement
01:57:59 2417/2 driver is busy: failed to transmit acknowledgement
01:57:59 2417/2 driver is busy: failed to transmit acknowledgement
01:58:00 2422/3 driver is busy: failed to transmit acknowledgement
01:58:00 2422/3 driver is busy: failed to transmit acknowledgement
01:58:01 2422/3 driver is busy: failed to transmit acknowledgement
01:58:01 2422/3 driver is busy: failed to transmit acknowledgement
01:58:02 2422/3 driver is busy: failed to transmit acknowledgement
01:58:02 2422/3 driver is busy: failed to transmit acknowledgement
01:58:03 2422/3 driver is busy: failed to transmit acknowledgement
01:58:03 2422/3 driver is busy: failed to transmit acknowledgement
01:58:04 2427/4 driver is busy: failed to transmit acknowledgement
01:58:05 2427/4 driver is busy: failed to transmit acknowledgement
01:58:05 2427/4 driver is busy: failed to transmit acknowledgement
01:58:06 2427/4 driver is busy: failed to transmit acknowledgement
01:58:06 2427/4 driver is busy: failed to transmit acknowledgement
01:58:07 2427/4 driver is busy: failed to transmit acknowledgement
01:58:07 2427/4 driver is busy: failed to transmit acknowledgement
01:58:08 2432/5 driver is busy: failed to transmit acknowledgement
01:58:08 2432/5 driver is busy: failed to transmit acknowledgement
01:58:09 2432/5 driver is busy: failed to transmit acknowledgement
01:58:09 2432/5 driver is busy: failed to transmit acknowledgement
01:58:10 2432/5 driver is busy: failed to transmit acknowledgement
01:58:10 2432/5 driver is busy: failed to transmit acknowledgement
01:58:11 2432/5 driver is busy: failed to transmit acknowledgement
01:58:11 2432/5 driver is busy: failed to transmit acknowledgement
01:58:12 2437/6 driver is busy: failed to transmit acknowledgement
01:58:12 2437/6 driver is busy: failed to transmit acknowledgement
01:58:13 2437/6 driver is busy: failed to transmit acknowledgement
01:58:13 2437/6 driver is busy: failed to transmit acknowledgement
01:58:14 2437/6 driver is busy: failed to transmit acknowledgement
01:58:14 2437/6 driver is busy: failed to transmit acknowledgement
01:58:15 2437/6 driver is busy: failed to transmit acknowledgement
01:58:15 2437/6 driver is busy: failed to transmit acknowledgement
01:58:16 2442/7 driver is busy: failed to transmit acknowledgement
01:58:16 2442/7 driver is busy: failed to transmit acknowledgement
01:58:17 2442/7 driver is busy: failed to transmit acknowledgement
01:58:17 2442/7 driver is busy: failed to transmit acknowledgement
01:58:18 2442/7 driver is busy: failed to transmit acknowledgement
01:58:18 2442/7 driver is busy: failed to transmit acknowledgement
01:58:19 2442/7 driver is busy: failed to transmit acknowledgement

So i get that error every time and not capture any PMKID, anyone know how to fix it ?

before start hcxdumptool i run:
airmon-ng check kill
airmon-ng start wlan0

thanks.

ZerBea, 2023-06-30:
Hcxdumptool 6.2.6 is completely outdated. Please use v6.3.0 or latest git head. This version will activate "active monitor mode" on MediaTek chipsets (do not use airmon-ng, because it doesn't provide this mode).

You are running hcxdumptool on a virtual monitor interface (wlan0mon).
That is nor recommended. From README.md:
```
        stop all services (e.g.: wpa_supplicant.service, NetworkManager.service) that take access to the interface
        do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
        do not use logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
        do not use virtual machines or emulators
        do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)
        do not use tools to change MAC (like macchanger)
        do not merge (pcapng) dump files, because this destroys assigned hash values!
```

Please notice:
aircrack-ng tools are designed to work together with other aircrack-ng tools inside the aircrack-ng sweet.
hcxdumptool is not part of this suite - this could lead to unexpected results.

Closed this issue report, because it is related to an outdated version on hcxdumptool, but you can still ask your questions here.

closed 2023-06-30 by ZerBea

ZerBea, 2023-06-30:
Looks like KALI is still running the outdated version:
https://bugs.kali.org/view.php?id=8360