ZerBea/hcxdumptool #329: No long options --active_beacon & --enable_status in hcxdumptool 6.3.1

piggy2011, 2023-07-03:
I have installed the latest hcxdumptool 6.3.1 successfully on WSL2 Kali Linux and the wireless adapter(rtl88XXau) is also installed. Now, I'd like to run the command "hcxdumptool hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15" but it seems that neither --active_beacon nor --enable_status is provided in  hcxdumptool 6.3.1.  Would you pls help suggest what the command should be used in this hcxdumptool version? Thanks a lot!

ZerBea, 2023-07-03:
active beacon is replaced by:
```
--beacontx=<digit>             : transmit BEACON of first n entries of ESSID list
                                  default: 10
```

status is enabled by default and can be sorted by --rds=1

usually this command line is enough:
`$ hcxdumptool -i wlan0 -o dumpfile.pcapng --rds=1`

If you need more than 10 ESSIDs to be beaconed, add becaontx:
`$ hcxdumptool -i wlan0 -o dumpfile.pcapng --becaontx=20 --rds=1`

piggy2011, 2023-07-03:
> active beacon is replaced by:
>
> ```
> --beacontx=<digit>             : transmit BEACON of first n entries of ESSID list
>                                   default: 10
> ```
>
> status is enabled by default [...]

Hi Zerbea, thanks a lot for your reply.  -o is replaced by -w according to hcxdumptool 6.3.1 help. Thanks!

ZerBea, 2023-07-03:
Correct, thanks - nasty copy and paste error (-o -> -w).

closed 2023-07-03 by ZerBea