ZerBea/hcxdumptool #331: Is captured real display (1, 2, 3, P, S) enough to crack the WiFi?

piggy2011, 2023-07-06:
[image]

I have captured the wireless information by the command " sudo hcxdumptool -i wlan0 -w dumpfile-2.pcapng --rds=1" (hcxdumptool 6.3.1), but I am not sure that real display (1, 2, 3, P, S) is enough to crack the Wifi Connie-ASUS. Would you pls help confirm it? Thanks a lot!

ZerBea, 2023-07-07:
Regarding the NETWORK: Connie-ASUS
You got:
```
a PMKID
an authorized handshake M1M2M3
and an EAPOL M2 ROGUE (AP LESS attack)
```
That is more than enough to recover the PSK:

This command show an information about the content of the hc 22000 file:
`$ hcxhashtool -i hash.hc22000 --info=stdout`

This command filter the target ESSID:
`$ hcxhashtool -i hash.hc22000 --essid="Connie-ASUS" -o target.hc22000 

Run hashcat on all hashes:
```
$ hcxpcapngtool -o hash.hc22000 dumpfile-2.pcapng
$ hashcat -m 22000 hash.hc22000 wordlist
```

Or run hashscat on the target:
```
$ hashcat -m 22000 target.hc22000 wordlist
```

To recover a PSK it is enough to get a PMKID (P), to get an authorized handshake (3)
or, if the target AP is not in RANGE, an EAPOL M2 ROGUE CHALLENGE (2).

Closed, because this is more a discussion than an issue and hcxdumptool is doing exactly what expected.

ZerBea, 2023-07-07:
BTW:
To find out if the PSK is inside the WiFi traffic you can do this:
```
$ hcxpcapngtool -o hash.hc22000 -E wl -I wl -U-wl -R wl dumpfile-2.pcapng
$ hashcat -m 22000 hash.hc22000 wl
```

To find out if the PSK is part of the BSSID or the ESSID you can use this command:
`$ hcxpsktool -c target.hc22000 | hashcat -m 22000 target.hc22000`

closed 2023-07-07 by ZerBea

ZerBea, 2023-07-07:
Explanation is here:
https://github.com/ZerBea/hcxdumptool/discussions/330