ZerBea/hcxdumptool #332: Doesn't work on Android Qualcomm chipset

ulist, 2023-07-22:
```
../hcxtools/hcxdumptool # echo 4 > /sys/module/wlan/parameters/con_mode                                  .../hcxtools/hcxdumptool # hcxdumptool -i  wlan0                                                          
Requesting interface capabilities. This may take some time.
Please be patient...

failed to arm interface                                                                                                                                                                                             2 errors during runtime                                                                                                                                                                                             bye-bye

```
I know that Qualcomm Is listed as unrecommened because of no packet Injection , but that doesn't prevent it from capturing pmkid right?

ZerBea, 2023-07-23:
hcxdumptool is an interactive tool. There is no single receive only mode. The driver must support full monitor mode and full packet injection via RTNETLINK and NL80211 as well as channel management via NL80211.

That is not he case, e.g.:
https://unix.stackexchange.com/questions/709392/does-qualcomm-qca9377-support-packet-injection
https://developer.qualcomm.com/forum/qdn-forums/hardware/qca9377/34827

I don' see a way that hcxdumptool will work on this driver.

ZerBea, 2023-07-23:
To record WiFi traffic, I recommend tshark.

ZerBea, 2023-07-24:
I think we can close this, because the driver doesn't support to set monitor mode by NL80211. Instead it use its own mode:
`echo 4 > /sys/module/wlan/parameters/con_mode`

Also it doesn't support full packet injection:
```
Known bugs/limitations

    firmware does not support association to the same AP from different virtual STA interfaces (driver prints “ath10k: Failed to add peer XX:XX:XX:XX:XX:XX for VDEV: X” in that case)
    packet injection isn't supported yet
    applying ath9k regulatory domain hack patch from OpenWRT causes firmware crash (reason: regulatory hint function is never called and ath10k never sends scan channel list to the firmware which in turn causes firmware to crash on scan)
    tx rate is reported as 6mbps due to firmware limitation (no tx rate information in tx completions); instead see /sys/kernel/debug/ieee80211/phyX/ath10k/fw_stats
    WEP doesn't work with AP_VLANs - frames are sent unencrypted (observed on: 999.999.0.636, 10.2.4.20-1, 10.1.467.2-1)
    TX speeds are extremely poor on certain chips (QCA6174 is one). A patch (https://gist.github.com/harrykipper/d1bedb234c4af0692f7ccd33329a02d7) solves the issue in most cases (https://bbs.archlinux.org/viewtopic.php?pid=1689990#p1689990))
```
https://wireless.wiki.kernel.org/en/users/Drivers/ath10k

closed 2023-07-24 by ZerBea

ulist, 2023-07-24:
Well it's a qcacld3.0 qca6390 and I can send deauth packets in aireplay-ng but must run airodump-ng in another session

ZerBea, 2023-07-24:
We can try this:
Search for this line:
`if(nl_set_monitormode() == false) return false;`

and comment it:
`# if(nl_set_monitormode() == false) return false;`

set monitor mode by qualcomm
`echo 4 > /sys/module/wlan/parameters/con_mode`

and run hcxdumptool.

This is the function where we arm the interface:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L3689

ZerBea, 2023-07-24:
It is mandatory that the driver support NL80211, To discover what NL80211 features are supported you can monitor this by nlmon:

activate nlmon:
$ sudo ip link add  nlmon0 type nlmon
$ sudo ip link set dev nlmon0 up

run Wireshark and capture traffic from nlmon
run
$ hcxdumptool -L 
$ hcxdumptool -I your_wlan_interface 

stop hcxdumptool
stop Wireshark capturing and save the dump file
add the dump file here.

ZerBea, 2023-07-24:
BTW:
hcxdumptool transmit several different types of frames because it does complete AUTHENTICATIONs with APs and CLIENTs.
Additional it has a channel management. All this is done via RTNETLINK and NL80211 messages.
The driver must support that.

ZerBea, 2023-07-24:
BTW:
Please comment the output of
$ hcxdumptool -L
and
$ hcxdumptool -I your_wlan_interface

This are NETLINK requests to retrieve the features of the driver/interface.

ZerBea, 2023-07-24:
> Well it's a qcacld3.0 qca6390 and I can send deauth packets in aireplay-ng but must run airodump-ng in another session

That explains a lot, because it is not a native Linux driver provided by the Linux kernel.

ZerBea, 2023-07-24:
You may have noticed that I changed the head line. Problem is the driver,

ZerBea, 2023-07-24:
Looks like there are some more problems on this driver:
https://github.com/aircrack-ng/aircrack-ng/issues/2179

ulist, 2023-07-24:
I think I need to recompile the kernel with CONFIG_NLMON=y for this to work 
```
# ip link add nlmon0 type nlmon
RTNETLINK answers: Operation not supported on transport endpoint
```

ulist, 2023-07-24:
I think I need to recompile the kernel with CONFIG_NLMON=y for this to work 
```
# ip link add nlmon0 type nlmon
RTNETLINK answers: Operation not supported on transport endpoint
```
I did comment and recompiled hcxdumptool
```
root@localhost:~/hcxdumptool# ./hcxdumptool  -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0  28 000000000000 00037f12d7fb + wlan0            cnss_pci (NETLINK)

* active monitor mode available                                                                                                      + monitor mode available
- no monitor mode available

bye-bye
root@localhost:~/hcxdumptool# ./hcxdumptool  -i wlan0

Requesting interface capabilities. This may take some time.
Please be patient...

failed to arm interface

2 errors during runtime

bye-bye
```
And Tshark isn't working 
```
root@localhost:~/hcxdumptool# tshark  -i wlan0
Running as user "root" and group "root". This could be dangerous.
Capturing on 'wlan0'
tshark: Child dumpcap process died: Segmentation violation
0 packets captured
```

ulist, 2023-07-24:
I do have an external wifi adapter and it does work but I always prefer to use internal interface

ulist, 2023-07-24:
I do have CONFIG_NL80211_TESTMODE=y in the kennel , not sure if it does anything

ZerBea, 2023-07-24:
Thanks for the tests.
The driver reported that it does support NETLINK and it does have monitor mode.

Attached a special debug version. Please decompress and replace hcxdumptool.c by this version.
[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/12149637/hcxdumptool.c.zip)
Than recompile and run hcxdumptool.

Now every call inside the "arm" function got a debug printf. It will show which RTNETLINK or NL80211 command is not supported.

Please comment the output of the debug messages.

ZerBea, 2023-07-24:
This
`  0  28 000000000000 00037f12d7fb + wlan0            cnss_pci (NETLINK)`
is definitely not good. It shows that at least the RTNETLINK command to retrieve the hardware MAC address failed.

ZerBea, 2023-07-24:
There is no longer need to activate nlmon, because we now can do this by the debug printf's mentioned above.

ulist, 2023-07-24:
```
root@localhost:~/hcxdumptool# make
cc -O3 -Wall -Wextra -Wpedantic -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.1-27-gd462972\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
hcxdumptool.c:4671:45: error: use of undeclared identifier 'HCX_EXIT_ON_EAPOL'
        {"exitoneapol",                 required_argument,      NULL,   HCX_EXIT_ON_EAPOL},
                                                                        ^
hcxdumptool.c:4809:8: error: use of undeclared identifier 'HCX_EXIT_ON_EAPOL'
                case HCX_EXIT_ON_EAPOL:
                     ^
hcxdumptool.c:4811:41: error: use of undeclared identifier 'EXIT_ON_EAPOL_PMKID'
                exiteapolpmkidflag |= exiteapolflag & EXIT_ON_EAPOL_PMKID;
                                                      ^
hcxdumptool.c:4812:38: error: use of undeclared identifier 'EXIT_ON_EAPOL_M2'
                exiteapolm2flag |= exiteapolflag & EXIT_ON_EAPOL_M2;
                                                   ^
hcxdumptool.c:4813:38: error: use of undeclared identifier 'EXIT_ON_EAPOL_M3'
                exiteapolm3flag |= exiteapolflag & EXIT_ON_EAPOL_M3;
                                                   ^
hcxdumptool.c:5122:22: error: use of undeclared identifier 'EXIT_ON_EAPOL_PMKID'
        if((wanteventflag & EXIT_ON_EAPOL_PMKID) == EXIT_ON_EAPOL_PMKID) fprintf(stdout, "\nexit on PMKID\n");
                            ^
hcxdumptool.c:5122:46: error: use of undeclared identifier 'EXIT_ON_EAPOL_PMKID'
        if((wanteventflag & EXIT_ON_EAPOL_PMKID) == EXIT_ON_EAPOL_PMKID) fprintf(stdout, "\nexit on PMKID\n");
                                                    ^
hcxdumptool.c:5123:22: error: use of undeclared identifier 'EXIT_ON_EAPOL_M2'
        if((wanteventflag & EXIT_ON_EAPOL_M2) == EXIT_ON_EAPOL_M2) fprintf(stdout, "\nexit on EAPOL M1M2\n");
                            ^
hcxdumptool.c:5123:43: error: use of undeclared identifier 'EXIT_ON_EAPOL_M2'
        if((wanteventflag & EXIT_ON_EAPOL_M2) == EXIT_ON_EAPOL_M2) fprintf(stdout, "\nexit on EAPOL M1M2\n");
                                                 ^
hcxdumptool.c:5124:22: error: use of undeclared identifier 'EXIT_ON_EAPOL_M3'
        if((wanteventflag & EXIT_ON_EAPOL_M3) == EXIT_ON_EAPOL_M3) fprintf(stdout, "\nexit on EAPOL M1M2M3\n");
                            ^
hcxdumptool.c:5124:43: error: use of undeclared identifier 'EXIT_ON_EAPOL_M3'
        if((wanteventflag & EXIT_ON_EAPOL_M3) == EXIT_ON_EAPOL_M3) fprintf(stdout, "\nexit on EAPOL M1M2M3\n");
                                                 ^
11 errors generated.
make: *** [Makefile:66: hcxdumptool] Error 1
root@localhost:~/hcxdumptool#
```

ZerBea, 2023-07-25:
You need to do a fresh git clone, before replacing hcxdumptool.c
```
$ ./hcxdumptool -v
hcxdumptool 6.3.1-30-g1a97430 (C) 2023 ZeroBeat
```

because there are a lot of changes:
```
24.07.2023
==========
hcxdumptool: added new option to EXIT on EAPOL
--exitoneapol=<type>           : exit on first EAPOL occurrence:
                                  bitmask:
                                   1 = PMKID
                                   2 = EAPOL M2
                                   4 = EAPOL M3
                                  target BPF filter is recommended
get more information here:
https://github.com/ZerBea/hcxdumptool/issues/333#issuecomment-1647684235
```
and hcxdumptool.c need the new includes.

ZerBea, 2023-07-25:
BTW: airodump-ng.strace.txt is very interesting, too
It looks like airodump-ng is running some deprecated WIRELESS EXTENSIONS (WEXT) tools in background (e.g. iwpriv):
`15885 execve("/bin/sh", ["sh", "-c", "iwpriv wlan0 2>/dev/null | grep "...], 0xffffec88d2b0 /* 10 vars */ <unfinished ...>`

WEXT is deprecated and has been replaced by ip and iw.
New development should be focused on cfg80211 and nl80211.
https://wireless.wiki.kernel.org/en/developers/Documentation/Wireless-Extensions

As a result, several WEXT calls failed:
```
15886 ioctl(3, SIOCGIWPRIV, 0xffffe67b0a08) = -1 E2BIG (Argument list too long)
15875 ioctl(6, SIOCGIWMODE, 0xffffec888818) = -1 EOPNOTSUPP (Operation not supported)
```

ulist, 2023-07-25:
```
root@localhost:~/hcxdumptool# ./hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0  42 000000000000 00037f12d7fb + wlan0            cnss_pci (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
root@localhost:~/hcxdumptool# ./hcxdumptool -i wlan0

Requesting physical interface capabilities. This may take some time.
Please be patient...

rtnetlink failed to set interface mac
failed to arm interface

2 errors during runtime

bye-bye
```
Iw dev doesn't show the channel 
```
root@localhost:~/hcxdumptool# iw dev wlan0  set channel 11
root@localhost:~/hcxdumptool# iw dev
phy#0
        Interface wlan0
                ifindex 42
                wdev 0x10
                addr 00:03:7f:12:d7:fb
                type monitor
                txpower 0.00 dBm
root@localhost:~/hcxdumptool#
```

ulist, 2023-07-25:
I have fixed this aircrack-ng/aircrack-ng#2555 by reflashing the stock rom and back to custom rom, I think the issue is tied to an old firmware labeled F.04 I had to flash before flashing the custom rom, when I flashed the stock rom it contained a newer firmware C.06 , and now airodump-ng works fine in the custom rom

ulist, 2023-07-25:
But I noticed the switching between managed and monitor mode 
```
echo 4 > /sys/module/wlan/parameters/con_mode
echo 0 > /sys/module/wlan/parameters/con_mode
```
in stock kernel happens instantly in the stock rom as compared to custom rom it takes a minute or so which Is annoying I think it's because of the custom kernel might be missing some  configs or misconfigured qcacld profile

ZerBea, 2023-07-25:
Ok, thanks for the information.

I see 2 problems
If we compare my iw output with your it looks like the driver does not allow to set/get frequency channel information:
We can't set a MAC, because the driver doe not allow this.

Attached another try that fixed:
`rtnetlink failed to set interface mac`
We do not need to set a virtual MAC because the driver does not support active monitor mode.

[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/12161498/hcxdumptool.c.zip)

Same procedure. Decompress, replace hcxdumptool.c by this version, compile and run it.
Let's see what happens.

ZerBea, 2023-07-25:
Thanks.
On the bright side we fixed one problem.
But unfortunately we run into another one:
Driver reported that it support NL80211, but it doesn't respond to a standard NL80211 request to get device information:

Added some more debug messages to the NL80211 standard request here: static bool nl_get_interfacestatus

[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/12163287/hcxdumptool.c.zip)

Same procedure as mentioned above. Let's see what happens.

ulist, 2023-07-25:
```
root@localhost:~/hcxdumptool# make
make: Warning: File 'hcxdumptool.c' has modification time 6906 s in the future
cc -O3 -Wall -Wextra -Wpedantic -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.1-30-g1a97430\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
make: warning:  Clock skew detected.  Your build may be incomplete.
root@localhost:~/hcxdumptool# ./hcxdumptool  -i wlan0

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0  74 000000000000 00037f12d7fb + wlan0            cnss_pci (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: 00

  2412 [  1] 19.0 dBm     2417 [  2] 19.0 dBm     2422 [  3] 19.0 dBm     2427 [  4] 19.0 dBm
  2432 [  5] 19.0 dBm     2437 [  6] 19.0 dBm     2442 [  7] 19.0 dBm     2447 [  8] 19.0 dBm
  2452 [  9] 19.0 dBm     2457 [ 10] 19.0 dBm     2462 [ 11] 19.0 dBm     2467 [ 12] 19.0 dBm
  2472 [ 13] 19.0 dBm     2484 [ 14] disabled     5180 [ 36] 19.0 dBm     5200 [ 40] 19.0 dBm
  5220 [ 44] 19.0 dBm     5240 [ 48] 19.0 dBm     5260 [ 52] 19.0 dBm     5280 [ 56] 19.0 dBm
  5300 [ 60] 19.0 dBm     5320 [ 64] 19.0 dBm     5500 [100] 19.0 dBm     5520 [104] 19.0 dBm
  5540 [108] 19.0 dBm     5560 [112] 19.0 dBm     5580 [116] 19.0 dBm     5600 [120] 19.0 dBm
  5620 [124] 19.0 dBm     5640 [128] 19.0 dBm     5660 [132] 19.0 dBm     5680 [136] 19.0 dBm
  5700 [140] 19.0 dBm     5720 [144] disabled     5745 [149] disabled     5765 [153] disabled
  5785 [157] disabled     5805 [161] disabled     5825 [165] disabled     5845 [169] disabled
  5865 [173] disabled

scan frequencies: frequency [channel] of Regulatory Domain: 00

  2412 [  1]      2437 [  6]      2462 [ 11]
failed to open raw packet socket

1 errors during runtime

bye-bye
root@localhost:~/hcxdumptool#
```

ZerBea, 2023-07-25:
Ok, thanks again for your patience.

According to this:
https://thomasw.dev/post/packet_ignore_outgoing/
PACKET_IGNORE_OUTGOING shopuld be available since Linux kernel 4.20

We check this here:
```
#if(LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
if(setsockopt(fd_socket_rx, SOL_PACKET, PACKET_IGNORE_OUTGOING, &enable, sizeof(int)) < 0)
	{
	perror("add set PACKET_IGNORE_OUTGOING failed");
	return false;
	}
#endif
```

and compile it only if the Linux kernel is >= 4.20.
What is your kernel version?

Removed PACKET_IGNORE_OUTGOING in this version. Please try.
[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/12163635/hcxdumptool.c.zip)

ulist, 2023-07-25:
My kernel is msm4.19

ulist, 2023-07-25:
Output before scan begins 
[image]

ZerBea, 2023-07-25:
Ok, thanks.

Unfortunately this problem "NL80211 failed to set frequency" is expected and confirmed by iw here:
https://github.com/ZerBea/hcxdumptool/issues/332#issuecomment-1649721463

The driver does not allow to set the frequency.
Automatic channel management via NL80211 is an integral part of hcxdumptool.
I can't disable, change or remove this without a huge impact on stock Linux kernel drivers which are working perfect.

Also R 1 3 P are empty which means that frame injection is not working as expected.

ulist, 2023-07-25:
I can execute iw to set the channel but I can't run it twice on the same channel
```
.../files/home # iw dev wlan0  set channel 11
command failed: Device or resource busy (-16)
.../files/home # ifconfig  wlan0 up
.../files/home # iw dev wlan0  set channel 11
.../files/home # iw dev wlan0  set channel 12
.../files/home # iw dev wlan0  set channel 6
.../files/home # iw dev wlan0  set channel 6
command failed: Connection timed out (-110)
.../files/home # time iw dev wlan0  set channel 6
command failed: Connection timed out (-110)

real    0m12.092s
user    0m0.008s
sys     0m0.008s
.../files/home # time iw dev wlan0  set channel 5

real    0m0.035s
user    0m0.007s
sys     0m0.007s
.../files/home # time iw dev wlan0  set channel 3

real    0m0.062s
user    0m0.007s
sys     0m0.007s
.../files/home # time iw dev wlan0  set channel 3
command failed: Connection timed out (-110)

real    0m12.155s
user    0m0.007s
sys     0m0.009s
.../files/home #
```
and airodump-ng can go through all the channels and I can run it more than once 
[image]

ZerBea, 2023-07-25:
You can't compare the behavior of airodump-ng (pasive) and hcxdumptool (interacttive).

I pushed all changes to latest git head. Please do a fresh git clone, compile and run:

./hcxdumptool  -i wlan0 --rcascan=active

If everything is working as expected you should see
timestamps on BEACON and  RESPONSE (frame injection is working)
on different channels CHA and  FREQ.

If not, the driver can't handle hcxdumptool's messages.

ulist, 2023-07-25:
[image]

ZerBea, 2023-07-25:
Thanks for the screenshot.
Packet injection is not working (RESPONSE timestamp does not increase)
Channel management is working (we can see different channels).

If you stop hcxdumptool and restart it, the driver is blocked. It can not handle hcxdumptool's requests any longer.

There is nothing more I can do to let hcxdumptool work on this driver.

BTW:
Is packet injection is working if you run airodump-ng and aireplay-ng -9 in parallel?

ZerBea, 2023-07-25:
No. Injection is not working:
```
21:28:50  Trying broadcast probe requests...
21:28:52  No Answer...
```
and
`21:28:19  Sending 64 directed DeAuth (code 7). STMAC: [C8:C2:FA:C6:4F:78] [ 5| 0 ACKs]`
0 ACKs = no packet received by the target.

ulist, 2023-07-25:
Why can he #251 retrieve his mac address and I don't even though hw-mac , virtual-mac are the same

ZerBea, 2023-07-25:
Old hcxdumptool v 6.2.6 got this information via ethtool API:
```
epmaddr = (struct ethtool_perm_addr*)calloc(1, sizeof(struct ethtool_perm_addr) +6);
if(!epmaddr)
	{
	perror("failed to malloc memory for permanent hardware address");
	return false;
	}
memset(&ifr, 0, sizeof(ifr));
memcpy(&ifr.ifr_name, interfacename, IFNAMSIZ);
epmaddr->cmd = ETHTOOL_GPERMADDR;
epmaddr->size = 6;
ifr.ifr_data = (char*)epmaddr;
if(ioctl(fd_socket, SIOCETHTOOL, &ifr) < 0)
	{
	perror("failed to get permanent hardware address, ioctl(SIOCETHTOOL) not supported by driver");
	return false;
	}
if(epmaddr->size != 6)
	{
	fprintf(stderr, "failed to get permanent hardware address length\n");
	return false;
	}
memcpy(&mac_orig, epmaddr->data, 6);
free(epmaddr);
```

Since v6.3.0 this has been removed in favor of RTNETLINK:
```
			#if(LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0))
			if(rta->rta_type == IFLA_PERM_ADDRESS)
				{
				if(rta->rta_len == 10) memcpy(hwmac, rta_data(rta), ETH_ALEN);
				}
			#else
			if(rta->rta_type == IFLA_ADDRESS)
				{
				if(rta->rta_len == 10) memcpy(hwmac, rta_data(rta), ETH_ALEN);
				}
			#endif
```

More information  is here:
https://kernel.org/doc/html/next/userspace-api/netlink/intro.html
e.g.:
```
Introduction to Netlink
Netlink is often described as an ioctl() replacement. It aims to replace fixed-format C structures as supplied to ioctl() with a format which allows an easy way to add or extended the arguments.
```
or here:
https://www.kernel.org/doc/html/latest/driver-api/ioctl.html
e.g.:
```
Alternatives to ioctl
There are many cases in which ioctl is not the best solution for a problem. Alternatives include:
...
netlink is the preferred way of configuring any network related objects through sockets.
```

ZerBea, 2023-07-26:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/1bc1768a6dbbe38917f0d41550ee275db32ccc39
I improved hcxdumptool -v to get some information about the environment at compile time.

On my system, output looks like this:
```
$ hcxdumptool -v
hcxdumptool 6.3.1-36-g1bc1768 (C) 2023 ZeroBeat
compiled by gcc 13.1.1
Linux API headers 6.3.0
```

It is not mandatory that the Linux API header match the actual kernel, e.g. on my system:
```
$ uname -r
6.4.4-arch1-1
```

But all values should not be below the requirements.

Can you please run latest git head and comment the output of hcxdumptool -v, because I still wonder why some defines are not working as expected on Android.

ulist, 2023-07-26:
```
root@localhost:~/hcxdumptool# ./hcxdumptool -v
hcxdumptool 6.3.1-40-gd399f14 (C) 2023 ZeroBeat
compiled by gcc 4.2.1
compiled with Linux API headers 5.15.99
root@localhost:~/hcxdumptool#
```

ZerBea, 2023-07-26:
Thanks.
It looks like we are running into a really ugly problem.
I can't fix that, because it is related to your environment / configuration and not to hcxdumptool.
hcxdumptool is compiled  using API headers 5.15.99 but your Android Kernel is 4.19
https://github.com/ZerBea/hcxdumptool/issues/332#issuecomment-1650195397

gcc compile functions and values of Linux API 5.15 to hcxdumptool which are not present in the active kernel 4.19.
I think you can imagine what happens if you execute hcxdumptool on kernel 4.19 and  it reach such a function...

ZerBea, 2023-07-26:
A real life example:

Here we check the Linux API header:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c

```
			#if(LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0))
			if(rta->rta_type == IFLA_PERM_ADDRESS)
				{
				if(rta->rta_len == 10) memcpy(hwmac, rta_data(rta), ETH_ALEN);
				}
			#else
			if(rta->rta_type == IFLA_ADDRESS)
				{
				if(rta->rta_len == 10) memcpy(hwmac, rta_data(rta), ETH_ALEN);
				}
			#endif
```

If the API version is > 5.10.0 we use IFLA_PERM_ADDRESS to get the permanent hardware MAC.
If it is not present, we use IFLA_ADDRESS to get the MAC

According to your API (5.15.99) IFLA_PERM_ADDRESS should be present and this code is compiled.
But your kernel is 4.19 and IFLA_PERM_ADDRESS is not available there. From now on, the results are unexpected.

ulist, 2023-07-27:
I'm not sure where to find the API headers but I found this https://docs.kernel.org/kbuild/headers_install.html
So I went to my kernel make install the headers pack it up and unpack it to /usr in chroot environment then I compiled hcxdumptool and it said this 
```
root@localhost:~/hcxdumptool# ./hcxdumptool  -v
hcxdumptool 6.3.1-40-gd399f14 (C) 2023 ZeroBeat
compiled by gcc 4.2.1
Linux API headers (LINUX_VERSION_MAJOR) is not defined
```
I looked it up and found this patchwork https://patchwork.kernel.org/project/linux-kbuild/patch/20220207115212.217744-1-jiaxun.yang@flygoat.com/
I applied it did the same thing again and when compile I get this
```
root@localhost:~/hcxdumptool# make
cc -O3 -Wall -Wextra -Wpedantic -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.1-40-gd399f14\" -DVERSION_YEAR=\"2023\" -DSTATUSOUT -DNMEAOUT
hcxdumptool.c:659:28: error: function-like macro 'KERNEL_VERSION' is not defined
        #if(LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0))
                                  ^
hcxdumptool.c:3583:30: error: function-like macro 'KERNEL_VERSION' is not defined
                        #if(LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0))
                                                  ^
hcxdumptool.c:3777:30: error: function-like macro 'KERNEL_VERSION' is not defined
                        #if(LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0))
                                                  ^
hcxdumptool.c:3873:27: error: function-like macro 'KERNEL_VERSION' is not defined
#if(LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
                          ^
hcxdumptool.c:3898:27: error: function-like macro 'KERNEL_VERSION' is not defined
#if(LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
                          ^
hcxdumptool.c:4454:28: error: function-like macro 'KERNEL_VERSION' is not defined
        #if(LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0))
                                  ^
6 errors generated.
make: *** [Makefile:66: hcxdumptool] Error 1
```

ZerBea, 2023-07-27:
Looks like this patch destroyed the macros.

Usually this macros are located here:
```
$ locate linux/version.h
/usr/include/linux/version.h
```
and the compiler (gcc) knows where to find them.

The header file "version.h" contain the version numbers and the macro (KERNEL_VERSION) to convert them:
```
$ cat /usr/include/linux/version.h
#define LINUX_VERSION_CODE 393984
#define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + ((c) > 255 ? 255 : (c)))
#define LINUX_VERSION_MAJOR 6
#define LINUX_VERSION_PATCHLEVEL 3
#define LINUX_VERSION_SUBLEVEL 0
```

Please notice that there are 2 different headers, e.g. on my system:
```
linux-api-headers 6.3-1
linux-headers 6.4.6.arch1-1
```
The difference is explained e.g. here:
https://bbs.archlinux.org/viewtopic.php?id=258173 

It is imperative that they fit together:
`linux-api-headers <= linux-headers`

I don't know how other distributions handle this, but it might be similar to Arch Linux (we are all on Linux).
If you cross compile, the target tool chain should use the matching header files, too. 

Please notice:
Even if you install the correct headers, hcxdumptool will not work as expected, because the driver doesn't support packet injection.
But it is always good to fix your environment/configuration if you would like to compile other tools than supplied by the package management system of your distribution.

ZerBea, 2023-07-27:
By latest commit I added an information about glibc (at compile time):
```
$ hcxdumptool -v
hcxdumptool 6.3.1-42-g3c00305 (C) 2023 ZeroBeat
compiled by gcc 13.1.1
compiled with Linux API headers 6.3.0
compiled with glibc 2.37
```

The version of glibc running on the target system should be greater or equal to this version, but never less:
```
$ ldd --version
ldd (GNU libc) 2.37
```