ZerBea/hcxdumptool #333: exit on success option please

AnotherWayIn, 2023-07-24:
great tool, thanks.

it would be nice however to have an exit upon hash option pls.
Will save having to keep running a check on the pcap then killing the process

thanks

ZerBea, 2023-07-24:
Nice idea. let me think about it how to implement it.
Maybe something like this:
`--oneapol=<TARGET_MAC>      : exit if a PMKID or a handshake has been received`

ZerBea, 2023-07-24:
Please test latest commit:
```
--exitoneapol=<type>           : exit on first EAPOL occurrence:
                                  bitmask:
                                   1 = PMKID
                                   2 = EAPOL M2
                                   4 = EAPOL M3
                                  target BPF filter is recommended
```

set monitor mode:
```
$ sudo hcxdumptool -m wlp22s0f0u9u3

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 c83a35dc9ef0 b025aae7f0ba + wlp22s0f0u9u3    rt2800usb (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] disabled

monitor mode is active...

bye-bye
```

create target BPF:
```
$ tcpdump -i wlp22s0f0u9u3 wlan addr3 08:96:d7:98:e1:9e -ddd > attack0896d798e19e.bpf
```

run hcxdumptool:
```
$ sudo hcxdumptool -c 10a --exitoneapol=4 --attemptclientmax==0 --bpf=attack0896d798e19e.bpf

  CHA    LAST   R 1 3 P S    MAC-AP    ESSID (last seen on top)   SCAN-FREQUENCY:   2457
-----------------------------------------------------------------------------------------
 [ 10] 12:06:05         + 0896d798e19e AP_7272

   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last seen on top)
-----------------------------------------------------------------------------------------

exit on EAPOL M1M2M3
$
```

Please notice:
Using option --attemptclientmax==0 will prevent that the CLIENT connect to hcxdumptool instead of connecting to its AP.

closed 2023-07-24 by ZerBea

AnotherWayIn, 2023-07-24:
awesome, thanks for the quick turnaround. how do i exit on any type hash? i'm only ever targeting 1 ssid at a time (multiple bssids) so i want the exit as soon as pmkid or eapol hash is captured. can i do --exitoneapol=1 --exitoneapol=2 --exitoneapol=4?

ZerBea, 2023-07-24:
We use a bitmask. Using 
"--exitoneapol=7"
hcxdumptool will exit on a PMKID or on AN EAPOL M1M2 or on an EAPOL M1M2M3

matrix:
1 = exit on PMKID
2 = exit on EAPOL M1M2
4 = exit on EAPOL M1M2M3

3 = exit on PMKID or EAPOL M1M2
5 = exit on PMKID or EAPOL M1M2M3

6 = exit on EAPOL M1M2 or EAPOL M1M2M3 - this doesn't make sence, because EAPOL M1M2M3 always include M1M2

ZerBea, 2023-07-24:
Please notice that some of the possible bitmask combinations do not make sense, e.g.:
exitoneapol=6 does not make sense, because you never get an EAPOL M1M2M3 because hcxdumptool terminate after and EAPOL M1M2 has been received.
  
If an AP transmit a PMKID you will never get an EAPOL M1M2 on option --exitoneapol=3, because hcxdumptool terminate after the PMKID has been received.

AnotherWayIn, 2023-07-24:
yes sorry, i realised after i posted how bit masking works. thanks very much for the quick delivery of this. i'll test it out in the morning

ZerBea, 2023-07-24:
This new option in combination with a BPF is a nice improvement.

Sechorda, 2025-02-25:
@ZerBea Hello,

I am trying to use --exitoneapol=4 to exit when a client probing for an SSID captures the M1M2 (+ in the '2' column).

when I get a + in the '2' column it does not exit, am I missing something?

ZerBea, 2025-02-25:
@Sechorda , you haven't missed a thing. Thanks for reporting this issue:

Should be fixed by latest commit on hcxlabtool and hcxdumptool by expanding the option;
```
--exitoneapol=<type>      : exit on first EAPOL occurrence:
                             bitmask:
                               1 = PMKID (from AP)
                               2 = EAPOL M2M3 (authorized)
                               4 = EAPOL M1M2 (not authorized)
                               8 = EAPOL M1M2ROGUE (not authorized)
                              16 = EAPOL M1
                             target BPF filter is recommended
```

Now it should look like this:
```
$ sudo hcxdumptool -c 10a --exitoneapol=8
...
246 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
7 EPB written to pcapng dumpfile

exit on EAPOL M1M2ROGUE
```

Sechorda, 2025-02-25:
@ZerBea That was really fast - Thanks!

For my pipeline, I instead just parse for the `+` symbol in the output while using a BPF. Works fine for me! Glad we squashed this bug though :)

Sechorda, 2025-02-27:
@ZerBea I love the project. Thanks for your work.

hey, just one thought - Would it be advantageous to modify the MACaddr of the 'simulated AP' for probing clients? I know some clients are picky - any specific reason to randomize instead of grab the client's associated AP's-MAC addr?

cheers

ZerBea, 2025-02-27:
@Sechorda , this is already implemented:

Depending on the CLIENTs PROBEREQUEST
undirected:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L2361
hcxdumptool/hcxlabtool either generate a randomized MAC AP
https://github.com/ZerBea/hcxdumptool/blob/1f4b438e7c80d0649ea33391de5fddd67fdfbc76/hcxdumptool.c#L2387

or it uses the real one:
directed:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L1367
hcxdumptool/hcxlabtool us the MAC the CLIENT probes
https://github.com/ZerBea/hcxdumptool/blob/1f4b438e7c80d0649ea33391de5fddd67fdfbc76/hcxdumptool.c#L2347

Additional, as soon as hcxdumptool/hcxlabtool receives a MAC from a real AP, it replaces the randomized MAC stored in the internal ring buffer by the real one.

ZerBea, 2025-02-27:
@Sechorda , you can monitor this by watching the CLIENTs status display:
Sometimes you see an EAPOLM1M2ROGUE using the randomized MAC AP and sometimes you see an EAPOLM1M2 using the real MAC AP.

It is important to understand that this behavior is controlled by the BPF!
A strict BPF, filtering by MAC AP only, prevent the usage of randomized MACs completely!

Sechorda, 2025-02-27:
@ZerBea I got a working BPF, and also saw your recent comments briefly explaining basic ones, but essentially I'm using the client-mac in `addr1 addr2 addr3` - I spent some time doing one by one and couldn't find out which one was strictly for client-macs, not sure if my interface was bugging up and not capturing at all, or if my filter was wrong - but the catchall filter works for me.

Thanks for this explanation - after reading the docs and kicking the tires I think i'v got most of learning curve down.

ZerBea, 2025-02-27:
BTW:
This project is not a "stand alone project". It would not be possible to code the tools without the big GPU hash crackers "John the Ripper" and "hashcat" as well as the huge database of "wpa-sec".

Sechorda, 2025-02-27:
@ZerBea I found a defcon talk outlining the wpa-sec database. thanks for the tip