ZerBea/hcxdumptool #334: No output when scanning with AWUS036AXML

parttimer777, 2023-08-02:
Hello,

I'm getting this error when running hcxdumptool scanning with the new AWUS036AXML usb adapter. Firmware was updated as per https://store.rokland.com/pages/alfa-awus036axml-driver-support-page. The hcxdumptool returns no output, and this error scrolls constantly in dmesg/messages.

Wondering if anyone has any advice.  I'm open to returning this adapter and getting AWUS036ACHM instead if that's the general advice.

Here's the error:
`
$ dmesg
[  143.833329] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
[  143.833865] xhci_hcd 0000:00:0c.0: Looking for event-dma 00000000010a0070 trb-start 00000000010a0080 trb-end 00000000010a00b0 seg-start 00000000010a0000 seg-end 00000000010a0ff0
`

    # lsmod|grep mt
    mt7921u                20480  0
    mt7921_common          94208  1 mt7921u
    mt76_connac_lib        73728  2 mt7921_common,mt7921u
    mt76_usb               36864  1 mt7921u
    mt76                  106496  4 mt76_usb,mt7921_common,mt76_connac_lib,mt7921u

    $ uname -a
    Linux debian 6.1.0-10-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.38-2 (2023-07-27) x86_64 GNU/Linux
    
    $ lsb_release  -a
    No LSB modules are available.
    Distributor ID: Debian
    Description:    Debian GNU/Linux 12 (bookworm)
    Release:        12
    Codename:       bookworm
    
    
    
    # https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md
    /lib/firmware/mediatek$ ls -lart
    -rw-r--r--  1 root root   92192 Aug  1 23:53 WIFI_MT7961_patch_mcu_1_2_hdr.bin
    -rw-r--r--  1 root root  722420 Aug  1 23:54 WIFI_RAM_CODE_MT7961_1.bin
    
    
    
    $ iwconfig
    wlx00c0cab3c769  IEEE 802.11  ESSID:off/any
              Mode:Managed  Access Point: Not-Associated   Tx-Power=3 dBm
              Retry short limit:7   RTS thr:off   Fragment thr:off
              Encryption key:off
              Power Management:on
		      
    # lsusb
    Bus 002 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device
    Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
    Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
    Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    
    # hcxdumptool -v
    hcxdumptool 6.3.1-42-g3c00305 (C) 2023 ZeroBeat
    compiled by gcc 10.2.1
    Linux API headers (LINUX_VERSION_MAJOR) is not defined
    compiled with glibc 2.31
    
    # hcxdumptool -L
    
    Requesting physical interface capabilities. This may take some time.
    Please be patient...
    
    
    available wlan devices:
    
    phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
    ---------------------------------------------------------------------------------------------
      0   3 00c0cab3c769 00c0cab3c769 * wlx00c0cab3c769  mt7921u (NETLINK)
    
    * active monitor mode available
    + monitor mode available
    - no monitor mode available
    
    bye-bye
    
    
    # hcxdumptool -I wlx00c0cab3c769
    
    Requesting physical interface capabilities. This may take some time.
    Please be patient...
    
    
    interface information:
    
    phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
    ---------------------------------------------------------------------------------------------
      0   3 00c0cab3c769 00c0cab3c769 * wlx00c0cab3c769  mt7921u (NETLINK)
    
    
    available frequencies: frequency [channel] tx-power of Regulatory Domain: 00
    
      2412 [  1] 20.0 dBm     2417 [  2] 20.0 dBm     2422 [  3] 20.0 dBm     2427 [  4] 20.0 dBm
      2432 [  5] 20.0 dBm     2437 [  6] 20.0 dBm     2442 [  7] 20.0 dBm     2447 [  8] 20.0 dBm
      2452 [  9] 20.0 dBm     2457 [ 10] 20.0 dBm     2462 [ 11] 20.0 dBm     2467 [ 12] 20.0 dBm
      2472 [ 13] 20.0 dBm     2484 [ 14] 20.0 dBm     5180 [ 36] 20.0 dBm     5200 [ 40] 20.0 dBm
      5220 [ 44] 20.0 dBm     5240 [ 48] 20.0 dBm     5260 [ 52] 20.0 dBm     5280 [ 56] 20.0 dBm
      5300 [ 60] 20.0 dBm     5320 [ 64] 20.0 dBm     5500 [100] 20.0 dBm     5520 [104] 20.0 dBm
      5540 [108] 20.0 dBm     5560 [112] 20.0 dBm     5580 [116] 20.0 dBm     5600 [120] 20.0 dBm
      5620 [124] 20.0 dBm     5640 [128] 20.0 dBm     5660 [132] 20.0 dBm     5680 [136] 20.0 dBm
      5700 [140] 20.0 dBm     5720 [144] 20.0 dBm     5745 [149] 20.0 dBm     5765 [153] 20.0 dBm
      5785 [157] 20.0 dBm     5805 [161] 20.0 dBm     5825 [165] 20.0 dBm     5845 [169] disabled
      5865 [173] disabled     5955 [  1] disabled     5975 [  5] disabled     5995 [  9] disabled
      6015 [ 13] disabled     6035 [ 17] disabled     6055 [ 21] disabled     6075 [ 25] disabled
      6095 [ 29] disabled     6115 [ 33] disabled     6135 [ 37] disabled     6155 [ 41] disabled
      6175 [ 45] disabled     6195 [ 49] disabled     6215 [ 53] disabled     6235 [ 57] disabled
      6255 [ 61] disabled     6275 [ 65] disabled     6295 [ 69] disabled     6315 [ 73] disabled
      6335 [ 77] disabled     6355 [ 81] disabled     6375 [ 85] disabled     6395 [ 89] disabled
      6415 [ 93] disabled     6435 [ 97] disabled     6455 [101] disabled     6475 [105] disabled
      6495 [109] disabled     6515 [113] disabled     6535 [117] disabled     6555 [121] disabled
      6575 [125] disabled     6595 [129] disabled     6615 [133] disabled     6635 [137] disabled
      6655 [141] disabled     6675 [145] disabled     6695 [149] disabled     6715 [153] disabled
      6735 [157] disabled     6755 [161] disabled     6775 [165] disabled     6795 [169] disabled
      6815 [173] disabled     6835 [177] disabled     6855 [181] disabled     6875 [185] disabled
      6895 [189] disabled     6915 [193] disabled     6935 [197] disabled     6955 [201] disabled
      6975 [205] disabled     6995 [209] disabled     7015 [213] disabled     7035 [217] disabled
      7055 [221] disabled     7075 [225] disabled     7095 [229] disabled     7115 [233] disabled
    
    bye-bye

ZerBea, 2023-08-02:
The ALFA seams to be ok and the driver seams to be ok, but you're running into a kernel regression regarding the XHCI subsystem as reported here:
https://bugzilla.kernel.org/show_bug.cgi?id=217122

There is nothing I can do and you have to wait for a kernel fix.

ZerBea, 2023-08-02:
Looks like there are several XHCI related reports:
https://duckduckgo.com/?q=ERROR+Transfer+event+TRB+DMA+ptr+not+part+of+current+TD&t=h_&ia=web

I think we can close this report because it is related to XHCI.

parttimer777, 2023-08-02:
ZerBea, thanks for the info. I updated the kernel to latest debian unstable:

Linux debian 6.4.0-1-amd64

Still no go. If you have any advice for things i can try it would be appreciated.

ZerBea, 2023-08-02:
A workaround for me is to connect the WiFi device (in may case an ALFA AWUS036ACM) to an USB 2 port.

$ uname -r
6.4.7-arch1-2

$ hcxdumptool -v
hcxdumptool 6.3.1-42-g3c00305 (C) 2023 ZeroBeat
compiled by gcc 13.2.1
compiled with Linux API headers 6.4.0
compiled with glibc 2.37

ZerBea, 2023-08-02:
After some research (kernel 6.4.7):

USB 3 WiFi devices are not working as expected if connected to an USB 3 port
USB 3 WiFi devices are working as expected if connected to an USB 2 port
and
USB 2 WiFi devices are working as expected if connected to an USB 3 port
USB 2 WiFi devices are working as expected if connected to an USB 2 port

BTW:

On kernel 6.4.7 there is no dmesg warning like this "ERROR Transfer event TRB DMA ptr not part of current TD".
That's all if running an ALFA AWUS036ACM:
```
$ lsusb
Bus 002 Device 002: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter

$ sudo dmesg
[  829.710132] usbcore: registered new interface driver mt76x2u
[  829.723549] mt76x2u 2-3:1.0 wlp22s0f0u3: renamed from wlan0
[  850.148134] mt76x2u 2-3:1.0 wlp22s0f0u3: entered promiscuous mode
[  860.063034] mt76x2u 2-3:1.0 wlp22s0f0u3: left promiscuous mode
```

and hcxdumptool got nothing from the driver:
```
 CHA    LAST   R 1 3 P S    MAC-AP    ESSID (last seen on top)   SCAN-FREQUENCY:   2462
-----------------------------------------------------------------------------------------

   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last seen on top)
-----------------------------------------------------------------------------------------
^C
exit on sigterm
```

Looks like the XHCI bug is still not fixed

ZerBea, 2023-08-02:
To confirm that the xhci system is not working as expected, you can run either tshark or Wireshark in capture mode in parallel:
terminal 1:
`$ sudo hcxdumptool  -i wlp22s0f0u3 -c 11a --disable_beacon`

terminal 2:
```
$ tshark -i wlp22s0f0u3
Capturing on 'wlp22s0f0u3'
 ** (tshark:5365) 17:33:07.722885 [Main MESSAGE] -- Capture started.
 ** (tshark:5365) 17:33:07.722952 [Main MESSAGE] -- File: "/tmp/wireshark_wlp22s0f0u3OLEP81.pcapng"
...
```
tshark will receive nothing, too.

parttimer777, 2023-08-02:
Oh man thanks for that info! 

Verified: it's not receiving anything either with "--disable_beacon". Though I thought I observed it working in kernel 5.10 or so.

On VirtualBox i disabled USB3 (XHCI) but when I plugged in this USB3 adapter, it issued an error "no available usb ports".

When i plugged it in via an old USB 2.0 extension cable, the system recognized the device (lsusb) but didn't recognize it as a wifi adapter (no output in iwconfig). Also, that produced errors in dmesg (unrecognized vendor id..).

Yeah, I feel this wraps it up for me, i dont have USB2.0 ports on this SurfacePro so will have to wait for proper XHCI support or swap out for the AWUS036ACHM.

ZerBea, 2023-08-02:
We got a response on bugzilla:
https://bugzilla.kernel.org/show_bug.cgi?id=217122
Looks like a patch is available.

parttimer777, 2023-08-07:
Hey ZerBea,

I went ahead and applied that patch. I haven't compiled a kernel in god knows how long but felt I needed to give this a good try.

Unfortunately, still not getting a positive feedback, seeing same error ` [   71.155209] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13`

Double checked that XHCI is enabled in virtualbox, and kernel config has TPM enabled.

Running this command `sudo hcxdumptool -i $IFACE -w dumpfile.pcapng --rds=1` returns no APs. Figure i'd post an update, just in case it's useful. I'll be purchasing the AWUS036ACHM for the time being.
    
    :~/work/orig/linux-6.5~rc4$ patch -p1 < xhci.patch 
    patching file drivers/char/tpm/tpm-chip.c
    patching file drivers/char/tpm/tpm_crb.c
    patching file include/linux/tpm.h
    
    
     68.592550] usb 1-2: new high-speed USB device number 3 using xhci_hcd
    [   69.002509] usb 1-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
    [   69.002514] usb 1-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
    [   69.002516] usb 1-2: Product: Wireless_Device
    [   69.002530] usb 1-2: Manufacturer: MediaTek Inc.
    [   69.002531] usb 1-2: SerialNumber: 000000000
    [   69.095801] alg: No test for fips(ansi_cprng) (fips_ansi_cprng)
    [   69.532023] cfg80211: Loading compiled-in X.509 certificates for regulatory database
    [   69.532126] Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
    [   69.532211] Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
    [   69.532364] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
    [   69.537975] platform regulatory.0: firmware: direct-loading firmware regulatory.db
    [   69.540529] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
    [   70.108165] mt7921u 1-2:1.3: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
    [   70.360126] Bluetooth: Core ver 2.22
    [   70.360184] NET: Registered PF_BLUETOOTH protocol family
    [   70.360185] Bluetooth: HCI device and connection manager initialized
    [   70.360189] Bluetooth: HCI socket layer initialized
    [   70.360191] Bluetooth: L2CAP socket layer initialized
    [   70.360194] Bluetooth: SCO socket layer initialized
    [   70.478629] usb 1-2: reset high-speed USB device number 3 using xhci_hcd
    [   71.134189] usbcore: registered new interface driver mt7921u
    [   71.135337] usbcore: registered new interface driver btusb
    [   71.155190] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
    [   71.155207] xhci_hcd 0000:00:0c.0: Looking for event-dma 00000000011da020 trb-start 00000000011da040 trb-end 00000000011da070 seg-start 00000000011da000 seg-end 00000000011daff0
    [   71.155209] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
    [   71.155219] xhci_hcd 0000:00:0c.0: Looking for event-dma 00000000011da030 trb-start 00000000011da040 trb-end 00000000011da070 seg-start 00000000011da000 seg-end 00000000011daff0
    [   71.158885] bluetooth hci0: firmware: direct-loading firmware mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
    [   71.161690] mt7921u 1-2:1.3: firmware: direct-loading firmware mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
    [   71.161928] mt7921u 1-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
    
    
    # iwconfig
    wlx00c0cab3c769  IEEE 802.11  ESSID:off/any
              Mode:Managed  Access Point: Not-Associated   Tx-Power=3 dBm
              Retry short limit:7   RTS thr:off   Fragment thr:off
              Encryption key:off
              Power Management:on
    
    
    $ hcxdumptool -L
    
    Requesting physical interface capabilities. This may take some time.
    Please be patient...
    
    
    available wlan devices:
    
    phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
    ---------------------------------------------------------------------------------------------
      0   3 00c0cab3c769 00c0cab3c769 * wlx00c0cab3c769  mt7921u (NETLINK)
    
    * active monitor mode available
    + monitor mode available
    - no monitor mode available

parttimer777, 2023-08-07:
Also, this is on my desktop which has usb2.0 ports. I disabled XHCI (usb3.0) in virtualbox and tried to run hcxdumptool still no output.

Got an error on the console:

[image]

ZerBea, 2023-08-07:
Thanks for this information.
Can you please join bugzilla.kernel.org and report this directly to:
https://bugzilla.kernel.org/show_bug.cgi?id=217122
because it is related to the XHCI sub system.

ZerBea, 2023-08-07:
BTW:
I joined the conversation on bugzilla.kernel.org, too, because tshark and Wireshark failed to receive packets, too if doing the standard procedure:
iw set monitor mode
ip link set device up
iw set channel
tsahrk capture traffic
https://bugzilla.kernel.org/show_bug.cgi?id=217122#c32

I don't think it is related to ALFA devices or the mt76 driver, because other USB 3.0 devices rt2800usb driver) failed, too.
Also I don't think that it is related to hcxdumptool, because tshark/Wireshark failed, too.

parttimer777, 2023-08-07:
Still no go, I'll post the results in the bugzilla. Dusted off my old sony vaio, it had a couple of usb3.0 ports.

ZerBea, 2023-08-08:
Great, thanks.
I mentioned it here, too:
https://github.com/morrownr/USB-WiFi/discussions/260#discussioncomment-6617868

I don't think the problem is related to the driver.

Tested using an USB 2 device on the same port:
```
0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]

[  688.556201] usb 1-4: new high-speed USB device number 8 using xhci_hcd
[  688.803627] usb 1-4: New USB device found, idVendor=0b05, idProduct=17d1, bcdDevice= 1.00
[  688.803637] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  688.803639] usb 1-4: Product: WiFi
[  688.803641] usb 1-4: Manufacturer: MediaTek
[  688.803643] usb 1-4: SerialNumber: 1.0
[  689.059681] usb 1-4: reset high-speed USB device number 8 using xhci_hcd
[  689.309168] mt76x0u 1-4:1.0: ASIC revision: 76100002 MAC revision: 76502000
[  692.889421] mt76x0u 1-4:1.0: EEPROM ver:02 fae:01
[  693.145486] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[  693.146044] usbcore: registered new interface driver mt76x0u
[  693.152083] mt76x0u 1-4:1.0 wlp22s0f0u4: renamed from wlan0
[  702.833854] mt76x0u 1-4:1.0 wlp22s0f0u4: entered promiscuous mode

$ tshark -i wlp22s0f0u4
Capturing on 'wlp22s0f0u4'
 ** (tshark:3617) 08:31:46.306978 [Main MESSAGE] -- Capture started.
 ** (tshark:3617) 08:31:46.307034 [Main MESSAGE] -- File: "/tmp/wireshark_wlp22s0f0u4Z6AH91.pcapng"
^Xtshark: 
48 packets captured

[  709.590017] mt76x0u 1-4:1.0 wlp22s0f0u4: left promiscuous mode
```

while all USB 3 devices failed.

ZerBea, 2023-08-08:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/7faffcd5da0f554a71a5a6b75a96335dfefdc79d
I added an additional information how many packets are received (like tshark) and stored:
```
SHB blocks written to pcapng dumpfile: 1
IDB blocks written to pcapng dumpfile: 1
ECB blocks written to pcapng dumpfile: 1
EPB blocks written to pcapng dumpfile: 5

exit on sigterm

bye-bye
```

ZerBea, 2023-08-11:
Running the same configuration (kernel, driver, ALFA WiFi device) on my Raspberry Pi, everything is working as expected.

parttimer777, 2023-08-11:
RPI is kernel 5.x? i recall it working on a debian i had with kernel 5.x with an older hcxdumptool and --disable_active_monitor flag

By any chance, anything you think i should try?

ZerBea, 2023-08-11:
```
Linux hcxpi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Aug  3 14:26:49 2023

$ hcxdumptool -v
hcxdumptool 6.3.1-42-g3c00305 (C) 2023 ZeroBeat
compiled by gcc 12.2.0
compiled with Linux API headers 6.1.21
compiled with glibc 2.36
```
is working fine.

Information is here:
https://www.raspberrypi.com/software/operating-systems/

32 bit:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-lite.img.xz
or 64 bit
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-05-03/2023-05-03-raspios-bullseye-arm64-lite.img.xz

parttimer777, 2023-08-12:
I just pulled the latest hcxdumptool, tried it on my pi3 and it says it was able to capture packets but not showing any output.

I'm using AWUS036AXML with linux firmware updated as per https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md .

[image]

$ uname -a
Linux pi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

________

Also tried it again on a box with the previously patched debian kernel 6.5:

[   14.914156] mt7921u 3-4:1.3: firmware: direct-loading firmware mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
[   14.914736] mt7921u 3-4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
[   14.927496] bluetooth hci1: firmware: direct-loading firmware mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
[   15.175838] mt7921u 3-4:1.3: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[   15.183894] mt7921u 3-4:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958

$ hcxdumptool -v
hcxdumptool 6.3.1-48-g26c41f0 (C) 2023 ZeroBeat
compiled by gcc 13.2.0
compiled with Linux API headers 6.4.4
compiled with glibc 2.37

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 00c0cab3c769 00c0cab3c769 * wlx00c0cab3c769  mt7921u (NETLINK)

  sudo systemctl stop wpa_supplicant
  sudo systemctl stop NetworkManager
 IFACE=wlx00c0cab3c769
  sudo hcxdumptool -i $IFACE -w dumpfile.pcapng --rds=1

[image]

_______

Not really sure what to try next, but open for any advice

parttimer777, 2023-08-12:
Just checked the WN722n wifi card on the patched kernel 6.5 and that works well:

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1   4 98ded00948e5 98ded00948e5 + wlx98ded00948e5  ath9k_htc (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

[image]

ZerBea, 2023-08-12:
Thanks for that information. The screen shots confirm that hcxdumptool is working as expected on TL722WN (USB 2) but not on AWUS036AXML (USB 3.2 Gen 1).
It still looks like an XHCI/USB 3.x problem that has to to be fixed by kernel developers.

parttimer777, 2023-08-12:
No prob, thank you!