ZerBea/hcxdumptool #337: Issue with finding any networks in Active Monitor mode

eng1n88r, 2023-08-23:
Tried to use hcxdumptool on clean installation of Kali and facing some weird behavior.

Tool version:
[image]

Adapter is AWUS036AXM (chipset is MediaTek MT7921AUN).
[image]

[image]

[image]

`dmesg` after plugin in the adapter:
[image]

Switching to monitor mode with `-m` (Reg Domain was set to US):
[image]

Trying to run `airodump-ng wlan0` to confirm that monitor mode works:
[image]

Next trying to run `hcxdumptool -i wlan0 -w test_dump.pcapng --rds=1 -F` and it doesn't finding anything: 
[image]

Terminate the process and try to run `airodump-ng wlan0` again (nothing is found too):
[image]

At this point monitoring just stops working completely using other tools. The only way to make it work is to re-plug the adapter.

After reconnecting the adapter if I set monitor mode with `-m` and then run it with `-p` option i.e. `hcxdumptool -i wlan0 -w test_dump.pcapng --rds=1 -F -p` it works:
[image]

From what I understand `-p` is passive scanning, right?
```
-p             : do not set monitor mode: active (do not ACK incoming frames addressed to the device MAC)
                 default monitor mode: active (ACK all incoming frames addressed to the device MAC)
```
I'm not sure where the disconnect is. Any ideas?

ZerBea, 2023-08-23:
This is a kernel issue.Please read more here:
https://github.com/ZerBea/hcxdumptool/issues/334

And it is reported to bugzilla, too:
https://bugzilla.kernel.org/show_bug.cgi?id=217122

I think we can close it here, because we need a kernel fix.

closed 2023-08-23 by ZerBea

ZerBea, 2023-09-14:
I got an ALLNET WA1200AC:
https://www.reichelt.de/de/en/wlan-adapter-usb-1167-mbit-s-allnet-wa1200ac-p298295.html?GROUPID=5839&START=0&OFFSET=16&SID=9625c0575e3885f3439033b181fcfcbb2c6f5c5b20fc74c8989fa&LANGUAGE=EN&&r=1

ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter

First tests are not impressive if connected to an USB 3x controller
`16:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] 500 Series Chipset USB 3.1 XHCI Controller
`

running
```
$ uname -r
6.5.2-arch1-1
```
dmesg showing no errors:
```
$ sudo dmesg
[107179.838508] usb 2-4: new SuperSpeed USB device number 19 using xhci_hcd
[107179.855584] usb 2-4: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[107179.855588] usb 2-4: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[107179.855590] usb 2-4: Product: Wireless 
[107179.855591] usb 2-4: Manufacturer: MediaTek Inc.
[107179.855593] usb 2-4: SerialNumber: 000000000
[107180.028133] usb 2-4: reset SuperSpeed USB device number 19 using xhci_hcd
[107180.049919] mt76x2u 2-4:1.0: ASIC revision: 76120044
[107180.071242] mt76x2u 2-4:1.0: ROM patch build: 20141115060606a
[107180.231993] mt76x2u 2-4:1.0: Firmware Version: 0.0.00
[107180.231999] mt76x2u 2-4:1.0: Build: 1
[107180.232000] mt76x2u 2-4:1.0: Build Time: 201507311614____
[107180.949901] ieee80211 phy21: Selected rate control algorithm 'minstrel_ht'
[107180.960286] mt76x2u 2-4:1.0 wlp22s0f0u4: renamed from wlan0
```

No packets received!
```
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile

exit on sigterm
bye-bye
```

If connected to an USB 2.0 interface on the same machine:
```
146 packet(s) captured
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
20 EPB written to pcapng dumpfile

exit on sigterm
bye-bye
```

Looks like USB 3.x support is completely broken on mt76 USB adapters.