ZerBea/hcxdumptool #338: AP List Table Explaination

marbletissue, 2023-08-24:
Hello,
When running hcxdumptool it outputs a table which displays the APs its scanning.
Is there a guide that explains what the headers of the table mean?
For example, R 1 3 P S which are marked with a + - what does this mean?
Also, the lower table has E 2, what does that mean?
If I am targeting an AP, should I ensure that all + are marked for a successful crack of the hash?
Many thanks

ZerBea, 2023-08-24:
The status is explained in legend of  --help:
Legend
real time display:
```
 R = + AP display:     AP is in TX range or under attack
 S = + AP display:     AUTHENTICATION KEY MANAGEMENT PSK
 P = + AP display:     got PMKID hashcat / JtR can work on
 1 = + AP display:     got EAPOL M1 (CHALLENGE)
 3 = + AP display:     got EAPOL M1M2M3 (AUTHORIZATION) hashcat / JtR can work on
 E = + CLIENT display: got EAP-START MESSAGE
 2 = + CLIENT display: got EAPOL M1M2 (ROGUE CHALLENGE) hashcat / JtR can work on
```

If you target an AP, a successful ATTACK will show a + in column P and/or 3 of AP display.

marbletissue, 2023-08-24:
Thanks for the quick reply. I must have read the --help a dozen times and missed that lol..
I have been capturing for 2 days now, but still missing the PMKID on my target.
I will look at what I can do to make this happen - if anything?
I have successfully cracked the hash of other APs without the PMKID but I assume this makes the process quicker/more efficient for hashcat?

Thanks for adding the section on the workflow. I'm working to: 
hcxdumptool - capture everything
hcxpcapngtool - convert everything (--all)
hcxhashtool - filter my target (--essid=xxxx)
At that point, I can attempt the crack the hash using the wordlist from hcxpcapngtool, or bruteforce it.
Do you have a post anywhere of how you would target an AP?
Thanks :)

ZerBea, 2023-08-24:
Running hashcat on a PMKID is a little bit faster than running hashcat on an EAPOL MESSAGEPAIR with --nonce-error-corrections=0. Using higher NC values will increase the task time.
Filtering the hash by ESSID is a good idea to get full benefit of reuse of PBKDF2. Slow PBKDF2 is doing once and than the PMK is compared to all PMKIDs /MESSAGEPAIRs using the same ESSID.

Targeting an AP via Berkeley Packet Filter is explained here:
https://github.com/ZerBea/hcxdumptool/issues/301
starting with comment:
https://github.com/ZerBea/hcxdumptool/issues/301#issuecomment-1594653855

ZerBea, 2023-08-24:
Closed this report, because it is rather a discussion
https://github.com/ZerBea/hcxdumptool/discussions
and not a bug.