ZerBea/hcxdumptool #340: Short, greppable outputs

ApplePie420, 2023-09-02:
It would be nice to have outputs, that are easily greppable and short, for example `hcxdumptool -L` returns stuff line-by-line, clearly meant for it to be "interactive" and pretty to look at. But that output is basically not greppable, unless you save the output to a file and then you can work with that - and even after that, it is a huge pain to do anything with, since the output is "nice" and readable. Having an additional flag that would toggle "greppable" output, i.e. for interfaces list, it would just return "MAC;IFNAME;DRIVER" would be extremely nice. 
Since there aren't many other flags that would output anything that one might use in a shell script, this is the only ocassion where it might (and will) be useful.

ZerBea, 2023-09-02:
Done latest commit:
```
$ hcxdumptool -h
-L             : show INTERFACE list
-l              : show INTERFACE list (tabulator separated and greppable)
```

```
$ hcxdumptool -l
  0	  3	f81a670f291b	f81a670f291b	+	wlp22s0f0u4     	rtl8192cu	NETLINK
  1	  4	0018e748c9d8	0018e748c9d8	+	wlp22s0f0u3     	rtl8187	NETLINK
  2	  5	00e0614861dd	00e0614861dd	*	wlp22s0f0u9u3   	mt7601u	NETLINK
  3	  6	00e62d05131a	00e62d05131a	*	wlp22s0f0u9u4   	mt7601u	NETLINK
```
get field via awk:
```
$ hcxdumptool -l | awk 'BEGIN { FS = "\t" } ; { print "Interfacename: " $4 }'
Interfacename: f81a670f291b
Interfacename: 0018e748c9d8
Interfacename: 00e0614861dd
Interfacename: 00e62d05131a
```

ApplePie420, 2023-09-02:
I love the commit message, _shart_ output :D :D 
Thank you very very much :)

closed 2023-09-02 by ApplePie420