ZerBea/hcxdumptool #342: can't accept options

Runnerblade84, 2023-09-12:
Hi,

it seems that 6.3.1 version can't accept parameters

hcxdumptool -i wlan0mon --enable_status=1 --filterlist_ap=/tmp/target.txt --filtermode=2 -o /tmp/pmkid.pcapng
hcxdumptool: unrecognized option '--enable_status=1'
hcxdumptool 6.3.1 (C) 2023 by ZeroBeat
usage: hcxdumptool -h for help

Same if omit enable_status

Runnerblade84, 2023-09-12:
sorry, it is just not supported anymore

closed 2023-09-12 by Runnerblade84

ZerBea, 2023-09-12:
That is correct, wanted and explained in help and changelog starting here:
```
06.04.2023
==========
release v6.2.9
several bug fixes

This is the last version:
that use WIRELESS EXTENSIONS
that use ETHTOOL to get/set virtual MAC address
that use old style status output
that use soft coded filter lists
that use msec timestamp
that use crypto stuff
that use server/client mode to display status

Next version will go back to the roots:
set focus on WPA PSK (WPA1, WPA2, WPA2 key version 3)
set bandwidth to 20MHz to increase range
set bitrate to lowest values to increase range
use active monitor mode
use NL80211 stack
use RTNETLINK
band a, b, c, d, e support
use NMEA messages:
 $GPRMC: Position, velocity, time and date
 $GPGGA: Position, orthometric height, fix related data, time
 $GPWPL: Position and MAC AP
 $GPTXT: ESSID in HEX ASCII
remove options that slow hcxdumptool down
```

To filter (what every you want) it is now mandatory to set a Berkeley Packet Filter (BPF). Old style (slow) filter modes are removed completely. The status display is completely rewritten, too ( see --help for more information about the columns).

More information is here:
https://github.com/ZerBea/hcxdumptool/issues/301

BTW:
I don't recommend to run hcxdumptool on a virtual monitor interface (wlan0mon), because it has a build in monitor mode that get full benefit of "active monitor mode" if the driver support this.

To discover if active monitor mode is supported by driver run either
```
$ hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  3   6 74da38f2038e 74da38f2038e * wlp22s0f0u4      mt7601u (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available
```

or
```
$ hcxdumptool -I wlp22s0f0u4

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  3   6 74da38f2038e 74da38f2038e * wlp22s0f0u4      mt7601u (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm
```

Do not set monitor mode by third party tools like airmon-ng, iw or iwconfig!