ZerBea/hcxdumptool #344: Capturing PMKID not working anymore ?

spookycheekybreeky, 2023-09-23:
Hey, I been trying to captures pmkid's for the past 2 days but without any success. 

I have routers with wps and WPA/2 - AES enabled and weak password for testing but I'm not able to capture anything. I Tries with multiple version of hxcdumptools but without any success. 

Do you have any route on why i can't capture anything?

My setup is a Orangepi 3 LTS with armbian and a ALPHA Network AWUS036ACH. I have attached some screenshot.
[image]
[image]

ZerBea, 2023-09-23:
hcxdumptool 4.2 is outdated because it depends on WIRELESS EXTENSIONS.
AWUS036ACH is based on RTL8812AU. This driver highly depend on NETLINK.

Please update to latest git head or at least v 6.3.1 (NETLINK version)
More information is here:
https://github.com/ZerBea/hcxdumptool/discussions/343

spookycheekybreeky, 2023-09-23:
Hi @ZerBea thanks you for answering me that quickly,

I been trying to use the new version of hcxdumptools too (v6.3.1) but without any success too, maybe you can help me on the command should use ?

Thanks -SCB

ZerBea, 2023-09-23:
Make sure you're running latest driver:
https://github.com/aircrack-ng/rtl8812au

```
$ hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 74da380645e7 74da380645e7 + wlp48s0f4u2u4    rtl88XXau (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```

```
$ hcxdumptool -I wlp48s0f4u2u4

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 74da380645e7 74da380645e7 + wlp48s0f4u2u4    rtl88XXau (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] 20.0 dBm	  5075 [ 15] 30.0 dBm	  5080 [ 16] 30.0 dBm
  5085 [ 17] 30.0 dBm	  5090 [ 18] 30.0 dBm	  5100 [ 20] 30.0 dBm	  5120 [ 24] 30.0 dBm
  5140 [ 28] 30.0 dBm	  5160 [ 32] 23.0 dBm	  5180 [ 36] 23.0 dBm	  5200 [ 40] 23.0 dBm
  5220 [ 44] 23.0 dBm	  5240 [ 48] 23.0 dBm	  5260 [ 52] 20.0 dBm	  5280 [ 56] 20.0 dBm
  5300 [ 60] 20.0 dBm	  5320 [ 64] 20.0 dBm	  5340 [ 68] 20.0 dBm	  5360 [ 72] 30.0 dBm
  5380 [ 76] 30.0 dBm	  5400 [ 80] 30.0 dBm	  5420 [ 84] 30.0 dBm	  5440 [ 88] 30.0 dBm
  5460 [ 92] 30.0 dBm	  5480 [ 96] 26.0 dBm	  5500 [100] 26.0 dBm	  5520 [104] 26.0 dBm
  5540 [108] 26.0 dBm	  5560 [112] 26.0 dBm	  5580 [116] 26.0 dBm	  5600 [120] 26.0 dBm
  5620 [124] 26.0 dBm	  5640 [128] 26.0 dBm	  5660 [132] 26.0 dBm	  5680 [136] 26.0 dBm
  5700 [140] 26.0 dBm	  5720 [144] 13.0 dBm	  5745 [149] 13.0 dBm	  5765 [153] 13.0 dBm
  5785 [157] 13.0 dBm	  5805 [161] 13.0 dBm	  5825 [165] 13.0 dBm	  5845 [169] 13.0 dBm
  5865 [173] 13.0 dBm	  5885 [177] 30.0 dBm

bye-bye
```

```
$ sudo hcxdumptool -i wlp48s0f4u2u4 -F --rds=1
  CHA    LAST   R 1 3 P S    MAC-AP    ESSID (last EAPOL on top)  SCAN-FREQUENCY:   2437
-----------------------------------------------------------------------------------------
 [  6] 23:22:05 + +   + + 5004b81a2fd3 TEST_NETWORK
```

column P = + means PMKID successful captured.

From --help:
```
real time display:
 R = + AP display:     AP is in TX range or under attack
 S = + AP display:     AUTHENTICATION KEY MANAGEMENT PSK
 P = + AP display:     got PMKID hashcat / JtR can work on
 1 = + AP display:     got EAPOL M1 (CHALLENGE)
 3 = + AP display:     got EAPOL M1M2M3 (AUTHORIZATION) hashcat / JtR can work on
 E = + CLIENT display: got EAP-START MESSAGE
 2 = + CLIENT display: got EAPOL M1M2 (ROGUE CHALLENGE) hashcat / JtR can work on
```

ZerBea, 2023-09-23:
```
$ hcxdumptool -v
hcxdumptool 6.3.1-55-gde8e8b2 (C) 2023 ZeroBeat
compiled by gcc 13.2.1
compiled with Linux API headers 6.4.0
compiled with glibc 2.38
```

spookycheekybreeky, 2023-09-23:
I'm running hcxdumptools now with the following command:
`sudo hcxdumptool -i wlp48s0f4u2u4 -F --rds=1 --disable_deauthentication`
[image]

How long does capturing a pmkid take? I saw people wardriving with hcxdumptool so i just gussed a few secound but i cant get anyphing from 2 routers right now.

ZerBea, 2023-09-24:
It should take less than 5 seconds if the scan reached the AP channel.

Your entire R column is empty.
`R = + AP display:     AP is in TX range or under attack
`

Possible reasons:
Packet injection is not working
AP is not in range.

You are on 5GHZ. hcxdumptool respects the wireless regulatory domain which must not be unset.
The impact of this regulatory domain is huge:
https://github.com/aircrack-ng/aircrack-ng/discussions/2430#discussioncomment-5007839

hcxdumptool will not transmit on prohibited frequencies or if RADAR is detected on this channels.

To find out if PACKET injection is working, run hcxdumptool in active scan mode:
$ sudo hcxdumptool -i interface_name -F --rcascan=active

If packet injection is working as expected and the regulatory domain is correct and the AP is in Range, you should see a + in the A column. If that is not the case, you're running into one of the problems mentioned above.

spookycheekybreeky, 2023-09-24:
I executed the command you sent above and i got this as a output, The beacon is increasing but not the reponse. Does that mean my injection is not working ?
[image]

ZerBea, 2023-09-24:
If the response time does not increase, the AP does not respond to hcxdumptool requests due to the above mentioned reasons.

I pushed an update to hcxdumptool:
Now RESPONSE column is empty if no response was received.

ZerBea, 2023-09-24:
Please notice that not all Freeboxes transmit a PMKID:
https://wpa-sec.stanev.org/?search=Freebox-

spookycheekybreeky, 2023-09-24:
Yes i cheacked that and i saw other freebox was transmitiing pmkid also everyphing inside my config is on and by default and i never touched a fing in the wifi section except the password. I Just ran the new update of hcxdumptool for 5 minute and this is the output i have. 

[image]
[image]

ZerBea, 2023-09-24:
Some do this:
https://wpa-sec.stanev.org/?search=dc00b08b9de0
and some not:
https://wpa-sec.stanev.org/?search=8c97ea90ae00

ZerBea, 2023-09-24:
Is the regulatory domain configured?
```
$ sudo iw reg get
global
country FR: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 71000 @ 2160), (N/A, 40), (N/A)
```

spookycheekybreeky, 2023-09-24:
yes i cheacked that like 20 minutes ago
[image]

ZerBea, 2023-09-24:
It should not be 00 or 99, because hcxdumptool will not transmit on "PASSIVE SCAN only channels" or on DFS channels if a RADAR signal is detected.

spookycheekybreeky, 2023-09-24:
How i can change it so i can work with what i got ?

ZerBea, 2023-09-24:
$ sudo iw reg set FR
and to confirm it is correct
$  iw reg get

spookycheekybreeky, 2023-09-24:
Still having 99 on my interface 
[image]

ZerBea, 2023-09-24:
If it is still on 00 (DFS-UNSET) neither crda nor wireless regulatory domain is installed and/or configured.

ZerBea, 2023-09-24:
https://forum.armbian.com/topic/1677-crda-and-wireless-regdb/

spookycheekybreeky, 2023-09-24:
I didnt install anyphing except for your insane tools and armbian. Im gonna cheack out https://forum.armbian.com/topic/1677-crda-and-wireless-regdb/ and see if i can fix it

ZerBea, 2023-09-24:
Maybe the link mentioned above is helpful. I don't use armbian.
Runnning Arch Linux or raspbian, crda must be installed and the kernel must be compiled with option to use it.

spookycheekybreeky, 2023-09-24:
Does something like this can work if i change the years to 2022 ? 
https://gist.github.com/kylemcdonald/b4ef99605fdf32d872dbc72d966421a0

ZerBea, 2023-09-24:
to figure out whether AP cachin is activated or not, capture a 4way handshake between AP and a CLIENT.
Open the dump file by Wireshark.
Apply this filter:
wlan_rsna_eapol.keydes.msgnr == 1
left click on a packet coming from the target AP

AP dos not transmit a PMKID:
`WPA Key Data Length: 0
`

AP transmit a PMKID:
`WPA Key Data Length: 22
`

To check if PMKID is not zeroed left click on WPA Key Data:
```
Tag: Vendor Specific: Ieee 802.11: RSN PMKID
    Tag Number: Vendor Specific (221)
    Tag length: 20
    OUI: 00:0f:ac (Ieee 802.11)
    Vendor Specific OUI Type: 4
    Data Type: PMKID KDE (4)
    PMKID: should contain a not zeroed PMKID
```

ZerBea, 2023-09-24:
If the kernel is configured to use crda, it will work.

E.g. on Arch Linux:
```
$ zcat /proc/config.gz | grep CONFIG_CFG80211
CONFIG_CFG80211=m
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
CONFIG_CFG80211_DEFAULT_PS=y
CONFIG_CFG80211_DEBUGFS=y
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
```

spookycheekybreeky, 2023-09-24:
> to figure out whether AP cachin is activated or not, capture a 4way handshake between AP and a CLIENT. Open the dump file by Wireshark. Apply this filter: wlan_rsna_eapol.keydes.msgnr == 1 left [...]

I tried to capture traffic arounf me and reconnect and disconnect multiple time with my phone and i got this output.
[image]

and the command i ran is 
`tcpdump -i wlx00c0cab02c9a -s 65535 -w test.pcapng`

and for the crda here the output i get:
[image]

ZerBea, 2023-09-24:
First screenshot:
AP doesn't use PMKID caching.

Second screenshot:
crda is supported - you need to install it via your package management system.

ZerBea, 2023-09-24:
The get full capabilites (if crda is installed) set country IN:

```
$ sudo iw reg set IN
$ sudo iw reg get
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS
	(5470 - 5725 @ 160), (N/A, 24), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 30), (N/A)
```

There is no need to patch the regulatory domain need third party tools!

spookycheekybreeky, 2023-09-24:
> The get full capabilites (if crda is installed) set country IN:
> 
> ```
> $ sudo iw reg set IN
> $ iw reg get
> global
> country IN: DFS-UNSET
> 	(2402 - 2482 @ 40), (N/A, 30), (N/A)
> 	(5150 - 5250 @ 80), (N/A, 30), (N/A)
> 	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS
> 	(5470 - 5725 @ 160), (N/A, 24), (0 ms), DFS
> 	(5725 - 5875 @ 80), (N/A, 30), (N/A)
> ```
> 
> There is no need to patch the regulatory domain need third party tools!

I downloaded crda with the external deb package 
[image]

but without any success on changing the reg to IN
[image]

ZerBea, 2023-09-24:
There is no need to run deauthentication attacks. The build in REASSOCIATION attack is working fine to get a 4way handshake.

BTW:
depending on options, hcxdumptool will stop an attack immediately if a PMKID or a 4was handshake was captured.
```
--attemptclientmax=<digit>     : set maximum of attempts to request an EAPOL M2
                                  default: 10 attempts
                                  to disable CLIENT attacks set 0
--attemptapmax=<digit>         : set maximum of received BEACONs to request a PMKID or to get a 4-way handshake
                                  default: stop after 4 received BEACONs
                                  attemptapmax=0 include this options:
                                   disable_deauthentication: do not transmit DEAUTHENTICATION/DISASSOCIATION frames
                                   disable_proberequest    : do not transmit PROBEREQUEST frames
                                   disable_association     : do not AUTHENTICATE/ASSOCIATE
                                   disable_reassociation   : do not REASSOCIATE a CLIENT

```

spookycheekybreeky, 2023-09-24:
> There is no need to run deauthentication attacks. The build in REASSOCIATION attack is working fine to get a 4way handshake.
> 
> BTW: Depending on options, hcxdumptool will stop an attack immediately if a PMKID or a 4-way handshake was captured.
> 
> ```
> --attemptclientmax=<digit>     : set maximum of attempts to request an EAPOL M2
>                                   default: 10 attempts
>                                   to disable CLIENT attacks set 0
> --attemptapmax=<digit>         : set maximum of received BEACONs to request a PMKID or to get a 4-way handshake
>                                   default: stop after 4 received BEACONs
>                                   attemptapmax=0 include this options:
>                                    disable_deauthentication: do not transmit DEAUTHENTICATION/DISASSOCIATION frames
>                                    disable_proberequest    : do not transmit PROBEREQUEST frames
>                                    disable_association     : do not AUTHENTICATE/ASSOCIATE
>                                    disable_reassociation   : do not REASSOCIATE a CLIENT
> ```

yes but the REASSOCIATION attack is to force a client to reconnect to the AP and that is not clientless? or i am wrong. I dont see my self waitting for client to reconnect to the AP when warwalking/driving.

if it completly clientless yea that could work i just need to get my dam crda working

ZerBea, 2023-09-24:
Some Linux distributions split the package to crda and wireless-regulatory domain, e.g. Debian:
https://packages.debian.org/search?keywords=regdb&searchon=names&suite=stable&section=all

ZerBea, 2023-09-24:
You're not wrong. Except of the PMKID attack all attacks are not CLIENT less.
But some attacks are AP less  - useful if the target AP is not in RANGE while the CLIENT is in RANGE.
All M1M2ROGUE attacks are valid, because the MESSAGE PAIR is calculated based on the REQUEST of hcxdumptool and the RESPONSE of the CLIENT. That allow hashcat to run with --nonce-error-corrections=0.

spookycheekybreeky, 2023-09-24:
> You're not wrong. Except of the PMKID attack all attacks are not CLIENT less. But some attacks are AP less - useful if the target AP is not in RANGE while the CLIENT is in RANGE. All M1M2ROGUE [...]

For a wardriving setup i cant see my self asking client to reconnect to there routers. Is this really a good way to capture 4way handshake and how mush time i need to sniff out the info needed ?

ZerBea, 2023-09-24:
I don't want to answer this public. Please send me a PM I can respond to.
 
You get get the email address directly from git AP:
https://api.github.com/users/ZerBea/events/public

spookycheekybreeky, 2023-09-24:
> I don't want to answer this public. Please send me a PM I can respond to.
> 
> You get get the email address directly via git API: https://api.github.com/users/ZerBea/events/public

done.

ZerBea, 2023-09-24:
Thanks. 
Answered via PM (with attached PGP public key).

ZerBea, 2023-09-24:
I think we can close this report here, because hcxdumptool is working as expected.

closed 2023-09-24 by ZerBea

spookycheekybreeky, 2023-09-24:
> I think we can close this report here, because hcxdumptool is working as expected.

yes totaly