ZerBea/hcxdumptool #350: Output file is not being generated

henriqueArrazao, 2023-10-07:
Version 6.3.1
Command:
`sudo hcxdumptool -I wlan0 -w dumpfile.pcapng`
Problem:
The output file is not being created

ZerBea, 2023-10-07:
wrong command line ( I == capital):
```
$ hcxdumptool -h
-I <INTERFACE> : show detailed information about INTERFACE
```

correct command line (i == lower case):
`$ sudo hcxdumptool -i wlan0 -w dumpfile.pcapng`

henriqueArrazao, 2023-10-07:
> wrong command line ( I == capital):
> 
> ```
> $ hcxdumptool -h
> -I <INTERFACE> : show detailed information about INTERFACE
> ```
> 
> correct command line (i == lower case): `$ sudo hcxdumptool -i wlan0 -w dumpfile.pcapng`

Gee you're right, closing here, thank you.

ZerBea, 2023-10-07:
No problem.

ZerBea, 2023-10-07:
Pleas notice:
Default scan channels are 1a,6a and 11a only.

closed 2023-10-07 by ZerBea

henriqueArrazao, 2023-10-07:
> Please notice: Default scan channels are 1a, 6a and 11a only.

To be honest I don't have any experience on this, I'm following this [video](https://www.youtube.com/watch?v=Usw0IlGbkC4&t=850s&ab_channel=DavidBombal) where the guy says to enter this command (in a previous version of the package):
`sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15`

For the lastest version, I'm adapting it to this (In which by the way, does not find anything):
sudo hcxdumptool -i wlan0 -w dumpfile.pcapng --beacontx=15

Should I change something?

ZerBea, 2023-10-08:
Since v 6.3.0 hcxdumptool is highly optimized and most of the options are changed.
An advanced command line looks like this:
`$ hcxdumptool --bpf=own.bpfc --essidlist=standard.essidlist --beacontx=5 -c 1a,6a,11a,2a,1a,6a,11a,13a,1a,6a,11a,3a,1a,6a,11a,12a,1a,6a,11a,4a,1a,6a,11a,10a,1a,6a,11a,5a,1a,6a,11a,9a,1a,6a,11a,7a,1a,6a,11a,8a --rds=1
`
standard.essidlist
This list should contain some common or target ESSIDs to init hcxdumptool's internal ring buffer
It is not mandatory, because this ring buffer is filled with received ESSIDs.

own.bpfc
This list should contain own devices to be protected.
It is mandatory that MAC randomization on your own devices is disabled - otherwise they are not protected

c
The channel list is self explained

rds
This option makes sure that successful attacked NETWORKs are always on top of the AP status display