ZerBea/hcxdumptool #351: Enable Status

MattStr11, 2023-10-10:
I updated my tools and the "enable_status" is no longer a recognized option. Using the help command, I couldn't find enable status but I also checked the man page. I was able to find the command in the man page. I am just trying to figure out what changed as I know my syntax is correct because this is exactly what I used before. Thanks 

─# hcxdumptool -i wlan1 -w cap.pcap --enable_status=1
hcxdumptool: unrecognized option '--enable_status=1'
hcxdumptool 6.3.1 (C) 2023 by ZeroBeat
usage: hcxdumptool -h for help

ZerBea, 2023-10-11:
Very interesting, because there is no entry inside the man page and the man page refers to help.
https://github.com/ZerBea/hcxdumptool/blob/master/man/hcxdumptool.1
Arch Linux does not install this man page:
https://archlinux.org/packages/extra/x86_64/hcxdumptool/files/

Since v6.3.0 nearly everything changed in favor of speed. The code is highly optimized and everything that makes it slow has been removed. This is explained in the changelog:
```
06.04.2023
==========
release v6.2.9
several bug fixes

This is the last version:
that use WIRELESS EXTENSIONS
that use ETHTOOL to get/set virtual MAC address
that use old style status output
that use soft coded filter lists
that use msec timestamp
that use crypto stuff
that use server/client mode to display status

Next version will go back to the roots:
set focus on WPA PSK (WPA1, WPA2, WPA2 key version 3)
set bandwidth to 20MHz to increase range
set bitrate to lowest values to increase range
use active monitor mode
use NL80211 stack
use RTNETLINK
band a, b, c, d, e support
use NMEA messages:
 $GPRMC: Position, velocity, time and date
 $GPGGA: Position, orthometric height, fix related data, time
 $GPWPL: Position and MAC AP
 $GPTXT: ESSID in HEX ASCII
remove options that slow hcxdumptool down
```

There is also a discussion about the new style:
https://github.com/ZerBea/hcxdumptool/discussions/343

From no on, hcxdumptool is able to run completely headless, e.g. on a penetration testing system like this one:
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-2
The status display can be compiled completely out.  Remove -DSTATUSOUT from Makefile.
The same applies to GPS support. If you don't need it remove -DNMEAOUT from Makefile.

That means you now have two choices:
If you don't need a status display (headless system) - compile it out and hcxdumptool is really fast
If you really need a status display there is no longer need to enable it - just change its sorting by --rds

ZerBea, 2023-10-11:
Please notice:
If you need/like an expanded waterfall realtime display, hcxdumptool is designed to run in parallel with tshark and/or Wireshark.
You can do this either on hcxdumptool's monitor interface (real time) or on the hcxdumptool's pcapng dump file while hcxdumptool is running.

Example:
view all already captured PMKIDs while hcxdumptool is running:
run hcxdumptool in terminal 1:
`$ sudo  hcxdumptool -i interface -w test.pcapng`
run tshark in terminal 2:
`tshark -r test.pcapng -Y "wlan.rsn.ie.pmkid" -T fields -e wlan.ra -e wlan.ta -e wlan.rsn.ie.pmkid | sort | uniq`

More information is here:
https://tshark.dev/analyze/packet_hunting/packet_hunting/

ZerBea, 2023-10-11:
workflow:
Do not create interface by third party tools!
Do not use a virtual monitor interface (wlanXmon)!

hcxdumptool and tshark/Wireshark running on the same dump file:
`terminal1: hcxdumptool & BPF dump file pcapng -> terminal2: tshark -r dump file & display filter`

hcxdumptool and tshark/Wireshark running on the same interface:
`terminal1: hcxdumptool & BPF -> terminal2: tshark -i & -w & capture filter`

entire tshark wlan filter reference:
https://www.wireshark.org/docs/dfref/w/wlan.html

ZerBea, 2023-10-25:
Closed, because it is not an issue, but more a discussion.

closed 2023-10-25 by ZerBea