ZerBea/hcxdumptool #352: hcxdumptool not working for me pliz help

karimHI, 2023-10-12:
[image]
every time I start it it freezes here and  freezes my adapter and i need to unplug it and plug it again to make it work 
everything is updated 

distro : kali 2023.3 
kernel : 6.5.0-kali2-amd64

ZerBea, 2023-10-12:
I need some more information. Please comment output of:
```
$ lsusb (if the adapter is an USB adapter)
$ lspci (if the adapter is a PCIe card)
$ hcxdumptool -L
$ hcxdumptool -I YOUR_INTERFACE_NAME
$ hcxdumptool -v
```

Please also make sure that:
- all service are stopped that take access to the interface
- you have not used a third party tool to set monitor mode
- the driver is an official Linux kernel driver (third party drivers are unsupported)
- you don't use a virtual machine

karimHI, 2023-10-12:
lsusb : Bus 003 Device 008: ID 0e8d:7961 MediaTek Inc. Wireless_Device

hcxdumptool -L : 
***phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  3   9 90de80d696db 90de80d696db * wlan0            mt7921u (NETLINK)
***
 hcxdumptool -i wlan0 [image]
 
 hcxdumptool -v : 
 ***hcxdumptool 6.3.1-63-g2bd0496 (C) 2023 ZeroBeat
compiled by gcc 13.2.0
compiled with Linux API headers 6.5.3
compiled with glibc 2.37***

all service are stopped that take access to the interface    YES
you have not used a third party tool to set monitor mode    YES
the driver is an official Linux kernel driver (third party drivers are unsupported)     YES
you don't use a virtual machine   I'm using kali on vm ware

about the command i'm using even if i type only hxdumptool or --rcascan it freezes i tried with airggedon same issue

ZerBea, 2023-10-12:
Thanks for the information.
It looks like a driver problem. There are several problems with this driver.
Some problems are fixed:
https://www.spinics.net/lists/linux-wireless/msg221533.html
some not.

Please read more here:
https://github.com/morrownr/USB-WiFi/issues/107

Unfortunately  there is nothing I can do. We have to wait for a driver fix.

ZerBea, 2023-10-13:
Some more information about this driver problem is here:
https://github.com/ZerBea/hcxdumptool/issues/337

ZerBea, 2023-10-19:
I think we can close this here and wait for a driver fix.

closed 2023-10-19 by ZerBea