ZerBea/hcxdumptool #353: OLd targetBSSID.is NOW?

iamyangyiok, 2023-10-19:
Thanks for the update
It's a bit uncomfortable to jump from the old version to the new version. Please tell me what format should be used now in the old version of --filterlist=targetBSSID --filtermode=2 to lock the BSSID. grateful.

ZerBea, 2023-10-19:
To answer your question:
`--filterlist=targetBSSID --filtermode=2`
has been replaced by
`--bpf=filter.bpf`

The kernel include a fantastic filter method and hcxdumptool use it. The entire filtering is now done inside the kernel space via a Berkeley Packet filter which can be applied to hcxdumptool by the option bpf (--bpf=filter.bpf). That is fast as lightning and more flexible than the old user space filtering method.

There are several ways to build this filter, e.g.:
```
via tcpdump == the simplest way and easy to use for a beginner
via filter code that needs to be compiled == the most flexible way
```

How to create this filter is described in --help:
```
--bpf=<file>                   : input kernel space Berkeley Packet Filter (BPF) code
                                  steps to create a BPF (it only has to be done once):
                                  $ hcxdumptool -m <interface>
                                  create BPF to protect MACs
                                  $ tcpdump -i <INTERFACE> not wlan addr2 11:22:33:44:55:66 -ddd > protect.bpf
                                  recommended to protect own devices
                                  create BPF to attack a MAC
                                  $ tcpdump -i <INTERFACE> wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 or wlan addr3 11:22:33:44:55:66 -ddd > attack.bpf
                                  it is strongly recommended to allow all PROBEREQUEST frames (wlan_type mgt && wlan_subtype probe-req)
                                  $ tcpdump -i <interface> wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 or wlan addr3 11:22:33:44:55:66 or wlan addr3 ff:ff:ff:ff:ff:ff -ddd > attack.bpf
                                  see man pcap-filter for a list of all filter options
                                  add BPF code: 
                                  $ hcxdumptool -i <INTERFACE> --bpf=attack.bpf ...
```
and here:
https://github.com/ZerBea/hcxdumptool/discussions/343
https://github.com/ZerBea/hcxdumptool/discussions/348

BPF is explained here:
https://en.wikipedia.org/wiki/Berkeley_Packet_Filter
https://www.kernel.org/doc/html/latest/bpf/index.html
https://www.kernel.org/doc/html/latest/networking/filter.html
https://andreaskaris.github.io/blog/networking/bpf-and-tcpdump/

Closed this here, because it is not a bug. If you have some more questions, you please open a discussion.

closed 2023-10-19 by ZerBea

ZerBea, 2023-10-19:
BTW:
Using a BPF is nothing new. It is present in hcxdumptool since 06.03.2020 (changelog):
```
06.03.2020
==========
added option to import Berkeley packet Filter code from tcpdump as alternative to apfilterlist and clientfilterlist
```