ZerBea/hcxdumptool #355: Hcxdumptool on nethunter

axeldog, 2023-10-23:
Need help with hcxdumptool on nethunter. When running command "hcxdumptool -i wlan0 -w pmkid.pcapng" it runs and pumps in pmkid.pcapng but if i add any more to the command like --filterlist_ap=my filter.txt --filternode=3 or --enable_status=1 it throws "2 errors during run time" "failed to arm interface" .

ZerBea, 2023-10-23:
This options (--filterlist_ap=my filter.txt --filternode=3 or --enable_status=1) have been removed since v6.3.0.
From now on, it is mandatory to add a Berkeley Packet Filter (BPF) that do the entire filtering.

More information is here:
https://github.com/ZerBea/hcxdumptool/discussions/343
https://github.com/ZerBea/hcxdumptool/discussions/330

as well as here:
`$ hcxdumptool --help`

and in changelog:
https://github.com/ZerBea/hcxdumptool/blob/master/changelog#L83

Closed,this report here, because this changes are wanted and announced.

ZerBea, 2023-10-23:
Option "enable_status=1" has been replaced by "rds=1".
Options "filterlist_ap=my filter.txt and "filternode=3" have been replaced by "bpf=filter.bpfc".
How to build the Berkeley Packet Filter is explained in help.

ZerBea, 2023-10-23:
A typical command line (scan all frequencies supported by the interface) is
`$ sudo hcxdumptool -i INTERFACENAME -w dumpfile.pcapng --rds=1 -F`

In case of filtering:
`$ sudo hcxdumptool -i INTERFACENAME -w dumpfile.pcapng --rds=1 -F --bpf=filter.bpf`

In case if using a single channel, e.g. channel 1:
`$ sudo hcxdumptool -i INTERFACENAME -w dumpfile.pcapng --rds=1 -c 1a`

Please take a look at help to get an overview of all options and the legend of the new split screen.

axeldog, 2023-10-24:
Quick follow up question. On nethunter and kalilinux pc, any command with a "--" before it throws "unrecognised option " example --disable_deauthentication or --do_rcascan

ZerBea, 2023-10-24:
This option "--do_rcascan" has been replaced by "--rcascan=active".
If you don't change that, you'll get an ERROR:
```
$ sudo hcxdumptool -i wlp48s0f4u2u4 --do_rcascan
hcxdumptool: unrecognized option '--do_rcascan'
hcxdumptool 6.3.1-69-g532cc54 (C) 2023 by ZeroBeat
usage: hcxdumptool -h for help
```

Running the new style, everything should be fine:
```
$ sudo hcxdumptool -i wlp48s0f4u2u4 --rcascan=active

Requesting physical interface capabilities. This may take some time.
Please be patient...
```

axeldog, 2023-10-24:
Also attack.bpf is empty. Use command hcxdumptool -m wlan0 
tcpdump -i wlan0 wlan addr 11:4C:69:08:50:BB -ddd > attack.bpf writes an empty file.

axeldog, 2023-10-24:
Has the man page not been updated yet

ZerBea, 2023-10-24:
Command line is wrong.
Your command line:
```
$ sudo tcpdump -i wlan0 wlan addr 11:4C:69:08:50:BB -ddd > attack.bpf
tcpdump: can't parse filter expression: syntax error
```

correct command line:
```
$ sudo tcpdump -i wlan0 wlan addr3 11:4C:69:08:50:BB -ddd > attack.bpf

$ cat attack.bpf
16
48 0 0 3
100 0 0 8
7 0 0 0
48 0 0 2
76 0 0 0
2 0 0 0
7 0 0 0
80 0 0 0
84 0 0 12
21 5 0 4
64 0 0 18
21 0 3 1762152635
72 0 0 16
21 0 1 4428
6 0 0 262144
6 0 0 0
```

ZerBea, 2023-10-24:
The default man page of hcxdumptool points to "--help":
https://github.com/ZerBea/hcxdumptool/blob/master/man/hcxdumptool.1
It doesn't need an update.

If you manpage is different it is not created by me.

axeldog, 2023-10-24:
Sorry in the command line I do add "addr1" 
Also "man hcxdumptool" brings a whole new list of comands

ZerBea, 2023-10-24:
Arch Linux don't need a manpage (like Debian). So it is not installed by default.
makefile doesn't have an entry to install this manpage.
https://github.com/ZerBea/hcxdumptool/blob/master/Makefile

Maybe you have an old manpage or a manpage created by a packages maintainer.

axeldog, 2023-10-24:
Ahhh so confusing lol. 
New error. 
Tcpdump: eithernet address used in non-ether expression.  After trying to make bpf file.

ZerBea, 2023-10-24:
tcpdump will show this error if the interface is not in monitor mode and not up.

ZerBea, 2023-10-24:
If the target AP is not on channel 1a, 6a or 11a, don't forget to set the operation channel of the AP by option "-c" or scan all frequencies by option "-F".

axeldog, 2023-10-24:
I didn't insert wlan before addr3 just -I wlan0 addr3

ZerBea, 2023-10-24:
Ok

ZerBea, 2023-10-24:
Please also notice that running this filter:
`$ sudo tcpdump -i wlan0 wlan addr3 11:4C:69:08:50:BB -ddd > attack.bpf`

You'll get a warning (missing frames) by hcxpcapngtool when converting the dump file to hc22000 format because undirected proberequest frames are filtered out, too.

To avoid that, add a BROADCAST MAC to your filter:
`$ sudo tcpdump -i wlan0 wlan addr3 11:4C:69:08:50:BB or wlan addr3 ff:ff:ff:ff:ff:ff -ddd > attack.bpf`

Additional you can tell hcxdumptool to terminat if you got a PMKID or a handshake:
```
--exitoneapol=<type>           : exit on first EAPOL occurrence:
                                  bitmask:
                                   1 = PMKID
                                   2 = EAPOL M2
                                   4 = EAPOL M3
```

ZerBea, 2023-10-24:
The RTL8812AU driver is not part of the stock Linux kernel.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek?h=v6.5.8
and you need to install a third party driver.

Please read the adapter section of README.md (recommended and not recommended chipsets/drivers).

Please also read this:
https://github.com/morrownr/USB-WiFi/issues/314

There is a difference between addr1, addr2, addr3 and add4:
```
addr1 = to 
addr2 = from
addr3 = BSSID
addr4 = wireless distribution systems
```

https://www.geeksforgeeks.org/ieee-802-11-mac-frame/
https://www.rfwireless-world.com/Articles/WLAN-MAC-layer-protocol.html

axeldog, 2023-10-24:
Your a legend thanks so much for your help. Is there a dule band adaptor you would recomend. Currently using the tplink tl wn722n 2.4ghs.

axeldog, 2023-10-24:
Is there any other solutions to single out a specific ap to attack this dosent seem to work like you said.

ZerBea, 2023-10-24:
ALFA AWUS036ACM
ALFA AWSU036ACHM
Both drivers are well maintained and part of the Linux kernel.
Both support active monitor mode.

Give me a few minutes to run a test.

ZerBea, 2023-10-24:
here we go:
TP-Link TL-W722N v1:
```
$ lsusb
ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n
```

build filter:
```
$ sudo hcxdumptool -m wlp48s0f4u2u4
$ sudo tcpdump -i wlp48s0f4u2u4 wlan addr3 08:96:d7:98:e1:9e -ddd > attack.bpf
```
run attack against target test AP:
```
$ sudo hcxdumptool -i wlp48s0f4u2u4 --rds=1 -w tlwn722v1.pcapng -c 10a --bpf=attack.bpf

 CHA    LAST   R 1 3 P S    MAC-AP    ESSID (last EAPOL on top)  SCAN-FREQUENCY:   2457
-----------------------------------------------------------------------------------------
 [ 10] 18:55:06 + + +   + 0896d798e19e AP_7272

   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
 18:54:59     0896d798e19e 00e62d021987 AP_7272

282 packet(s) captured
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
31 EPB written to pcapng dumpfile

exit on sigterm
bye-bye

```

conver to hc22000 format:
```
$ hcxpcapngtool -o test.22000 tlwn722v1.pcapng
hcxpcapngtool 6.3.1-109-gac0db30 reading from tlwn722v1.pcapng...

summary capture file
--------------------
file name................................: tlwn722v1.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.5.8-arch1-1
application..............................: hcxdumptool 6.3.1-69-g532cc54
interface name...........................: wlp48s0f4u2u4
interface vendor.........................: f81a67
openSSL version..........................: 1.1
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 000da76303f4 (incremented on every new client)
MAC CLIENT...............................: c02250d406bb
REPLAYCOUNT..............................: 62460
ANONCE...................................: 4655d441f5e986d1c7102b72799bd537abc7a9c8dc150391bf988a2d0a8db87b
SNONCE...................................: 3c2c9d4853f5cfad58cde9c27e0aef9b0c80899f33e2664a21ea8a04c8df4464
timestamp minimum (GMT)..................: 24.10.2023 18:54:57
timestamp maximum (GMT)..................: 24.10.2023 18:54:59
duartion of the dump tool (seconds)......: 1
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianness (capture system)..............: little endian
packets inside...........................: 31
frames with correct FCS..................: 31
packets received on 2.4 GHz..............: 31
ESSID (total unique).....................: 1
BEACON (total)...........................: 1
BEACON on 2.4 GHz channel (from IE_TAG)..: 10 
PROBERESPONSE (total)....................: 1
AUTHENTICATION (total)...................: 3
AUTHENTICATION (OPEN SYSTEM).............: 3
ASSOCIATIONREQUEST (total)...............: 2
ASSOCIATIONREQUEST (PSK).................: 2
EAPOL messages (total)...................: 24
EAPOL RSN messages.......................: 24
EAPOLTIME gap (measured maximum msec)....: 52
EAPOL ANONCE error corrections (NC)......: not detected
REPLAYCOUNT gap (measured maximum).......: 4
EAPOL M1 messages (total)................: 18
EAPOL M2 messages (total)................: 2
EAPOL M3 messages (total)................: 2
EAPOL M4 messages (total)................: 2
EAPOL M4 messages (zeroed NONCE).........: 2
EAPOL pairs (total)......................: 4
EAPOL pairs (best).......................: 1
EAPOL pairs written to 22000 hash file...: 1 (RC checked)
EAPOL M32E2 (authorized).................: 1

frequency statistics from radiotap header (frequency: received packets)
-----------------------------------------------------------------------
 2457: 31	

Information: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

session summary
---------------
processed pcapng files................: 1
```

Please notice that BROADCAST frames are filtered out and hcxpcapngtool told us that.

Try to recover the PSK:
```
$ hashcat -m 22000 test.22000 -a 6 12345678
hashcat (v6.2.6-812-g8a206a9c0) starting
...
12e5828236c9a5a54795dde9b2b9d0f1:0896d798e19e:00e62d021987:AP_7272:12345678
                                                          
Session..........: hashcat
Status...........: Cracked
Hash.Mode........: 22000 (WPA-PBKDF2-PMKID+EAPOL)
Hash.Target......: test.22000
Time.Started.....: Tue Oct 24 19:01:45 2023 (0 secs)
Time.Estimated...: Tue Oct 24 19:01:45 2023 (0 secs)
Kernel.Feature...: Pure Kernel
Guess.Mask.......: 12345678 [8]
Guess.Queue......: 1/1 (100.00%)
Speed.#1.........:       73 H/s (0.42ms) @ Accel:32 Loops:256 Thr:64 Vec:1
Recovered........: 1/1 (100.00%) Digests (total), 1/1 (100.00%) Digests (new)
Progress.........: 1/1 (100.00%)
Rejected.........: 0/1 (0.00%)
Restore.Point....: 0/1 (0.00%)
Restore.Sub.#1...: Salt:0 Amplifier:0-1 Iteration:0-1
Candidate.Engine.: Device Generator
Candidates.#1....: 12345678 -> 12345678
Hardware.Mon.#1..: Temp: 42c Fan:  0% Util: 59% Core:2835MHz Mem:10802MHz Bus:16

Started: Tue Oct 24 19:01:44 2023
Stopped: Tue Oct 24 19:01:46 2023
```
For this test I configured a very simple PSK 12345678 which was successful recovered by hashcat.

ZerBea, 2023-10-24:
Same test using a TP-LINK TL-WN722N v2/v3:

device information
```
$ lsusb
ID 2357:010c TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]
```

attack using the already created filter:
```
$ sudo hcxdumptool -i wlp22s0f0u9u3 --rds=1 -w tlwn722v3.pcapng -c 10a --bpf=attack.bpf
  CHA    LAST   R 1 3 P S    MAC-AP    ESSID (last EAPOL on top)  SCAN-FREQUENCY:   2457
-----------------------------------------------------------------------------------------
 [ 10] 19:09:45 + + +   + 0896d798e19e AP_7272

   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
 19:09:43     0896d798e19e 00e62d021987 AP_7272
^C
209 packet(s) captured
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
25 EPB written to pcapng dumpfile

exit on sigterm
bye-bye
```

convert to hc22000:
```
$ hcxpcapngtool -o testv3.22000 tlwn722v3.pcapng
hcxpcapngtool 6.3.1-109-gac0db30 reading from tlwn722v3.pcapng...

summary capture file
--------------------
file name................................: tlwn722v3.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.5.8-arch1-1
application..............................: hcxdumptool 6.3.1-69-g532cc54
interface name...........................: wlp22s0f0u9u3
interface vendor.........................: 9ca2f4
openSSL version..........................: 1.1
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 111111300a88 (incremented on every new client)
MAC CLIENT...............................: a4a6a9dcefde
REPLAYCOUNT..............................: 64838
ANONCE...................................: 3b82ddb85e3fa8b4fd787bab084e310bec96f5fb4d008ec3f7766ecb55b69ff1
SNONCE...................................: 672a67b17e2bd2cd43e9b258b2bf066c8ecace54ba76b48c3cf2a1771a06c10c
timestamp minimum (GMT)..................: 24.10.2023 19:09:39
timestamp maximum (GMT)..................: 24.10.2023 19:09:43
duartion of the dump tool (seconds)......: 3
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianness (capture system)..............: little endian
packets inside...........................: 25
packets received on 2.4 GHz..............: 25
ESSID (total unique).....................: 1
BEACON (total)...........................: 1
BEACON on 2.4 GHz channel (from IE_TAG)..: 10 
PROBERESPONSE (total)....................: 1
AUTHENTICATION (total)...................: 3
AUTHENTICATION (OPEN SYSTEM).............: 3
ASSOCIATIONREQUEST (total)...............: 1
ASSOCIATIONREQUEST (PSK).................: 1
EAPOL messages (total)...................: 19
EAPOL RSN messages.......................: 19
EAPOLTIME gap (measured maximum msec)....: 51
EAPOL ANONCE error corrections (NC)......: not detected
REPLAYCOUNT gap (measured maximum).......: 2
EAPOL M1 messages (total)................: 14
EAPOL M2 messages (total)................: 1
EAPOL M3 messages (total)................: 2
EAPOL M4 messages (total)................: 2
EAPOL M4 messages (zeroed NONCE).........: 2
EAPOL pairs (total)......................: 2
EAPOL pairs (best).......................: 1
EAPOL pairs written to 22000 hash file...: 1 (RC checked)
EAPOL M32E2 (authorized).................: 1

frequency statistics from radiotap header (frequency: received packets)
-----------------------------------------------------------------------
 2457: 25	

Information: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

session summary
---------------
processed pcapng files................: 1
```

recover the PSK:
```
$ hashcat -m 22000 testv3.22000 -a 3 12345678
hashcat (v6.2.6-812-g8a206a9c0) starting
...
5ad4260e96fd4d538a11545708a5bb1b:0896d798e19e:00e62d021987:AP_7272:12345678
                                                          
Session..........: hashcat
Status...........: Cracked
Hash.Mode........: 22000 (WPA-PBKDF2-PMKID+EAPOL)
Hash.Target......: testv3.22000
Time.Started.....: Tue Oct 24 19:14:11 2023 (0 secs)
Time.Estimated...: Tue Oct 24 19:14:11 2023 (0 secs)
Kernel.Feature...: Pure Kernel
Guess.Mask.......: 12345678 [8]
Guess.Queue......: 1/1 (100.00%)
Speed.#1.........:       73 H/s (0.42ms) @ Accel:64 Loops:256 Thr:32 Vec:1
Recovered........: 1/1 (100.00%) Digests (total), 1/1 (100.00%) Digests (new)
Progress.........: 1/1 (100.00%)
Rejected.........: 0/1 (0.00%)
Restore.Point....: 0/1 (0.00%)
Restore.Sub.#1...: Salt:0 Amplifier:0-1 Iteration:0-1
Candidate.Engine.: Device Generator
Candidates.#1....: 12345678 -> 12345678
Hardware.Mon.#1..: Temp: 37c Fan:  0% Util: 75% Core:2850MHz Mem:10802MHz Bus:16

Started: Tue Oct 24 19:14:10 2023
Stopped: Tue Oct 24 19:14:12 2023
```
PSK has been successful recovered by hashcat.

ZerBea, 2023-10-24:
TP-LINK TLWN722Nv1 and v2/v3 driver overview:
```
$ hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  6   9 9ca2f4093af0 9ca2f4093af0 + wlp22s0f0u9u3    rtl8xxxu (NETLINK)
  7  10 f81a67077d0e f81a67077d0e + wlp22s0f0u9u1    ath9k_htc (NETLINK)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```

ZerBea, 2023-10-24:
Please notice:
It is mandatory that you know which services on your system are configured and running.
It is also mandatory that you stop them if they take access to the device.

axeldog, 2023-10-25:
Bit of a problem with or without -c 11a  on the scan screen it puts me on to chanel 8

ZerBea, 2023-10-25:
That is more related to the hardware and caused by overlapped channels.
Take a look at this picture:
https://technologyuk.net/telecommunications/telecom-principles/images/telecom-principles-0075.gif

If the center frequency is on 2.437 GHz (channel 6), you receive signals from channel 4, 5 and 7, 8, too.
That depend on the bandwidth and the proximity to the transmitter.

There are explanations in www, e.g.:
https://electronics.stackexchange.com/questions/155643/why-do-wifi-channels-overlap
https://serverfault.com/questions/152920/what-happens-when-wifi-channels-overlap
https://witestlab.poly.edu/blog/ieee-802-11-wlans-using-partially-overlapped-channels/

BTW:
The actual operation frequency is displayed on the top right corner of the status display.

ZerBea, 2023-10-25:
BTW:
hcxpcapngtool print both information after reading a pcapng dump file:
```
BEACON on 2.4 GHz channel (from IE_TAG)..: 1 2 3 4 5 6 7 8 9 11 13 
BEACON on 5/6 GHz channel (from IE_TAG)..: 36 40 44 
...
frequency statistics from radiotap header (frequency: received packets)
-----------------------------------------------------------------------
 2412: 1833	 2417: 30	 2422: 4	 2427: 32	
 2432: 216	 2437: 1280	 2442: 35	 2447: 312	
 2452: 16	 2457: 2	 2462: 3890	 2467: 7	
 2472: 8	 5180: 27	 5200: 11	 5220: 20	
 5240: 4	
```

ZerBea, 2023-10-25:
If you take a look at the frequency statistics from radio tap header, you can imagine why hcxdumptool scans channels 1a (2412), 6a (2437) and 11a (2462)  by default only.

axeldog, 2023-10-25:
Cool thanks will research it. All working on kalilinux and nethunter.

axeldog, 2023-10-26:
What dose hcxdumptool do that it must be used in a safe environment?

ZerBea, 2023-10-27:
I guess you mean this:
`BPF is unset! Make sure hcxdumptool is running in a 100% controlled environment!`

Simply explained: _hcxdumptool is an angry and hungry beast - by default it attacks everything_

But it can be completely controlled by a BPF and by its options and it's up to the user to control it.

axeldog, 2023-10-27:
But what dose it do by default without a bpf filter to make it such a hungry beast.

ZerBea, 2023-10-27:
- Capture every EAPOL (that include EAP hashes).
- REASSOCIATE EVERY connected CLIENT and request all EAPOL M2.
- Allow not connected CLIENTs to connect and request all EAPOL M2.
- REQUEST EAP-IDs.

on every WPA-PSK system (APs and CLIENTs) in range.

axeldog, 2023-10-27:
Dose it deauthenticate clients from ap to make them reconnect to get handshakes?

ZerBea, 2023-10-27:
DEAUTHENTICATION and DISASSOCIATION attacks (old school) are implemented, too.
State of the art APs and CLIENTs are hardened against this ancient attacks.
Either they run Management Frame Protection (MFP) or they ignore into data sequences injected DEAUTHENTICATION frames.

axeldog, 2023-10-27:
Oh wow so David bombals youtube video could get you in to alot of trouble for Denial of service because he dosent use a filter. 
Dose the rcascan just scan to gather information without any attacks.

ZerBea, 2023-10-27:
rcascan provide 2 modes:
```
active scan
passive scan
```

Running active scan it transmit PROBEREQUESTs and count BEACONs and responses
Running passive scan it BEACONs.

Nearly every CLIENT is doing active scans to discover APs in range.

axeldog, 2023-10-27:
Ok cool. My thinking is I want to start a business testing wifi passwords with real life tests. Do you have any advice.

ZerBea, 2023-10-27:
hcxdumptool runs layer 2 attacks only.
It neither runs DoS attacks nor layer 1 attacks nor attacks > layer 2.

ZerBea, 2023-10-27:
Real life attacks should include all layers (1 to 7) and you have to cover them all.
https://cdn.rohde-schwarz.com/pws/dl_downloads/dl_application/application_notes/1ma69/1MA69_2e.pdf

Therefore you have to learn the entire stuff (much more than I mentioned in README.md).

Regarding 802.11, a good starting point is:
`802.11® Wireless Networks: The Definitive Guide`
https://www.oreilly.com/library/view/80211-wireless-networks/0596100523/index.html

Regarding Linux, a good starting point is:
https://man7.org/tlpi/

Regarding coding, I suggest C and Assembler.
https://gcc.gnu.org/onlinedocs/gcc/

axeldog, 2023-10-27:
Yeah that's alot to learn. Oh yeah I'm testing this on my home router wich I'm pretty sure is wifi6 and it works but it shouldn't work should it ?

ZerBea, 2023-10-27:
hcxdumptool is state of the art and running state of the art attacks against all WPA-PSK secured NETWORKs.
Even if the AP is hardened, most of the CLIENTs are weak (and hcxdumptool show a successful attack with a "+" in the M2 column.

axeldog, 2023-10-27:
Do you mind if i put some of our conversation in a youtube video it's is all interesting and I'm sure will help others.

ZerBea, 2023-10-27:
Maybe. But like Arch Linux, hcxdumptool and hcxtools are rolling release tools.
If I can do something better or faster, I'll change everything. In that case, a video will be outdated (like David*s video).
That will cause that new users run into the same problems like you did.

axeldog, 2023-10-27:
Yeah I get what you're saying. So are you the actual creator of the hcx project. Is there anything you're working on at the moment improvements or otherwise.

ZerBea, 2023-10-27:
I'll say, watching a video or installing KALI by graphical installer is far, far away from handling hcxdumptool/hcxtools.

For beginners I recommend to run aircrack-ng and to learn 802.11 instead of doing
hcxdumptool -> hcpcapngtool -> hashcat/JtR

Neither hcxdumptool, nor hcxtools, nor hashcat or JtR will satisfy a newbee.
https://hashcat.net/forum/thread-11669-post-59437.html#pid59437

ZerBea, 2023-10-27:
We're working on:
https://github.com/hashcat/hashcat/issues/3887
and  on WPA3.

axeldog, 2023-10-27:
Yeah I'm defo a newbie. I typically don't use a kali gui as I use a nethunter phone and the gui is crap. I use hashcat on windows for the gpu speed and have used aircrack to attack capture and crack hashes. Currently I'm trying to get an alfa wifi adaptor to work on nethunter as iv got it working on kalilinux with drivers and headers but am stuck.

ZerBea, 2023-10-27:
My top ranking due to excellent driver support (if not connected to an USB3 port due to USB3 problems) on Linux stock kernel:
ALFA AWUS036ACHM and ALFA AWUS036ACM

axeldog, 2023-10-27:
The kali docs for nethunter say the RTL8812AU supported in nethunter but dosent seem to work. Iv reached out to David bombal as he has a working nethunter phone and the WiFi adaptor I want to use but not had a reply. Also reached out to someone on hear as its his name that pops up in the error but again not had a response. Not everyone is as nice as you on giving help. 
[image]

ZerBea, 2023-10-27:
RTL8812AU  is not part of the Linux stock kernel.
I guess that KALI install this driver by default:
https://github.com/aircrack-ng/rtl8812au

axeldog, 2023-10-27:
Yeah when I go to make the aircrack driver that's where the problems start. I guess it's something to do with the header but can't find a nethunter supported one it's not on the kale pool list.

ZerBea, 2023-10-27:
Maybe it's a good idea to ask Christian (kimocoder).

axeldog, 2023-10-27:
In that picture I sent cyberknight777 name pops up in the error and have messaged him but no reply. Who is Christian?

ZerBea, 2023-10-27:
Ok, I compiled the driver. Lookinf good, so far:
[25457.065189] usbcore: registered new interface driver rtl88XXau
[25465.063996] usb 1-9.3: new high-speed USB device number 9 using xhci_hcd
[25465.207410] usb 1-9.3: New USB device found, idVendor=7392, idProduct=a812, bcdDevice= 2.00
[25465.207414] usb 1-9.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[25465.207417] usb 1-9.3: Product: Edimax AC600 USB
[25465.207419] usb 1-9.3: Manufacturer: Realtek
[25465.207421] usb 1-9.3: SerialNumber: 00e04c000001
[25469.850382] usb 1-9.3: 88XXau 74:da:38:06:45:e7 hw_info[107]
[25469.904766] rtl88XXau 1-9.3:1.0 wlp22s0f0u9u3: renamed from wlan0
[25472.258862] mt7601u 5-2.4:1.0 wlp48s0f4u2u4: entered promiscuous mode

ZerBea, 2023-10-27:
hcxdumptool is running fine, too:
```
$ sudo hcxdumptool -I wlp22s0f0u9u3

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  3   6 74da380645e7 74da380645e7 + wlp22s0f0u9u3    rtl88XXau (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] 20.0 dBm	  5075 [ 15] 30.0 dBm	  5080 [ 16] 30.0 dBm
  5085 [ 17] 30.0 dBm	  5090 [ 18] 30.0 dBm	  5100 [ 20] 30.0 dBm	  5120 [ 24] 30.0 dBm
  5140 [ 28] 30.0 dBm	  5160 [ 32] 23.0 dBm	  5180 [ 36] 23.0 dBm	  5200 [ 40] 23.0 dBm
  5220 [ 44] 23.0 dBm	  5240 [ 48] 23.0 dBm	  5260 [ 52] 20.0 dBm	  5280 [ 56] 20.0 dBm
  5300 [ 60] 20.0 dBm	  5320 [ 64] 20.0 dBm	  5340 [ 68] 20.0 dBm	  5360 [ 72] 30.0 dBm
  5380 [ 76] 30.0 dBm	  5400 [ 80] 30.0 dBm	  5420 [ 84] 30.0 dBm	  5440 [ 88] 30.0 dBm
  5460 [ 92] 30.0 dBm	  5480 [ 96] 26.0 dBm	  5500 [100] 26.0 dBm	  5520 [104] 26.0 dBm
  5540 [108] 26.0 dBm	  5560 [112] 26.0 dBm	  5580 [116] 26.0 dBm	  5600 [120] 26.0 dBm
  5620 [124] 26.0 dBm	  5640 [128] 26.0 dBm	  5660 [132] 26.0 dBm	  5680 [136] 26.0 dBm
  5700 [140] 26.0 dBm	  5720 [144] 13.0 dBm	  5745 [149] 13.0 dBm	  5765 [153] 13.0 dBm
  5785 [157] 13.0 dBm	  5805 [161] 13.0 dBm	  5825 [165] 13.0 dBm	  5845 [169] 13.0 dBm
  5865 [173] 13.0 dBm	  5885 [177] 30.0 dBm

bye-bye
```

ZerBea, 2023-10-27:
And frame injection is working like a charm:
```
$ sudo hcxdumptool --rcascan=active
...
377 packet(s) captured

exit on sigterm
bye-bye
```

ZerBea, 2023-10-27:
No, that is not fair to say.
KALI is an excellent distribution ... if you are an experienced Linux user (!!!) and an experienced penetration tester(!):

"As the distribution’s developers, you might expect us to recommend that everyone should be using Kali Linux. The fact of the matter is, however, that Kali is a Linux distribution specifically geared towards professional penetration testers and security specialists, and given its unique nature, it is NOT a recommended distribution if you’re unfamiliar with Linux or are looking for a general-purpose Linux desktop distribution for development, web design, gaming, etc."

https://www.kali.org/docs/introduction/should-i-use-kali-linux/#is-kali-linux-right-for-you

But shame on all tutorials and blogs that create the impression that it is good for a Linux beginner.

axeldog, 2023-10-27:
True.

[50 more replies]