ZerBea/hcxdumptool #356: Active beacons and enable staus issue

anonymouse-257, 2023-10-24:
Hey I've read on other issue that hcxdumptool updated and removed all slow thing something like this,
And I'm watching some videos on YouTube (I've watched David Bombal tutorial) and he typed the following command:sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=1.
Now I've switched the -o in -w but the error is unrecognized option (that error appears to active beacon and enable status) and I know that hcxdumptool updated just idk what to do rn, how can I fix this

ZerBea, 2023-10-24:
You may have noticed that I have changed the head line  because there is already a discussion about that:
https://github.com/ZerBea/hcxdumptool/discussions/343

So we can close this report, because the new style is not a bug.

closed 2023-10-24 by ZerBea

anonymouse-257, 2023-10-24:
Aight tnx for the help and sorry for my bad English I'm more thing is how can I still use the enable staus option in the latest version what cmd do I need to type?

ZerBea, 2023-10-24:
`--enable_status has been replaced by --rds=1`

Typical commands are shown here:
https://github.com/ZerBea/hcxdumptool/issues/355#issuecomment-1776020019

ZerBea, 2023-10-24:
BTW:
My english is not much better.

anonymouse-257, 2023-10-24:
Aight thanks and you're English is very well I'm sure, now this is a fun question,what language this tool has written?

ZerBea, 2023-10-24:
I'm German.

anonymouse-257, 2023-10-24:
No,what is the code language that tool has written?

ZerBea, 2023-10-24:
Also I know that the entire 802.11 stuff is really hard to understand.
And everything is described in English which doesn't make things easier.

ZerBea, 2023-10-24:
Completely written in C.

anonymouse-257, 2023-10-24:
Wow, how old are you?

anonymouse-257, 2023-10-24:
btw how can I enable beacon in the latest version? Or the scan includes beacon in by default

ZerBea, 2023-10-24:
Close to 70 years

anonymouse-257, 2023-10-24:
fr?

anonymouse-257, 2023-10-24:
Man, I'm 16 😂

ZerBea, 2023-10-24:
Default options include transmitting a BEACON.
It can be disabled by --beacontx=0

anonymouse-257, 2023-10-24:
Aight tnx

ZerBea, 2023-10-24:
Either use the default value (10 BEACONs) or reduce it to 5 which will increase the chance that the target really find hcxdumptool during its scan.

anonymouse-257, 2023-10-24:
But wow completely on C man you're genius asf

ZerBea, 2023-10-24:
I have learned Assembler, but C is ok.

anonymouse-257, 2023-10-24:
Oh thanks

anonymouse-257, 2023-10-24:
Man, assembler? For real? You're Albert Einstein or something

anonymouse-257, 2023-10-24:
Oh and btw do you think that hashcat is faster on windows instead of VM?

ZerBea, 2023-10-24:
No, that was part of my of my education.

I don't use Windows and I don't use a VM. All my systems are running Linux. So I can't answer this questions.

anonymouse-257, 2023-10-24:
Oh you're runnin Linux as a default operating system btw what Linux Type you use?

ZerBea, 2023-10-24:
Desktop, notebook and netbook: Arch Linux
Raspberry Pi (Zero, A, B): Raspberry Pi OS Lite (Debian derivative)
Smart Phone: Lineage

anonymouse-257, 2023-10-24:
Wow man you are my inspiration