ZerBea/hcxdumptool #362: Capturing wifi using hcxdumptools says invalid option help

zotim1, 2023-11-16:
Writing hcxdumptools -i wlan0 -o test1.pcapng -- enable_status=1 says invalid option --'o' can anyone help on how to solve this problem

v74863, 2023-11-17:
Please use the latest version of the hcxdumptool
https://github.com/ZerBea/hcxdumptool/discussions/330

zotim1, 2023-11-17:
Sorry but I am a real beginner at this stuff so what should I use instead of -o

v74863, 2023-11-18:
hcxdumptool has built-in help.
Just type "hcxdumptool -h" and press Enter
Also read here, please
https://github.com/ZerBea/hcxdumptool/issues/342
https://github.com/ZerBea/hcxdumptool/issues/347
https://github.com/ZerBea/hcxdumptool/issues/355
https://github.com/ZerBea/hcxdumptool/issues/356

hcxdumptool is not for beginners. You have to know Linux.
If it is too complicated for you, start with wifite
https://null-byte.wonderhowto.com/how-to/automate-wi-fi-hacking-with-wifite2-0191739/

ZerBea, 2023-11-20:
As mentioned by @v74863 it is always a good idea to read --help, README.md and the changelog to get some information.

In your case, the simplest command line is:
```
$ hcxdumptool -i INTERFACE_NAME -w dump.pcapng -F --rds=1
```
By this options hcxdumptool use the interface named by "INTERFACE_NAME", store all information to a dump file named dump.pcapng, scan all available frequencies (frequencies supported by the driver and allowed by wireless regulatory domain) and sort the status display by successful attacks.

Please notice:
It is mandatory that all services that take access to the device are stopped before you run hcxdumptool.

I close this report, because it is more a discussion and not a bug.

closed 2023-11-20 by ZerBea