ZerBea/hcxdumptool #364: need  obtain to  additional  package

LLH-l, 2023-11-19:
In capturing  handshake, needs  obtain to following additional  package 

`0x00, 0x02, 0x04, 0x05, 0x08, 0x020, 0x028, 0x01, 0x0b `

ZerBea, 2023-11-20:
hcxdumptool is not designed to be a monitoring tool that captures all frame types because this makes hcxdumptool slow.
Other tools (tsahrk/Wireshark) can do this much better.

But it is designed to work in perfect harmony with tshark and/or Wireshark. Both tools are working fine, using hcxdumptools's monitor mode. You can run them in parallel (on the same interface) to hcxdumptool.

A simple example:
```
$ sudo hcxdumptool -i wlp48s0f4u2u4 & tshark -w test.pcapng -i wlp48s0f4u2u4
```
When finished:
```
$ sudo killall hcxdumptool
$ killall tshark
```

ZerBea, 2023-11-22:
Suggestion is now shown in help:
```
$ hcxdumptool --help
hcxdumptool 6.3.2-18-g1266cf2  (C) 2023 ZeroBeat
Additional information:
-----------------------
first stop all services that take access to the interface, e.g.:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service
run hcxdumptool
press ctrl+c to terminate
press GPIO button to terminate
 hardware modification is necessary, read more:
 https://github.com/ZerBea/hcxdumptool/tree/master/docs
stop all services (e.g.: wpa_supplicant.service, NetworkManager.service) that take access to the interface
do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
do not use logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
do not use virtual machines or emulators
do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)
do not use tools to change MAC (like macchanger)
do not merge (pcapng) dump files, because this destroys assigned hash values!
to store entire traffic, run <tshark -i <interface> -w allframes.pcapng> in parallel on the same interface
```

closed 2023-11-22 by ZerBea

ZerBea, 2023-11-23:
I checked the list mentioned in your commend.
All this frames types (and some more) are captured and stored by hcxdumptool.

Frame types overview:
https://github.com/ZerBea/hcxdumptool/blob/master/include/ieee80211.h#L11
This 802.11 frames are stored to the pcapng file:
```
management frames:
0x00 == IEEE80211_STYPE_ASSOC_REQ
0x01 == IEEE80211_STYPE_ASSOC_RESP
0x02 == IEEE80211_STYPE_REASSOC_REQ
0x03 == IEEE80211_STYPE_REASSOC_RESP (not in your list)
0x04 == IEEE80211_STYPE_PROBE_REQ
0x05 == IEEE80211_STYPE_PROBE_RESP
0x08 == IEEE80211_STYPE_BEACON (only on first occurrence)
0x0b == IEEE80211_STYPE_AUTH
0x0d == IEEE80211_STYPE_ACTION (not in your list)
data frames:
0x20 == IEEE80211_STYPE_DATA (filtered by EAP/EAPOL)
0x28 == IEEE80211_STYPE_QOS_DATA  (filtered by EAP/EAPOL)
```

BTW:
It doesn't make sense to handle all data frames during an attack, because it slow down the attack.
To offline decrypt the traffic, it is mandatory that an entire session (handshake and following traffic) is recorded.
That is mostly not the case if hcxdumptool moves to the next channel.