ZerBea/hcxdumptool #369: beware! linux-firmware has changed mt76 fw path

kimocoder, 2023-11-22:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=8451c2b1d529dc1a49328ac9235d3cf5bb8a8fcb

they made changes...  but openwrt/mt76 hasnt followed.

https://github.com/openwrt/mt76/blob/master/mt76x2/mt76x2.h

ZerBea, 2023-11-22:
Thanks for that information.
Usually I take the source directly from the Linux kernel tree (stable and mainline) to test patches.
In case of mt7921u monitor problem we (@morrownr) decided to collect this problems on upstream directly:
https://github.com/morrownr/USB-WiFi/discussions/260#discussioncomment-7575090
That's much better than handling this problem here:
https://github.com/morrownr/USB-WiFi/issues
or here:
https://github.com/ZerBea/hcxdumptool/issues

I also noticed that there was/is work in progress (during the last past year):
https://lore.kernel.org/all/YapTmM3EztojTS9F@lore-desk/T/
https://patchwork.kernel.org/project/linux-wireless/patch/c61deba952e23bcfa98e28778bc11af524ad2f1e.1645832790.git.objelf@gmail.com/#24755669

The driver is working fine (managed mode):
```
[120392.004344] mt7921u 1-9.3:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
[120393.638531] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: renamed from wlan0
```
only monitor mode and frame injection failed.

ZerBea, 2023-11-22:
Out of scope.
Have you seen this patch?
https://bugzilla.kernel.org/show_bug.cgi?id=217205#c77