ZerBea/hcxdumptool #371: Lists (aplist, clientlist, etc) are sorted AFTER item is evicted, potentially resulting in something not LRU being dropped

roboman2444, 2023-11-23:
I might be making some incorrect assumptions, or misunderstanding something. But it seems to me that most places that enter something into a list (aplist, clientlist, etc) do the eviction before the sort on tsakt. This could potentially result in something recently used being evicted, when if it was sorted prior (like a recent call to show_realtime), it wouldn't have.

I'm going to use process80211beacon_rca, as an example because it is reasonably simple, but many other process80211* functions are similar. The sort is done all the way at the end of the function, after the old entry in the list has been zero'd and replaced.

https://github.com/ZerBea/hcxdumptool/blob/1266cf28f4a4ff69f3b1b3c04c0473305491e689/hcxdumptool.c#L2381-L2414

I'm not sure what a good fix for this would be. Sorting ahead of replacing (right after the linear search) would result in the new item being inserted at the end... but that may not matter if everything is going to be sorted before any further modifications are made. Double sorting, once before, and once after, for completeness? Locating the index of the LRU for overwriting while doing the initial linear search?

Lemme know if i've made mistakes with understanding what's going on, or if this really doesn't matter and isn't an issue at all.

roboman2444, 2023-11-23:
I'd also add that the linear search does not check the very last entry. Most techniques to fix this issue are likely going to need to change the i < APLIST_MAX - 1 to just i< APLIST_MAX, just so it doesn't miss anything, and doesn't assume the final element is getting evicted.

ZerBea, 2023-11-23:
Exactly this behavior is wanted and all process80211* functions are similar (except the .
The list is actin as a ring buffer in which the last entry is always replaced by a new entry.

If the ring buffer is not completely filled, a new entry is always at position last entry +1
https://github.com/ZerBea/hcxdumptool/blob/1266cf28f4a4ff69f3b1b3c04c0473305491e689/hcxdumptool.c#L2400
Than we do the sort on ap list +1
https://github.com/ZerBea/hcxdumptool/blob/1266cf28f4a4ff69f3b1b3c04c0473305491e689/hcxdumptool.c#L2412
Doing this, we make sure that the actual entry is on top of the list. 
On the next occurrence we don't need to run through the entire list. That will speed up hcxdumptool on e.g. a Raspberry Pi.

If the ring buffer is filled completely, we replace the oldest entry by the new one.
Than we do the sort and the new entry went on rank 1. 

We are not interested in the last entry, because it will always be replaced by a new entry if the ring buffer is filled.

You can add some debug printf's to monitor this behavior.

ZerBea, 2023-11-23:
I forgot to mention that these ring buffers have only one purpose: keep the dump file as small as possible.
e.g. only one BEACON from an AP instead of flooding the dump file with BEACONs from the same AP,
e.g. only one PROBERESPONSE from an AP instead of flooding the dump file with PROBERESONSES from the same AP.

You'll notice that this fails if more than 250 APs are in range. In that case we see more than one BEACON of an AP. But this is much better than storing a BEACON every 100ms.

ZerBea, 2023-11-23:
We can say that the behavior is more likely LFU than LRU.

ZerBea, 2023-11-23:
But anyway, based on your report I discovered an ugly copy and paste error
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L544
and fixed it by this commit:
https://github.com/ZerBea/hcxdumptool/commit/be378e1b67ff85708b8581c7aba3128b95f14676

ZerBea, 2023-11-23:
[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/13451781/hcxdumptool.c.zip)

This debug code demonstrate LFUDA.
Make a copy of hcxdumptool folder and replace hcxdumptool.c by the attached debug code
```
$ make
$ hcxdumptool
```

ZerBea, 2023-11-24:
In case of LRU it might be an issue, but not in case of LFDUA.
I think we can close this report, because the debug code exactly shows the expected behavior.

closed 2023-11-24 by ZerBea

roboman2444, 2023-11-28:
> It also will fail if you e.g. do a war walk and came back to the starting point. In that case you have at least two BEACONs (or more, dependent on the duration of hcxdumptool - but never ever two equal BEACONs in a row) of the same AP.

Yeah, This was the exact failure case i was thinking of. Thanks for clarifying that this isn't really a big issue in the grand scheme of things.

ZerBea, 2023-11-29:
I had to make a compromise:
Performance (fast attacks - small ring buffers) on the one hand.
As little duplicate storage (BEACON frames) as possible on the other hand.

For sure, we can avoid this (duplicate storage) completely, but the price tag will be huge.
Passive dumpers (if they accept a packet loss) can do this.
But hcxdumptool is an interactive tool. It is mandatory that it respond just in time (instead of walking through endless buffers).

ZerBea, 2023-11-29:
The size of the ring buffer is defined here:
https://github.com/ZerBea/hcxdumptool/blob/master/include/hcxdumptool.h#L87
If you have a fast CPU and fast RAM, you can increase it to avoid dupes completely. The default value is a good compromise if hcxdumptool is running on a Raspberry Pi model 1.