ZerBea/hcxdumptool #376: driver is busy: failed to transmit proberesponse

karimHI, 2023-11-27:
anyway to fix this error while using an mt7921u usb wifi adapter on openwrt 

everything on the latest version (openwrt , hcxdumptool , and mediatek firmware and driver )

`18:06:04   4 driver is busy: failed to transmit proberesponse
18:06:04   4 driver is busy: failed to transmit proberesponse`

`root@OpenWrt:~# hcxdumptool -i wlan0 --enable_status=15 --error_max=999999999999
99999 -o haha2.pcapng
initialization of hcxdumptool 6.2.4...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 3 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 90de80d696db (not used for the attack)
INTERFACE VIRTUAL MAC.....: 90de80d696db (not used for the attack)
DRIVER....................: mt7921u
DRIVER VERSION............: 5.10.176
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.1
ERRORMAX..................: 2147483647 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 146e0a544f36 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 146e0a544f37 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 146e0a544f38 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b4e1eba21f2d
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62113
ANONCE....................: 0da8e210c87c410481f5c8b868171f04e263120fac124373ba585bb3b5a72f7b
SNONCE....................: e7211afef8e67e95d60ef0dd34ba54841c1cd9ab7939576498da0dfc15606674

18:05:53   1 ffffffffffff c4ea1d1f432f ************* [BEACON]
18:05:53   1 ffffffffffff f0b4d214b2b2 ************** [BEACON]
18:05:53   1 ffffffffffff bcf88b6253bc ****************** [BEACON]
18:05:53   1 ffffffffffff 505d7abbc80a ************** [BEACON]
18:05:53   1 ffffffffffff 14007df9afc8 ***************** [BEACON]
18:05:53   1 ffffffffffff a0092e954762 ******************* [BEACON]
18:05:53   1 ffffffffffff 14007df6434a ******************** [BEACON]
18:05:53   1 ffffffffffff 4033062f435f **************** [BEACON]
18:05:53   1 ffffffffffff 54ce82b1d8ce **************** [BEACON]
18:05:53   1 ffffffffffff 56ce82b1d8ce [HIDDEN BEACON]
18:05:54   1 ffffffffffff 90916478a563 *************** [BEACON]
18:05:54   1 ffffffffffff 92916478a563 [HIDDEN BEACON]
18:05:54   1 ffffffffffff 729c2789a2e8 ******************2 [BEACON]
18:05:54   1 ffffffffffff 42e469b25756 ****************[BEACON]
18:05:54   1 ffffffffffff 98482790abd7 ***************** [BEACON]
18:05:54   1 88b5ffce31de 729c2789a2e8 ************ [PROBERESPONSE]
18:05:55   1 driver is busy: failed to transmit acknowledgement
18:05:56   2 driver is busy: failed to transmit acknowledgement
18:05:56   2 driver is busy: failed to transmit acknowledgement
18:05:57   2 driver is busy: failed to transmit acknowledgement
18:05:57   2 driver is busy: failed to transmit acknowledgement
18:05:58   2 driver is busy: failed to transmit acknowledgement
 18:05:59   2 driver is busy: failed to transmit proberesponse
18:05:59   2 167ca3e141fd f0b4d214b2b2 ************** [PROBERESPONSE]
18:06:00   3 ffffffffffff 14007df60534 ERIN [BEACON]
18:06:01   3 aa36d1ed2179 bcf88b6253bc **************** [PROBERESPONSE]
18:06:01   3 driver is busy: failed to transmit proberesponse
18:06:02   3 driver is busy: failed to transmit proberesponse
18:06:02   3 driver is busy: failed to transmit proberesponse
18:06:03   3 driver is busy: failed to transmit proberesponse
18:06:03   3 driver is busy: failed to transmit proberesponse
18:06:04   4 driver is busy: failed to transmit proberesponse
18:06:04   4 driver is busy: failed to transmit proberesponse
18:06:05   4 driver is busy: failed to transmit reassociationrequest
18:06:05   4 driver is busy: failed to transmit deauthentication
`

ZerBea, 2023-11-27:
The mt7921u driver is broken:
https://github.com/openwrt/mt76/issues/839
https://github.com/morrownr/USB-WiFi/discussions/260#discussioncomment-7567057
We have to wait for a fix. There is nothing I can do.

BTW:
hcxdumptool is v6.2.4 is outdated.
Please use v 6.3.2 or latest git head.

ZerBea, 2023-11-27:
We have to wait for a fix. There is nothing I can do.

Please report driver problems
https://github.com/openwrt/mt76/issues/839

Closed this report, because it is related to the driver.

closed 2023-11-27 by ZerBea

ZerBea, 2023-12-08:
Please update to latest git head of hcxdumptool. I pushed a workaround:
https://github.com/ZerBea/hcxdumptool/commit/8d3f24e5a10ebdcc75211ae9214ee30ff9e4b517

Now it should work (until the driver get a fix):
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7567045

karimHI, 2023-12-08:
it's working tnx but we can no longer target APs on this latest update or what ?