ZerBea/hcxdumptool #377: Inconsistencies in (de?)referencing buffers for fd_socket_tx writes (send_80211_* functions)

roboman2444, 2023-11-30:
Some of the send_80211_* writes use & (address-of), and some of them don't.

For example:
In send_80211_associationrequest it uses "&wltxbuffer"
https://github.com/ZerBea/hcxdumptool/blob/d45799b6c81e60824fa68e0952623ce6ef2cd9d1/hcxdumptool.c#L1146

but in send_80211_eap_request_id (right below), it does not. (just "wltxbuffer")
https://github.com/ZerBea/hcxdumptool/blob/d45799b6c81e60824fa68e0952623ce6ef2cd9d1/hcxdumptool.c#L1169

I'm pretty sure these all should NOT be using &, but I am not 100% sure on that. Changing them does not create any compiler warnings, and since it is just reads, valgrind or other tools would not throw any errors.

Either way, the inconsistency in them seems like an issue to me.

```console
~/hcxdumptool$ grep 'write(fd_socket_tx' ./hcxdumptool.c 
if((write(fd_socket_tx, &wltxnoackbuffer, ii)) == ii) return;
if((write(fd_socket_tx, &wltxbuffer, ii)) == ii) return;
if(write(fd_socket_tx, wltxbuffer, ii) == ii) return;
if(write(fd_socket_tx, wltxbuffer, ii) == ii) return;
if(write(fd_socket_tx, &wltxbuffer, ii) == ii) return;
if(write(fd_socket_tx, &wltxbuffer, ii) == ii) return;
if((write(fd_socket_tx, &wltxbuffer, RTHTX_SIZE + MAC_SIZE_NORM + AUTHENTICATIONRESPONSE_SIZE)) == RTHTX_SIZE + MAC_SIZE_NORM + AUTHENTICATIONRESPONSE_SIZE) return;
if((write(fd_socket_tx, &wltxnoackbuffer, ii)) == ii) return;
if((write(fd_socket_tx, &wltxnoackbuffer, RTHTXNOACK_SIZE + MAC_SIZE_NORM + AUTHENTICATIONREQUEST_SIZE)) == RTHTXNOACK_SIZE + MAC_SIZE_NORM + AUTHENTICATIONREQUEST_SIZE) return;
if((write(fd_socket_tx, &wltxbuffer, RTHTX_SIZE + MAC_SIZE_NORM + AUTHENTICATIONREQUEST_SIZE)) == RTHTX_SIZE + MAC_SIZE_NORM + AUTHENTICATIONREQUEST_SIZE) return;
if((write(fd_socket_tx, &wltxnoackbuffer, ii)) == ii) return;
if((write(fd_socket_tx, &wltxnoackbuffer, ii)) == ii) return;
if((write(fd_socket_tx, &wltxnoackbuffer, RTHTXNOACK_SIZE + MAC_SIZE_NORM + PROBEREQUEST_UNDIRECTED_SIZE)) == RTHTXNOACK_SIZE + MAC_SIZE_NORM + PROBEREQUEST_UNDIRECTED_SIZE) return;
if((write(fd_socket_tx, &wltxnoackbuffer, RTHTXNOACK_SIZE + MAC_SIZE_NORM + 2)) == RTHTXNOACK_SIZE + MAC_SIZE_NORM +2) return;
if((write(fd_socket_tx, &wltxnoackbuffer, RTHTXNOACK_SIZE + MAC_SIZE_NORM + 2)) == RTHTXNOACK_SIZE + MAC_SIZE_NORM +2) return;
if((write(fd_socket_tx, &wltxnoackbuffer, RTHTXNOACK_SIZE + MAC_SIZE_NORM + 2)) ==RTHTXNOACK_SIZE + MAC_SIZE_NORM +2) return;
if((write(fd_socket_tx, &wltxnoackbuffer, RTHTXNOACK_SIZE + MAC_SIZE_NORM + 2)) == RTHTXNOACK_SIZE + MAC_SIZE_NORM +2) return;

```

roboman2444, 2023-11-30:
It looks like these were part of a big change 7 months ago ( [311e093](https://github.com/ZerBea/hcxdumptool/commit/311e09325d70a0774d25f687b285c3af26ca7519) ) and would explain my (perceived) change in hcxdumptool attack performance around that time. If it was sending "garbage" ( just data/pointers in the GOT? ) instead of actual packet data, that would probably massively impact things.

ZerBea, 2023-11-30:
You're right. Thanks.
I'll fix that.

ZerBea, 2023-11-30:
Ugly (nasty) copy and paste error.
Should be fixed, now.

closed 2023-11-30 by ZerBea

roboman2444, 2023-11-30:
I ended up doing some tests to see which was the right one... & or not. Both seem to have the same behavior and output (strace to verify). I ended up comparing the output bytecode... and the compiler appears to be smart and results in the same exact code for both situations(&wltxbuffer and wltxbuffer). I would not have expected that.

Anyway, the same issue also appears on some of the memcpys before the writes. For example in send_80211_authenticationresponse ( https://github.com/ZerBea/hcxdumptool/blob/90a1e3a7cf628e042bdc7b291437ad664e48151a/hcxdumptool.c#L1267 ) there isn't an &, but in send_80211_associationresponse ( https://github.com/ZerBea/hcxdumptool/blob/90a1e3a7cf628e042bdc7b291437ad664e48151a/hcxdumptool.c#L1248 ) there is.

ZerBea, 2023-12-01:
Looks like gcc is a way too smart.
But anyway. that need a fix. I'll check that.

ZerBea, 2023-12-01:
Should be fixed by latest commit, now.