ZerBea/hcxdumptool #378: Some problems

mamob07, 2023-12-03:
I'm trying to use the following command:
```
sudo hcxdumptool -i eth0 -w capture.pcapng --active_beacon --enable_status=1
```
but `--active_beacon` and `--enable_status=1` options are unrecognized. Then I removed them but I got a new error:
```
Requesting interface capabilities. This may take some time.
Please be patient...

failed to get interface list

3 errors during runtime

bye-bye
```
How can I fix it? Thanks.

ZerBea, 2023-12-03:
hcxdumtool does not work on wired interfaces like eth0. You must select a WiFi interface.

"--active_beacon" and  "--enable_status" are deprecated and hey have been removed.
Starting with the release v6.3.0 (05.05.2023), hcxdumptool uses different options.
In other words, the syntax has changed.
Please see the built-in help.
Just type "hcxdumptool -h" and press Enter.
Or type "hcxdumptool --help" to get more information like a typical command line:
```
 $ hcxdumptool -i INTERFACENAME -w dumpfile.pcapng -F --rds=1
   i     : name of the interface to be used
   w     : name of file to which packets are written
   F     : use all available channels
   rds=1 : sort real time display by status (last PMKID/EAPOL on top)
```

ZerBea, 2023-12-03:
Looks like the WiFi adapter is not visible for KALI:
https://unix.stackexchange.com/questions/626328/are-usb-wi-fi-adapters-compatible-with-kali-in-wsl2

ZerBea, 2023-12-04:
Closed this report, because it is related to WSL and/or KALI in combination with WSL.

Please notice (from README.md):
`Unsupported: Windows OS, macOS, Android, emulators or wrappers!`
and WSL is Windows.

closed 2023-12-04 by ZerBea