ZerBea/hcxdumptool #391: hcxdumptool: invalid option -- 'o'

alqisar221, 2023-12-31:
Please help. I installed the hcxdumptool tool and it shows errors. Please help
---------------------------------------------------------------------------
hcxdumptool -i wlan0  -o dumpfile.pcapng -active_beacon -enable_status=15 
hcxdumptool: invalid option -- 'o'
hcxdumptool 6.3.2-142-gf977403 (C) 2023 by ZeroBeat
This is a penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!

usage:
 $ hcxdumptool -h for an overview of all options
 $ hcxdumptool --help for an example and trouble shooting

gemesa, 2024-01-01:
i think you copied this command from [this old reddit post](https://www.reddit.com/r/HowToHack/comments/ye16n5/hcxdumptool_when_do_i_have_enough_info_to_stop/)

`hcxdumptool` has changed a lot since then, check `hcxdumptool --help` to see what options you can use (`-o`, `--active_beacon` and `--enable_status` have all been removed/renamed)

alqisar221, 2024-01-01:
I did not copy anything from anyone, but this happened to me. Can you write me the complete code? 
@gemesa

hyekalhitech, 2024-01-02:
> I did not copy anything from anyone, but this happened to me. Can you write me the complete code? @gemesa

The one you use is the old command. Use 

`hcxdumptool --help`

for new command and option list,

ZerBea, 2024-01-03:
First step after an update should always be to take a look the the change log:
```
03.05.2023
==========
hcxdumptool: changed/added several options:
```

Second step should be to take a look at the short help "-h" and for additional options "--help"
```
$ hcxdumptool -h
hcxdumptool 6.3.2-142-gf977403  (C) 2023 ZeroBeat
usage: hcxdumptool <options>

most common options:
--------------------
...
-w <outfile>   : write packets to a pcapng-format file named <outfile>
                  existing dump file file will be overwritten
                  default outfile name: yyyyddmmhhmmss-interfacename.pcapng
                  existing dump file file will not be overwritten
                  get more information: https://pcapng.com/
...
```

The FAQ are helpful, too:
https://github.com/ZerBea/hcxdumptool/discussions/330

Closed this here, because it is already answered.

ZerBea, 2024-01-03:
Closed this here, because it is already answered.

closed 2024-01-03 by ZerBea