ZerBea/hcxdumptool #393: HCXDumpTool runtime error on MediaTEK MT7922 / MT7921E

jjbyrnes29, 2024-01-04:
[image]

**Describe the bug**
HCXDumpTool runs, recognizes the wlan card, does not populate SSIDs, crashes with runtime errors and saves a pcapng file. 

**To Reproduce**
> sudo hcxdumptool 

**Expected behavior**
Lack of runtime errors causing crashing / population of SSIDs of access points. 

**Screenshots**
Output of **sudo hcxdumptool**:
[image]
[image]

Wireshark packet dump analysis of *.pcapng file:
[image]

Output of **inxi -Fxa** for wireless card. 
[image]

Test of Packet Injection:
[image]
[image]

hyekalhitech, 2024-01-04:
what kind of network adapter is this?

jjbyrnes29, 2024-01-04:
It's the internal NIC of my laptop

jjbyrnes29, 2024-01-04:
It's the internal NIC of my laptop.
I accidentally fat fingered that close with comment button.

hyekalhitech, 2024-01-04:
> It's the internal NIC of my laptop. I accidentally fat fingered that close with comment button.

try with the external network adapter. Seems like something is wrong with your MTK network adapter driver that can possibly make the devices not fully utilize by hcxdumptool.

ZerBea, 2024-01-04:
The driver is broken. Active monitor mode capability is reported by the driver, but not supported as expected.
There are several bug reports around this problem, e.g:
https://github.com/openwrt/mt76/issues/839

There is a workaround on hcxdumptool since 08.12.2023 and hcxdumptool does not trust this report any longer.
Active monitor mode must now be confirmed by -A option.
```
08.12.2023
==========
Some drivers report incorrectly active monitor mode capability.
hcxdumptool: do not set active monitor mode by default if driver report that it is possible
             removed option -p
             added new option -A in case active monitor mode is working as expected
             -A             : ACK incoming frames
                               INTERFACE must support active monitor mode
```

ZerBea, 2024-01-04:
Please udate to latest git head and run hcxdumptool without -A option, even it the driver report that active monitor mode is possible. That is definitely not the case.

ZerBea, 2024-01-04:
BTW:
airodump-ng and aireplay-ng are working, because the do not use active monitor mode.

ZerBea, 2024-01-04:
Your second scrrenshot is a nice example why hcxdumptool uses either frequency input (2412)  or channel input with additional band information (13a).
Since WiFi6 channel numbers are not longer unique:
1 = 2412
1 = 5955

ZerBea, 2024-01-04:
If you set active monitor mode by iw, aireplay-ng should crash the driver, too.

jjbyrnes29, 2024-01-04:
Running build: hcxdumptool 6.3.2-157-gc228432

Manually setting monitor mode via
> iw dev wlan0 set type monitor
> iw dev wlan0 set monitor active
Or
> iwconfig wlan0 mode monitor
Causes no changes towards hcxdumptool.

ZerBea, 2024-01-04:
That will always crach the driver:
"iw dev wlan0 set monitor active"

ZerBea, 2024-01-04:
Once it has been crashed, you must reload the driver or restart the system.
Than run hcxdumptool without -A option.
Do not use airmon-ng or iw.

jjbyrnes29, 2024-01-04:
Suprisingly enough, after reloading the driver and manually setting it via
> iw def wlan0 set type monitor

And not appending -A, program works flawlessly and I'm seeing output.

ZerBea, 2024-01-04:
There is no need to set monitor mode by iw, because hcxdumptool will do that via NETLINK messages.

ZerBea, 2024-01-04:
It looks like active monitor mode is broken on the entire mt79xx driver series.

jjbyrnes29, 2024-01-04:
Well, I can also confirm no need for the iw command now. Thanks a ton.

ZerBea, 2024-01-04:
Glad to hear this. Unfortunately, we can't use -A feature on this driver until it received a fix.
Closed as solved.

closed 2024-01-04 by ZerBea

jjbyrnes29, 2024-01-04:
Wait a second, letting it run for a while causes runtime errors even though I'm seeing APs

ZerBea, 2024-01-04:
This runtime errors are caused by the driver and hcxdumptool's internal ERROR COUNTER will increase.
That also happens if another process tries to get access to the INTERFACE while hcxdumptool is running.

ZerBea, 2024-01-04:
I guess you're running a notebook. In that case make sure that the OS does not enable power save while hcxdumptool is running.

ZerBea, 2024-01-04:
hcxdumptool treis to set power save off:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L3373
It looks like this is ignored by the driver or overwritten by KALI.

ZerBea, 2024-01-04:
By latest commit, I have improved the status. If the internal ERROR count is > 0 this additional information is shown:
```
xx ERROR(s) during runtime
Possible reasons:
 driver is broken
 driver is busy (misconfigured system, other services access the INTERFACE)
```