ZerBea/hcxdumptool #394: HCXDumpTool & MediaTEK MT7922 / MT7921E (Part 2)

jjbyrnes29, 2024-01-04:
I'd like to preface this by saying that I deeply apologize for opening another bug report on the same topic, I just found this out as soon as we closed the other thread. 

Every instance of running the current build (6.3.2-157-gc228432) leads towards:

[image]

closed 2024-01-04 by jjbyrnes29

ZerBea, 2024-01-04:
By latest commit, I have improved the status. If the internal ERROR count is > 0 this additional information is shown:
```
xx ERROR(s) during runtime
Possible reasons:
 driver is broken
 driver is busy (misconfigured system, other services access the INTERFACE)
```

I'll add some debug information to hcxdumptool to narrow down why the driver crashes.

ZerBea, 2024-01-04:
By this commit I have added a full debug mode:
https://github.com/ZerBea/hcxdumptool/commit/028c747c46b4e54ccd56e174e5a114821331ee1f
Please do a git pull to fetch this commit
Edit Makefile
from
```
# uncomment to enable DEBUG log
#DEFS		+= -DHCXDEBUG
```
to
```
# uncomment to enable DEBUG log
DEFS		+= -DHCXDEBUG
```
run
```
$ make clean
$ make
and to make sure hcxdumtool is running in debug mode:
$ ./hcxdumptool -v
hcxdumptool 6.3.2-160-g028c747 (C) 2024 ZeroBeat
running on Linux kernel 6.6.9-arch1-1
running GNU libc version 2.38
compiled by gcc 13.2.1
compiled with Linux API headers 6.4.0
compiled with GNU libc headers 2.38
enabled REALTIME DISPLAY
enabled GPS support
disabled BPF compiler
running in debug mode
```
And final:
$ sudo hcxdumptool ....... your command line
After hcxdumptool has been terminated due to ERRORs, please attach hcxdumptool.log

ZerBea, 2024-01-04:
Please also take a look at dmesg.
E.g. I have a hardware broken interface. If connected to my notebook, hcxdumptool stops working after awhile  (when the device is getting hot)
```
send_80211_beacon failed: No such device or address
send_80211_beacon failed: No such device or address
send_80211_beacon failed: No such device or address
```
and dmesg showing this ERRORs:
```
$ sudo dmesg | grep mt7
[90375.205547] mt76x0u 3-2:1.0: MAC error detected
[90375.424639] mt76x0u 3-2:1.0: MAC stop failed
```