ZerBea/hcxdumptool #413: Add a gpiowait.svg alongside gpiowait.odg

vertexgamer, 2024-01-30:
I would like to propose the suggestion of converting gpiowait.odg to svg and adding it alongside the original, as .svg is a widely recognized vector format.
Otherwise it would be useful adding a note in readme.md pointing towards .odg being a vector file format

ZerBea, 2024-01-30:
The format odg is very common on Linux and several tools can open or convert it:
LibrOffice Draw
MuPDF
ImageMagick (convert gpiowait.odg gpiowait.pdf or convert gpiowait.odg gpiowait.jpg or convert gpiowait.odg gpiowait.png)

I'll add a note to README.md

ZerBea, 2024-01-31:
Done by this commit:
https://github.com/ZerBea/hcxdumptool/commit/21f7eaaf03fd6e7bedd66d2e3192b0c270755397

closed 2024-01-31 by ZerBea