ZerBea/hcxdumptool #414: Need add stop work prompt

LLH-l, 2024-01-31:
hcxdumptool attack default value : --attemptclientmax=10   --attemptapmax=4
run  . . .
hcxdumptool -i wlan0  -w cap.pcapng  -F --rds=1
hcxdumptool  attack default value completed, need prompt: Exit  stop work

```
CHA   LAST   R 1 3 P S    MAC-AP    ESSID (last EAPOL on top)    SCAN-FREQUENCY:   2412
-----------------------------------------------------------------------------------------
  1 11:36:06  + +      +  ffffffffffff test

 
   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
 11:35:14     ffffffffffff 74daffffffff test

 Exit  stop work .  .  .
```

This prompt includes using custom default values
hcxdumptool -i wlan0 -c 1a  -w cap.pcapng  -F --rds=1 --attemptclientmax=50   --attemptapmax=50

ZerBea, 2024-01-31:
Please take a look at the R column:
As long as there is a "+", the target is in range and under attack.

You know your command line options:
`--attemptclientmax=50 --attemptapmax=50`

If the "+" in the R column has disappeared you have to change your options.

It is not a good idea to terminate hcxdumptool if this attack fails. If a new CLIENT connect to the target (or the CLIENT change its MAC), the attack starts again.

closed 2024-01-31 by ZerBea

LLH-l, 2024-01-31:
Run 30min
sudo hcxdumptool -i wlan0 -c 1a,6a,11a -w cap.pcapng --rds=1 --bpf=attack.bpf
[image]

Using the same system device and wireless card
Use airodump + aireplay in 10s capture to them
[image]
[image]
[image]

Should try to reproduce the problem I mentioned
Many people have this phenomenon, not just me !

ZerBea, 2024-01-31:
Please comment the BPF code you have used to build the filter.

LLH-l, 2024-01-31:
> Please comment the BPF code you have used to build the filter.

Test each type
```
sudo tcpdump -i wlan0  wlan addr3  8ca6dfcf4a40  or wlan addr3 94e4ba32a23c or wlan addr3 1ca6811ac33c or wlan addr3 ffffffffffff   -ddd  > attack.bpf
or
sudo tcpdump -s 1024 -y IEEE802_11_RADIO -i wlan0  wlan addr3  8ca6dfcf4a40  or wlan addr3 94e4ba32a23c or wlan addr3 1ca6811ac33c or wlan addr3 ffffffffffff   -ddd  > attack.bpf
or
sudo tcpdump -s 1024 -y IEEE802_11_RADIO -i wlan0 wlan addr1  70039f9a1db8  or wlan addr2 70039f9a1db8 or wlan addr3 1ca6811ac33c or wlan addr3 ffffffffffff   -ddd  > attack.bpf
or
sudo tcpdump -i wlan0 wlan addr1  70039f9a1db8  or wlan addr2 70039f9a1db8 or wlan addr3 1ca6811ac33c or wlan addr3 ffffffffffff   -ddd  > attack.bpf
```

If your suspect that BPF is incorrect
Please write a correct BPF, my test it  !

LLH-l, 2024-01-31:
Your, should try to reproduce the problem I mentioned

Run command test
scan
Create BPF
sudo hcxdumptool -i wlan0 -c 1a -w cap.pcapng --rds=1 --bpf=attack.bpf
After launching an attack, my observe the client reconnecting to the AP (But only once),
I observe chart not show a captured handshake data, I also checked the pcapng file, no m1m2(at least)

```
CHA   LAST   R 1 3 P S    MAC-AP    ESSID (last EAPOL on top)    SCAN-FREQUENCY:   2412
-----------------------------------------------------------------------------------------
   1 11:36:06 + +       + ffffffffffff test

   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
 11:35:14     ffffffffffff 74daffffffff test
```

The problem is that after launching an attack, hcxdumptool seems never launched another attack, my observed the client no reconnection behavior ,causing no capture the handshake.
the continued attack behavior ineffective ? or auto stop attack ?
After launching one attack, the attack will auto stop ?
AP and client in my 2m

ZerBea, 2024-01-31:
$ hcxdumptool default scan list (as mentioned in -h) is channel 1, channel 6 and channel 11.
There is no need to add this to your command line:

Please don't turn a thousand screws at the same time.
For this test we use: 74daffffffff
Replace 74daffffffff with the correct MAC and double check that is the target MAC)

create filter:
`$ tcpdump -s 1024 -y IEEE802_11_RADIO wlan addr3 74daffffffff or wlan addr3 ffffffffffff -ddd > 74daffffffff.bpf`

run hcxdumptool:
`$ sudo hcxdumptool -i wlan0 --rds=1 -w 74daffffffff.pcapng --bpf=74daffffffff.bpf -t 10 --exitoneapol=7`

Wait until hcxdumptool termiates.
Important note: hcxdumptool will terminate straight after an EAPOL MESSAGE PAIR has been received. It doesn't wait for the status display.

`Your should try to reproduce the problem I mentioned`
I already did this sevral times (last one yesterday);
https://github.com/ZerBea/hcxdumptool/discussions/412#discussion-6154287

LLH-l, 2024-01-31:
Once again, it has nothing to do with BPF, this is hcxdumptool a attack issue
Even run following command, it still cannot be captured
sudo hcxdumptool -i wlan0  -w cap.pcapng --rds=1
[image]

You should review the attack code and marking rule code
Features:
The first attempt to launch a re association attack is effective, but regardless of whether it is captured or not, it will no longer launch a second re association attack, I discovered this behavior by observing the client

Using the same system device and wireless card
Use airodump + aireplay in 5s capture to them
[image]

ZerBea, 2024-01-31:
Please give me a couple of hours to code a test version to figure out why it doesn't work running your environment.

ZerBea, 2024-01-31:
$ git clone https://github.com/ZerBea/wifi_laboratory
$ cd wifi_laboratory
$ make
ignore all warnings
$ sudo ./hcxlabtool -i wlan0
do you see M1M2, M1M2RGUE or M1M2M3 or PMKID messages?

LLH-l, 2024-02-01:
I have seen this situation from other communities, and at that time, I didn't believe they said. . .
In order to confirm they said,in the past few days,  I have been try to use hcxdumptool capture test, Indeed, some signals capture fail
As for reason, I not know, time not allows me conduct more test

If need reproduce this problem:
if using hcxdumptool cannot captured some signals
Can try use airdump+airreplay or AngryOxide to test capture it, your may find this issues

ZerBea, 2024-02-01:
If you take a look at the issue reports and the discussions you'll see that 90% came from KALI users.
KALI is not (absolutely not) a distribution for a beginner!
https://www.kali.org/docs/introduction/should-i-use-kali-linux/

It is imperative that the requirements (mentioned in REAFME.md) are met.
The farther away you are from this requirements, the less hcxdumptool/hcxtools works.

I've done several tests and every test has been successful. I've coded an entire new attack engine and the results are the same.

Check your environment and make sure that the requirements are met!

And no, I do not use other tools like airdump+airreplay or AngryOxid.

LLH-l, 2024-02-01:
> mentioned in REAFME.md
All meet the conditions   mentioned in REAFME.md
If not meet the conditions, it cannot capture one handshake. Now it captures at least one handshake, indicating that all conditions are met

If doubting filters , doubting various conditions, doubting commands, etc.  . .  are all unrealistic,

ZerBea, 2024-02-01:
_All meet the conditions mentioned in REAFME.md_

Great. Let#s draw a conclusion:

You're running Arch Linux or Raspbian Lite

The Kernel is one of these ones mentioned here:
https://www.kernel.org/

The driver is part of the stock Linux kernel.

You system is well configured and all services that are not mandatory to run hcxdumptool are stopped.

Do you see M1M2, M1M2RGUE or M1M2M3 or PMKID messages?
```
$ git clone https://github.com/ZerBea/wifi_laboratory
$ cd wifi_laboratory
$ make
ignore all warnings
$ sudo ./hcxlabtool -i wlan0
```

LLH-l, 2024-02-01:
> 您是否看到 M1M2、M1M2RGUE 或 M1M2M3 或 PMKID 消息？
Yes, I see  m1m2
time sudo ./hcxlabtool -c10a --exitoneapol=7

LLH-l, 2024-02-01:
test 1a 6a 11a
I remember exit without any error prompts

ZerBea, 2024-02-01:
The new test version print a status after exit.
Please comment this status.
I can't help you if you don't follow these instructions.

LLH-l, 2024-02-01:
> It is your system that cause problems and we have to figure out why this happens

Please don't doubt me anymore
There is no need doubt, run hcxdumptool the is no errn prompt
Please attention the capture MAC

Run use command
the capture results each  different ( for each re attack )
sudo hcxdumptool - i wlan0- w a. pcapng -- rds=1
[image]

Run....
sudo hcxdumptool - i wlan0- w a. pcapng -- rds=1
[image]

Use filter  command test . . . . 
the capture results each  different ( for each re attack )
sudo hcxdumptool - i wlan0- w a. pcapng -- rds=1-- bpf=attach. bpf
[image]

sudo hcxdumptool - i wlan0- w a. pcapng -- rds=1-- bpf=attach. bpf
[image]

sudo hcxdumptool - i wlan0- w a. pcapng -- rds=1-- bpf=attach. bpf
[image]

ZerBea, 2024-02-01:
I don't doubt, but I can't help if you if turn several screws at the same time.
I requested the result of hcxlabtool:
https://github.com/ZerBea/hcxdumptool/issues/414#issuecomment-1919158071

But you gave me the result of hcxdumptool.
Please also take a look at the real time display of your screenshots:
First:
5cbb has been successful attacked, M1M2M3 has been captured, R column is empty -> the attack has been finished

Second:
9897 and 1ca6 have been successfull attacked, M1M2M3, R column is not empty -> the attack is still in progress, because there are some CLIENTs from which we can get additional information

Third:
94e4 has been successful attacked, M1M2M3 has been captured, R column is empty -> the attack has been finished

Fourth:
30fc0 has been successful attacked, M1M2M3 has been captured, R column is empty -> the attack has been finished
Additional hcxdumptool got an EAPOL M1M2ROGUE because the CLIENT connected to hcxdumptool

All the results are different, because hcxdumptool run several different attack modes to get a MESSAGE PAIR.
All attacks have been successful.

The results depend on:
Does the AP receive hcxdumptool? ([Why is the RSSI of the attack adapter completely useless:](https://github.com/ZerBea/hcxdumptool/discussions/406) 
Does the CLIENT receive hcxdumptool?
Is the channel busy or free?

ZerBea, 2024-02-01:
Please note:
If one of the targets is e.g. a mobile phone and if this phone moves a few centimeters, the conditions on the channel change!
It is possible that it doesn't receive hcxdumptool at the new place.

LLH-l, 2024-02-01:
> I requested the result of hcxlabtool:
But you gave me the result of hcxdumptool.

[image]

ZerBea, 2024-02-01:
Thanks. That's looking good.
I don't see any problems related to hcxlabtool/hcxdumptool.
You can use both versions and compare the results to each other.

Please take a closer look at line 7.
Exactly that is the wanted behavior
```
7460 <-> d4ee M1M2ROGUE
7460 <-> d4ee M1M2
```
The response from hcxlabtool (M1M2ROGUE) has been much faster as the response from the AP.
To get the best option values (for your environment) I recommend to play with all options an filters (on both tools).
There is no(!) standard value  that will work for all users.

The difference between hcxlabtool and hcxdumptool:
```
all CPU cycle intensive functions have been removed.
the PMKID is not implemented, yet
the deauthentication attack has been removed in favor of a disassociation attack
```

rcascan has been removed, because "nmcli dev wifi" can do this better.

And you are absolutely right:
hcxdumptool's attack behavior has become toothless to to too many features and the real time display.

LLH-l, 2024-02-02:
> I don't see any problems related to hcxlabtool/hcxdumptool.

Please see
[image]
This is my AP ! Within my 5m range
Why hcxdumptool cannot capture it
If attack is by R column "+" decide  , it is unreliable. (measuring distance not accurate)

Why can airodump+airreplay easily capture him
This is problem core,  and the direction for improvement

There also one issues
No scan found for my AP MAC: d4ee0744ffff
[image]

ZerBea, 2024-02-02:
First screenshot:
hcxdumptool does not measure the distance to the target.
Instead it asks the target "Hello, how are you, can you hear me"?
When the target answers  "Hello, I'm fine and I hear you good", the attack starts and a plus appears.
There are at least seven APs on this channels and a lot of CLIENTs which interfere with each other.

Second screenshot:
By default hcxdumptool scans channels 1,6,11 only. If your AP is not on one of these channels, it does not appear in the real time display. 

But I can see something that let me wonder:
I can see d4ee0744aa66 as AP and (first screen shot) and d4ee0744.... as CLIENT ( second screen shot).
Do you use a repeater? Or do you run aircrack in parallel to hcxdumptool?

For both cases, please comment your command line and the output of the command line. And the interface driver (hcxdumptool option -l (lower case of L)

ZerBea, 2024-02-02:
The best way to figure out why it doesn't work is to analyze the traffic - run tshark in parallel.
Set your AP to a less crowed channel, e.g. channel 4.
Set hcxdumptool to channel 4.
Run tshark and monitor what's going on.

I can't do this from here, because O don't see the traffic on your operating channel.
A hcxdumptool pcapng is not useful, because it only reflect what hcxdumptool sees.

ZerBea, 2024-02-02:
On such crowded channels it is mandatory to increase the scan time >= 15 seconds (-t) or to set the operating channel (-c).

ZerBea, 2024-02-02:
Please note that my real time display looks different from yours.
It shows only one entry: The filtered target AP and the filtered target CLINET(s) from my laboratory environment.
Everything else is against the law!
Misuse of hcxdumptool within a network, particularly without authorization, may cause irreparable damage and result in significant consequences. “Not understanding what you were doing” is not going to work as an excuse.

LLH-l, 2024-02-02:
Right.  AP 9897cc in my 15m within scope, this is also my AP
AP d4ee07 in my 5m within scope, this is also my AP
Not running simultaneously with airodmp and aireplay

Sorry
Due to information leakage,I need to delete all our conversations

ZerBea, 2024-02-02:
A good idea to delete all screenshots. It is not mandatory to have them to understand the basics of hcxdumptool.

I recommend to set up a laboratory environment:
```
own AP as target AP
simple PSK (12345678)
fixed channel (less crowded)
one or two own CLIENTs
```

The workflow:
```
run tshark/Wireshark to monitor what's going on
run hcxdumptool to attack your target NETWORK
take a look at tshark/Wireshark to check the impact of the options on your target NETWORK
repeat this until you get the best options
```

ZerBea, 2024-02-02:
AAAAAAxxxxxx (AAAAAA is the OUI and xxxxxx is the NIC of the MAC) is for comments here on git to keep privacy
High level BPF syntax (provided by tcpdump) does not support this.

If you would like to code such a BPF (filer by OUI) you need low level assembler code.

Example for MAC 112233445566

high level syntax (user friendly, simple and easy to use if you are familiar with tshark, Wireshark or tcpdump):
`"wlan addr3 112233445566`

c style syntax (good knowledge in programming with C):
```
{ 0x30, 0, 0, 0x00000003 },
{ 0x64, 0, 0, 0x00000008 },
{ 0x7, 0, 0, 0x00000000 },
{ 0x30, 0, 0, 0x00000002 },
{ 0x4c, 0, 0, 0x00000000 },
{ 0x2, 0, 0, 0x00000000 },
{ 0x7, 0, 0, 0x00000000 },
{ 0x50, 0, 0, 0x00000000 },
{ 0x54, 0, 0, 0x0000000c },
{ 0x15, 5, 0, 0x00000004 },
{ 0x40, 0, 0, 0x00000012 },
{ 0x15, 0, 3, 0x33445566 },
{ 0x48, 0, 0, 0x00000010 },
{ 0x15, 0, 1, 0x00001122 },
{ 0x6, 0, 0, 0x00000400 },
{ 0x6, 0, 0, 0x00000000 },
```

low level assembler code (extensive knowledge in programming with ASM):
```
(000) ldb      [3]
(001) lsh      #8
(002) tax      
(003) ldb      [2]
(004) or       x
(005) st       M[0]
(006) tax      
(007) ldb      [x + 0]
(008) and      #0xc
(009) jeq      #0x4             jt 15	jf 10
(010) ld       [x + 18]
(011) jeq      #0x33445566      jt 12	jf 15
(012) ldh      [x + 16]
(013) jeq      #0x1122          jt 14	jf 15
(014) ret      #1024
(015) ret      #0
```
This is the royal class of building a BPF.

ZerBea, 2024-02-02:
Why does hcxdumptool work for me but not for you?
I've analyzed the target before I run hcxdumptool:
my test AP is slow and some of my test CLIENTs are slow, too.
It takes about 3 seconds until they complete the 4way handshake (sometimes more).
hcxdumptool's default scan interval is 5 seconds which is a way too fast for the slow targets.
A solution is to set hcxdumptool to the target channel or to increase the stay time on the channel.
Running hcxdumptool unconfigured (no BPF) it attacks everything on a channel in order  of the received responses.
If there are 5 targets on the channel and each 4way handshake take 4 seconds (20 seconds to get them all), the default stay time is not enough.
All this ensures that hcxdumptool does not work for you.
I can't (and I don't want) program functions that compensate missing knowledge.

LLH-l, 2024-02-03:
Firstly, I am a supporter of hcxdumptool,and I am a passionate supporter ( Your should know I am supportive hcxdumptool )
I all the conversations related with hcxdumptool, 
I hope he gets better and better

I tried increase this value, once again attacked the behavior, it worked
--attemptapmax=50

Use default values, sometimes the attack behavior does not work.
I found through observing the client

LLH-l, 2024-02-03:
I found the reason !
--attemptapmax=4
Reason is this default value is too small, I tried to increase it to 10
`sudo hcxdumptool -i wlan0 -c 1a  --bpf=attack.bpf  --attemptapmax=10`
Occasionally have 9  to capture handshake fail (test 20 times)

I try increasing it to 15 (test20 times)
`sudo hcxdumptool -i wlan0 -c 1a  --bpf=attack.bpf  --attemptapmax=15`
Occasionally have 5 capture handshake fail

I  try increasing it to 20 (test 20 times)
`sudo hcxdumptool -i wlan0 -c 1a  --bpf=attack.bpf  --attemptapmax=20`
Basically it can be success capture !

To avoid capture handshake fail,this optimal value should be 20 ( minimum value )

ZerBea, 2024-02-03:
Great that you got it. Now you have your best suited values:
`To avoid capture handshake fail,this optimal value should be 20 ( minimum value )`
But keep in mind: This are your(!) optimized values. 

Maybe I didn't explain it well enough:
To avoid that a WiFi channel is intentionally jammed by the default values of hcxdumptool, this values are always best case values. They are measured in laboratory tests using off-the-shelf hardware and software.

BTW:
For additional reference tests, I'm running "Professional Hi Tech Hardware" and modified drivers. This measured values are of course always by far better than the off-the-shelf hardware competition. Off-the-shelf hardware will never reach this values. 

But anyway, every other scenario requires different values which depend on various factors:
```
performance of the hardware
receiver sensitivity of the WiFi adapter
antenna
WiFi channel (free or busy)
distance to target
```
I fully agree to what you have mentioned above, If there were no options.
But hcxdumptool has options to customize that. This allows a user to adapt it to any scenario tailored to the hardware, the target and the environment!

The same applies to hcxtools (hcxpcapngtool). The default values are best case values.
If these defaults are not working for you it is mandatory to customize the tool by options.

`To avoid capture handshake fail,this optimal value should be 20 ( minimum value )`
To avoid jamming a channel by default options of hcxdumptool I do not do this.

ZerBea, 2024-02-03:
And if you think that I don't listen to your comments, than you might be wrong.
Please take a look at hcxlabtool.

ZerBea, 2024-02-03:
BTW:
If write "inexperienced user" it was not meant to be negative.
It simply means that the user does not know all these factors mentioned above.

In any case, I advise you to run tshark or Wireshark to ensure that hcxdumptool is doing what you expect.
If that is not the case, .... that's why options are for!

ZerBea, 2024-02-03:
Please test hcxlabtool.
Running --rds=1 you'll see a [counter] after the ESSID.
This counter show the remaining attacks.

LLH-l, 2024-02-04:
> Please test hcxlabtool. Running --rds=1 you'll see a [counter] after the ESSID. This counter show the remaining attacks.

```
CHA
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[9]
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[6]
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[0]
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[1]
```

Add the remaining number of attack very good. (if add to hcxdumptool)
But In the next 2 weeks, me no more time for test
Do a simple test

I don't understand it how  works, it seems have some of remaining attack number always not  changed
But have some ESSID, remaining number attack changed

LLH-l, 2024-02-04:
hcxdumptool style should add M2 at the top
R column Change be  the remaining number of attacks

```
 CHA   LAST   R 1 2 3 P S    MAC-AP    ESSID (last EAPOL on top)    SCAN-FREQUENCY:   2432
-----------------------------------------------------------------------------------------
1	      0  + + +
6	      3  + + 
6	      9  +  
```

ZerBea, 2024-02-04:
For sure it will be more intuitive, but unfortunately sorting the status display takes a lot (too much) CPU cycles.
To vie only NETWORK from which hcxdumptool got a hash, you can use rds=2

LLH-l, 2024-02-04:
```
CHA                                                                			RE ASSOCIATION
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[9]	[0]
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[6]	[2]
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[0]	[1]
 6	11:00:00	+	ff:ff:ff:ff:ff:ff	ESSID	[1]	[5]
```

ZerBea, 2024-02-04:
I'll check it.

ZerBea, 2024-02-04:
Problem:
The latest version is going full stealth.
It is not possible to distinguish between a RESPONSE from AP to CLIENT or from AP to hcxlabtool.
It is not possible to distinguish between a RESPONSE from CLIENT to AP or from CLIENT to hcxlabtool.

If acting as an AP, hcxdumptool use the MAC of the AP. If acting as a CLIENT hcxdumptool use the MAC of the CLIENT.

ZerBea, 2024-02-04:
Problem 2:
We can't add a new column, because the line length exceed the line length of an Android device.

LLH-l, 2024-02-04:
Regardless of whether it is AP or client, as long as it happens re association action will be recorded once

ZerBea, 2024-02-04:
That will not work, due to hcxdumptool's re/association attack.
We use an RE/ASSOCIATION to disconnect all connected CLIENTs.
E.g. my test target responds a hundred times (within a second) to this attack.

ZerBea, 2024-02-04:
Please focus on the tshark monitor instead of the display: Form always follows function.

First of all this basic functions must work as expected:
```
attack vector must work (get M2, get PMKID, get M1M2M3 - exactly in this order)
attack vector must not produce unnecessary traffic (if the channel is busy, we get nothing)
attack vector should not be detected by intrusion detection systems
```

ZerBea, 2024-02-04:
This is the tshark/Wireshark filter to monitor the channel while hcxdumptool is running:
`eapol  || wlan.fc.type_subtype == 0x0000  || wlan.fc.type_subtype == 0x0001  || wlan.fc.type_subtype == 0x0002 || wlan.fc.type_subtype == 0x0003  || wlan.fc.type_subtype == 0x0004 ||  wlan.fc.type_subtype == 0x000a || wlan.fc.type_subtype == 0x000b || wlan.fc.type_subtype == 0x000c || radiotap.present.dbm_antsignal == False
`

ZerBea, 2024-02-04:
If you run this filter in Wireshark you'll see a lot of packets.
hcxdumptool needs to handle every single packet and it must respond to every packet just in time. That has to be done on-the-fly. We are talking about nanoseconds!

I think you can imagine that there is not much time left to present a beautiful and well sorted status display.

ZerBea, 2024-02-04:
While it (default values) doesn't work for you, it works well for others:
https://github.com/ZerBea/hcxdumptool/issues/421#issuecomment-1925745685

LLH-l, 2024-02-15:
Hi..one week later , I'm back
Use wireshark observation captured to m2m3,  But no report
If only captured m3m4 (m4 not reset to zero), how report ?

ZerBea, 2024-02-15:
Please attach both dump files:
hcxpcapngtool pcapng
and
Wireshark pcapng

LLH-l, 2024-02-16:
> Please attach both dump files: hcxpcapngtool pcapng and Wireshark pcapng

I forgot to correct wireshark the time zone, but the experimental file is real
[20240215222800-wlan0.zip](https://github.com/ZerBea/hcxdumptool/files/14303851/20240215222800-wlan0.zip)

LLH-l, 2024-02-16:
Reported for m1m2,   m2m3 no Report ?

Or there should at least one report
```
   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
 11:35:14    m23 ffffffffffff 74daffffffff test

```

LLH-l, 2024-02-16:
Wireshark displayed capture of m1m2, but hcxdumptool did not report
[image]

Wireshark displayed capture of m2m3, but hcxdumptool did not report
[image]

LLH-l, 2024-02-16:
Incomplete messages should displayed in the bottom box style

```
CHA   LAST   R 1 3 P S    MAC-AP    ESSID (last EAPOL on top)    SCAN-FREQUENCY:   2412
-----------------------------------------------------------------------------------------
  1 11:36:06  + +      +  ffffffffffff test

 
   LAST   E 2 MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
 11:35:14   m12  ffffffffffff afffffffffff test
11:35:14    m23  ffffffffffff bfffffffffff test
```

ZerBea, 2024-02-16:
We are on the fly and we do not have the time to evaluate and to show everything!
As a direct result, the real time display is limited to the conditions which are mentioned in help
It shows a + in R column if an AP responds
It shows a +  in 1 column if we got an M1 (because this EAPOL MESSGAES can  contain a PMKID.
It shows a + in P column if the M1 contain a PMKID
It shows a + in 3 column if the handshake is authorized (M1M2M3 or M1M2M3M4)
It shows a + in S column if the AKM is one of the by hashcat or JtR supported hash modes.

A handshake is shown if these conditions met:
time between M1 and M2 20000000 nanoseconds
time between M2 and M3 20000000 nanoseconds
time between M3 and M4 20000000 nanoseconds
and
replaycount M1 == M2
replaycount M2 == M3 -1
and
ANONCE M1 == ANONCE M3
additional:
ANONCE M4 not zeroed
and
replaycount M3 == M4
replaycount M4 == M1 -1

Than and only than a plus in column 3 is show, because we got a really good 4way handshake!

I've analyzed your attached monitor dump file wireshark.pcapng from:
https://github.com/ZerBea/hcxdumptool/files/14303851/20240215222800-wlan0.zip

And none of the conditions mentioned above are met!
```
$ tshark -r wireshark.pcapng -T fields -e frame.number -e frame.time -e wlan.ra -e wlan.ta -e wlan_rsna_eapol.keydes.msgnr -e eapol.keydes.replay_counter -e wlan_rsna_eapol.keydes.nonce
1	Feb 15, 2024 23:28:06.632483785 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
2	Feb 15, 2024 23:28:06.633361620 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
3	Feb 15, 2024 23:28:06.634109620 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
4	Feb 15, 2024 23:28:06.636360186 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
5	Feb 15, 2024 23:28:06.637732916 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
6	Feb 15, 2024 23:28:06.639858954 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
7	Feb 15, 2024 23:28:06.640982631 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
8	Feb 15, 2024 23:28:06.641982968 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
9	Feb 15, 2024 23:28:06.656387593 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
10	Feb 15, 2024 23:28:06.670863665 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
11	Feb 15, 2024 23:28:06.691110584 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
12	Feb 15, 2024 23:28:06.695486280 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
13	Feb 15, 2024 23:28:06.700356592 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
14	Feb 15, 2024 23:28:06.701232751 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	1	117c20ae52c2b71adb7cd6011fd08c870502f8a329773281f24a323503636e0a
15	Feb 15, 2024 23:28:07.626233549 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
16	Feb 15, 2024 23:28:07.628435155 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
17	Feb 15, 2024 23:28:07.628473289 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
18	Feb 15, 2024 23:28:07.629487594 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
19	Feb 15, 2024 23:28:07.630355791 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
20	Feb 15, 2024 23:28:07.631481283 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
21	Feb 15, 2024 23:28:07.633354147 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
22	Feb 15, 2024 23:28:07.634479639 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
23	Feb 15, 2024 23:28:07.636856129 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
24	Feb 15, 2024 23:28:07.644480771 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
25	Feb 15, 2024 23:28:07.645853222 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
26	Feb 15, 2024 23:28:07.647228816 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
27	Feb 15, 2024 23:28:07.648354099 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
28	Feb 15, 2024 23:28:07.649479242 CET	f4:9d:a7:2d:85:88	d4:ee:07:44:1a:66	1	2	7c1de6adb21d1455a3ae693287f9d3c06978387741e2e43a4449a10f14e0d6ab
[... cut ...]
                              1 = + AP display     : got EAPOL M1 (CHALLENGE)
                              3 = + AP display     : got EAPOL M1M2M3 (AUTHORIZATION) hashcat / JtR can work on
                              E = + CLIENT display : got EAP-START MESSAGE
                              2 = + CLIENT display : got EAPOL M1M2 (ROGUE CHALLENGE) hashcat / JtR can work on
```

To mention it again.
This is the target system hcxdumptool is designed work on:

[image]

It is not a 16 core RYZEN with a 1TB SSD and a 24"" monitor!
It is a Raspberry Pi Zero (single core)!

LLH-l, 2024-02-16:
> 2 = + CLIENT display : got EAPOL M1M2 (ROGUE CHALLENGE) 

If need improve performance, why check m12 once
When AP and client list appears in the box below, they be classified as incomplete messages, they may be m12 or m23 or m34 or m14 message

```
   LAST   E  MAC-AP-ROGUE   MAC-CLIENT   ESSID (last M2ROGUE on top)
-----------------------------------------------------------------------------------------
11:35:14      0fffffffffff  afffffffffff test
11:35:14      1fffffffffff  bfffffffffff test
11:35:14      2fffffffffff  cfffffffffff test
11:35:14      3fffffffffff  dfffffffffff test
```

ZerBea, 2024-02-16:
This is related to 2 different internal ring buffers:
Buffer 1
contain 4-way handshakes between AP and its CLIENT(s).
An authorized handshake is M1M2M3 or M1M2M3M4
An attack stops if this condition is met.

Buffer 2 contain a half handshake between hcxdumptool and CLIENT(s).
We don't know the PSK, so we never get an authorized handshake.
Instead we get M1M2 only.
An attack stops if this condition is met.

This 2 conditions are shown in the real time display:
AP display:
column 3 = + =  condition M1M2M3 or M1M2M3M4 is met

CLIENT display:
column 2 = + =  condition M1M2ROGUE is met

Adding more conditions like
```
M1M2 between AP and CLIENTs,
M2M3 between AP and CLIENTs,
M3M4 between AP and CLIENTs,
M1M4 between AP and CLIENTs,
```
will take a lot of time.

LLH-l, 2024-02-17:
Captured pmkid no prompt

ZerBea, 2024-02-17:
Thanks for reporting that bug.
It should be fixed by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/aa665ad6e157560e056700173fb08d21a70a74b5
Please try latest git head.

[7 more replies]