ZerBea/hcxdumptool #421: openwrt: Issue finding interfaces

kimocoder, 2024-02-04:
Hi!

We seem to have an issue with listing adapters on OpenWRT at the momen. I attached the strace log from "hcxdumptool -i wlan0 -w capture.pcapng" cmd output/trace :+1: 

airmon-ng and everything else works fine, but hcxdumptool has issues calling  ""/sys/class/ieee80211/phy1/device/driver" it seems.

kimocoder, 2024-02-04:
`root@OpenWrt:~# strace hcxdumptool -i wlan0 -w capture.pcapng
-----------------------------------------------------------------------------------------------------------------------------------------------------------

execve("/usr/sbin/hcxdumptool", ["hcxdumptool", "-i", "wlan0", "-w", "capture.pcapng"], 0x7fd7bffb60 /* 13 vars */) = 0
set_tid_address(0x7fa77b3258)           = 5718
brk(NULL)                               = 0x3bd98000
brk(0x3bd9a000)                         = 0x3bd9a000
mmap(0x3bd98000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x3bd98000
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=131088, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 200704, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fa76e0000
mmap(0x7fa770f000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1f000) = 0x7fa770f000
close(3)                                = 0
mprotect(0x7fa770f000, 4096, PROT_READ) = 0
mprotect(0x42f000, 4096, PROT_READ)     = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 3
ioctl(3, SIOCGIFINDEX, {ifr_name="wlan0", ifr_ifindex=11}) = 0
close(3)                                = 0
getpid()                                = 5718
writev(1, [{iov_base="", iov_len=0}, {iov_base="\nRequesting physical interface c"..., iov_len=92}], 2
Requesting physical interface capabilities. This may take some time.
Please be patient...

) = 92
rt_sigprocmask(SIG_UNBLOCK, [RT_1 RT_2], NULL, 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x409b04, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fa776485c}, NULL, 8) = 0
rt_sigaction(SIGTERM, {sa_handler=0x409b04, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fa776485c}, NULL, 8) = 0
rt_sigaction(SIGTSTP, {sa_handler=0x409b04, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fa776485c}, NULL, 8) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa77aa000
mmap(NULL, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa77a5000
mmap(NULL, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa779f000
mmap(NULL, 49152, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76d4000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa779e000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa779c000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa779b000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa779a000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa7799000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa7798000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa7797000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa7796000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa7795000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76d2000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76d0000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76ce000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76cc000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76c8000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76c4000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76c0000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76bc000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76b4000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76ac000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa76a4000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa769c000
mmap(NULL, 65536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa768c000
nanosleep({tv_sec=1, tv_nsec=0}, NULL)  = 0
openat(AT_FDCWD, "/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=4, ...}) = 0
mmap(NULL, 4, PROT_READ, MAP_SHARED, 3, 0) = 0x7fa768b000
close(3)                                = 0
socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC, NETLINK_ROUTE) = 3
setsockopt(3, SOL_SOCKET, SO_SNDBUF, [4095], 4) = 0
setsockopt(3, SOL_SOCKET, SO_RCVBUF, [65535], 4) = 0
getpid()                                = 5718
bind(3, {sa_family=AF_NETLINK, nl_pid=5718, nl_groups=00000000}, 12) = 0
socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC, NETLINK_GENERIC) = 4
setsockopt(4, SOL_SOCKET, SO_SNDBUF, [4095], 4) = 0
setsockopt(4, SOL_SOCKET, SO_RCVBUF, [65535], 4) = 0
[... 276 lines cut ...]
munmap(0x7f81748000, 8192)              = 0
munmap(0x7f81744000, 16384)             = 0
munmap(0x7f81740000, 16384)             = 0
munmap(0x7f8173c000, 16384)             = 0
munmap(0x7f81738000, 16384)             = 0
munmap(0x7f81730000, 32768)             = 0
munmap(0x7f81728000, 32768)             = 0
[... cut ...]
writev(2, [{iov_base="1 ERROR(s) during runtime\n", iov_len=26}, {iov_base=NULL, iov_len=0}], 21 ERROR(s) during runtime
) = 26
writev(1, [{iov_base="", iov_len=0}, {iov_base="\n", iov_len=1}], 2
) = 1
writev(1, [{iov_base="", iov_len=0}, {iov_base="bye-bye\n\n", iov_len=9}], 2bye-bye

) = 9
exit_group(0)                           = ?
+++ exited with 0 +++

----------------------------------------------------------------------------------------------------------------------------------------------------------
root@OpenWrt:~#`

kimocoder, 2024-02-04:
`root@OpenWrt:~# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

eth1      no wireless extensions.

lan4      no wireless extensions.

lan3      no wireless extensions.

lan2      no wireless extensions.

lan1      no wireless extensions.

lan5      no wireless extensions.

br-lan    no wireless extensions.

wlan0     no wireless extensions.

root@OpenWrt:~# airmon-ng 

PHY	Interface	Driver		Chipset

phy0	wlan0		mt798x-wmac	Not pci, usb, or sdio

root@OpenWrt:~# iw wlan0 info
Interface wlan0
	ifindex 11
	wdev 0x2
	addr ca:7f:54:bd:31:ab
	type managed
	wiphy 0
	txpower 6.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.148 #0 SMP Sun Feb 4 00:12:51 2024 aarch64 GNU/Linux
root@OpenWrt:~# 
root@OpenWrt:~# `

kimocoder, 2024-02-04:
`root@OpenWrt:~# strace hcxdumptool -i wlan0mon -w capture.pcapng
execve("/usr/sbin/hcxdumptool", ["hcxdumptool", "-i", "wlan0mon", "-w", "capture.pcapng"], 0x7fcb5c59c0 /* 13 vars */) = 0
set_tid_address(0x7fbb419258)           = 6519
brk(NULL)                               = 0x3284000
brk(0x3286000)                          = 0x3286000
mmap(0x3284000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x3284000
openat(AT_FDCWD, "/etc/ld-musl-aarch64.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=131088, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 200704, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fbb346000
mmap(0x7fbb375000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1f000) = 0x7fbb375000
close(3)                                = 0
mprotect(0x7fbb375000, 4096, PROT_READ) = 0
mprotect(0x42f000, 4096, PROT_READ)     = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 3
ioctl(3, SIOCGIFINDEX, {ifr_name="wlan0mon", ifr_ifindex=12}) = 0
close(3)                                = 0
getpid()                                = 6519
writev(1, [{iov_base="", iov_len=0}, {iov_base="\nRequesting physical interface c"..., iov_len=92}], 2
Requesting physical interface capabilities. This may take some time.
Please be patient...

) = 92
rt_sigprocmask(SIG_UNBLOCK, [RT_1 RT_2], NULL, 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x409b04, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fbb3ca85c}, NULL, 8) = 0
rt_sigaction(SIGTERM, {sa_handler=0x409b04, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fbb3ca85c}, NULL, 8) = 0
rt_sigaction(SIGTSTP, {sa_handler=0x409b04, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fbb3ca85c}, NULL, 8) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb410000
mmap(NULL, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb40b000
mmap(NULL, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb405000
mmap(NULL, 49152, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb33a000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb404000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb402000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb401000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb400000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb3ff000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb3fe000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb3fd000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb3fc000
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb3fb000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb338000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb336000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb334000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb332000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb32e000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb32a000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb326000
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb322000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb31a000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb312000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb30a000
mmap(NULL, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb302000
mmap(NULL, 65536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbb2f2000
nanosleep({tv_sec=1, tv_nsec=0}, NULL)  = 0
openat(AT_FDCWD, "/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=4, ...}) = 0
mmap(NULL, 4, PROT_READ, MAP_SHARED, 3, 0) = 0x7fbb2f1000
close(3)                                = 0
socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC, NETLINK_ROUTE) = 3
setsockopt(3, SOL_SOCKET, SO_SNDBUF, [4095], 4) = 0
setsockopt(3, SOL_SOCKET, SO_RCVBUF, [65535], 4) = 0
getpid()                                = 6519
bind(3, {sa_family=AF_NETLINK, nl_pid=6519, nl_groups=00000000}, 12) = 0
socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC, NETLINK_GENERIC) = 4
setsockopt(4, SOL_SOCKET, SO_SNDBUF, [4095], 4) = 0
setsockopt(4, SOL_SOCKET, SO_RCVBUF, [65535], 4) = 0
fcntl(4, F_SETFL, O_RDONLY|O_NONBLOCK|O_LARGEFILE) = 0
bind(4, {sa_family=AF_NETLINK, nl_pid=6519, nl_groups=00000000}, 12) = 0
[... 115 lines cut ...]
munmap(0x7fbb338000, 8192)              = 0
munmap(0x7fbb336000, 8192)              = 0
munmap(0x7fbb334000, 8192)              = 0
munmap(0x7fbb332000, 8192)              = 0
munmap(0x7fbb32e000, 16384)             = 0
munmap(0x7fbb32a000, 16384)             = 0
munmap(0x7fbb326000, 16384)             = 0
munmap(0x7fbb322000, 16384)             = 0
[... cut ...]
munmap(0x7fbb2f2000, 65536)             = 0
munmap(0x7fbb402000, 8192)              = 0
writev(1, [{iov_base="", iov_len=0}, {iov_base="\n\33[?25h", iov_len=7}], 2
) = 7
writev(2, [{iov_base="1 ERROR(s) during runtime\n", iov_len=26}, {iov_base=NULL, iov_len=0}], 21 ERROR(s) during runtime
) = 26
writev(1, [{iov_base="", iov_len=0}, {iov_base="\n", iov_len=1}], 2
) = 1
writev(1, [{iov_base="", iov_len=0}, {iov_base="bye-bye\n\n", iov_len=9}], 2bye-bye

) = 9
exit_group(0)                           = ?
+++ exited with 0 +++`

ZerBea, 2024-02-04:
strace is not helpful in this case, because hcxdumptool use NETLINK to retrieve the interface list instead of requesting it directly from  "/sys/class/ieee80211/phy1/device/driver":

```
static bool get_interfacelist(void)
{
static size_t i;

nl_get_familyid();
if(nlfamily == 0)
	{
	errorcount++;
	return false;
	}
nl_get_regulatorydomain();
if(nl_get_interfacecapabilities() == false) return false;
if(nl_get_interfacelist() == false) return false;
for(i = 0; i < INTERFACELIST_MAX -1; i++)
	{
	if((ifpresentlist +i)->index == 0) break;
	ifpresentlistcounter++;
	}
if(rt_get_interfacelist() == false) return false;
if(ifpresentlistcounter == 0) return false;
qsort(ifpresentlist, ifpresentlistcounter, INTERFACELIST_SIZE, sort_interfacelist_by_index);
return true;
}
```

First of all we need to find the exact error: RTNETLINK or NL80211.
Please add some debug printf to the function mentioned above:
```
static bool get_interfacelist(void)
{
static size_t i;

nl_get_familyid();
if(nlfamily == 0)
	{
	errorcount++;
       fprintf(stderr, "NETLINK general failed to get family ID\n");
	return false;
	}
nl_get_regulatorydomain();
if(nl_get_interfacecapabilities() == false)
	{
       fprintf(stderr, "NETLINK NL80211 failed to get interface capabilites\n");
       return false;
       }
if(nl_get_interfacelist() == false)
	{
       fprintf(stderr, "NETLINK NL80211 failed to get interface list\n");
       return false;
       }

for(i = 0; i < INTERFACELIST_MAX -1; i++)
	{
	if((ifpresentlist +i)->index == 0) break;
	ifpresentlistcounter++;
	}
if(rt_get_interfacelist() == false)
	{
       fprintf(stderr, "RTNETLINK failed to get interface capabilites\n");
       return false;
       }
if(ifpresentlistcounter == 0) return false;
qsort(ifpresentlist, ifpresentlistcounter, INTERFACELIST_SIZE, sort_interfacelist_by_index);
return true;
}
```
Than run: call function to get Iinterface (L)ist
$ hcxdumptool -L
and to:  call function to get interface (I)nformation
$ hcxdumptool -I INTERFACE 

If we need to debug NETLINK we have to add a NETLINK MONITOR INTERFACE:
```
$ sudo ip link add  nlmon0 type nlmon
$ sudo ip link set dev nlmon0 up
```
Now the communication between hcxdumptool and the kernel can be monitored via tshark or Wireshark or tcpdump.

Next step is to debug NETLINK:

ZerBea, 2024-02-04:
My candidates:

either RTNETLINK is broken on openwrt
socket(AF_NETLINK, SOCK_RAW | SOCK_CLOEXEC, NETLINK_ROUTE)

or GENERIC NETLINK (NL80211) is broken on openwrt
socket(AF_NETLINK, SOCK_RAW | SOCK_CLOEXEC, NETLINK_GENERIC)

It is mandatory that you use latest git head of hcxdumptool which received a lot of openwrt NETLINK fixes!

This version has been confirmed as working:
https://github.com/ZerBea/hcxdumptool/discussions/382#discussioncomment-7859354

kimocoder, 2024-02-04:
Latest git;

```
root@OpenWrt:~/hcxdumptool# make -j2
cc -O3 -Wall -Wextra -Wpedantic -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.2-196-g074916f\" -DVERSION_YEAR=\"2024\" -DHCXSTATUSOUT -DHCXNMEAOUT 
hcxdumptool.c:14:10: fatal error: linux/filter.h: No such file or directory
   14 | #include <linux/filter.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```

ZerBea, 2024-02-04:
I'll check it.

ZerBea, 2024-02-04:
That is the header file of the Linux Berkeley Packet Filter.
It should be present in every Linux kernel, e.g.:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/include/linux?h=v5.15.148

kimocoder, 2024-02-04:
`/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/libbpf-1.3.0/include/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/iproute2-iptiny/iproute2-6.7.0/include/uapi/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/iptables-1.8.8/include/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.148/include/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.148/include/uapi/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.148/tools/include/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.148/tools/include/uapi/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.148/user_headers/include/linux/filter.h
/root/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.148/usr/include/linux/filter.h
`

kimocoder, 2024-02-04:
it's in the build system, but not available for hcxdumptool specific to use?`I have no clue

ZerBea, 2024-02-04:
looks like a gcc include path configuration problem.
```
$ echo | gcc -E -Wp,-v -
...
 /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/include
 /usr/local/include
 /usr/lib/gcc/x86_64-pc-linux-gnu/13.2.1/include-fixed
 /usr/include
...
$ ls /usr/include/linux/filter.h
/usr/include/linux/filter.h
```

ZerBea, 2024-02-04:
```
#define _GNU_SOURCE
#include <stdio.h>
#include <stdlib.h>
#include <linux/filter.h>
#include <linux/genetlink.h>
#include <linux/if_packet.h>
#include <linux/limits.h>
#include <linux/nl80211.h>
#include <linux/rtnetlink.h>
#include <linux/version.h>

int main()
{
int a = 0;
printf("%02x\n", a);

return EXIT_SUCCESS;
}
```

```
$ gcc filter.c -o filter
$ 
```

ZerBea, 2024-02-04:
Since v6.3.0 hcxdumptool has the option to add a Berkeley Packet Filter.
This is flexible and really fast and thousand times better than doing this by soft coded filter lists.

ZerBea, 2024-02-04:
Out of scope: rtl8xxxu

Please take a look at this:
https://bugzilla.kernel.org/show_bug.cgi?id=217205#c77

and this:
https://bugzilla.kernel.org/show_bug.cgi?id=217205#c84

Could be a fix (driver crash) and improvement add TP-Link TL-WN8200ND v3) for your driver, too:
https://github.com/kimocoder/realtek_rtwifi

kimocoder, 2024-02-04:
"
PHY	Interface	Driver		Chipset

phy0	wlan0		mt798x-wmac	Not pci, usb, or sdio
		(mac80211 monitor mode vif enabled for [phy0]wlan0 on [phy0]wlan0mon)
		(mac80211 station mode vif disabled for [phy0]wlan0)

root@OpenWrt:~# hcxdumptool -i wlan0mon -w capture.pcapng

Requesting physical interface capabilities. This may take some time.
Please be patient...

failed to get interface list
PACKET_STATISTICS failed

1 ERROR(s) during runtime
Possible reasons:
 driver is broken
 driver is busy (misconfigured system, other services access the INTERFACE)
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 no transmitter in range
 frames are filtered out by BPF
 driver is broken
0 SHB written to pcapng dumpfile
0 IDB written to pcapng dumpfile
0 ECB written to pcapng dumpfile
0 EPB written to pcapng dumpfile
"

After reinstall of OpenWRT (full flash) and clean install, I dont know what up

kimocoder, 2024-02-04:
"
root@OpenWrt:~# hcxdumptool -v
hcxdumptool 6.3.2-197-gf2d12ac (C) 2024 ZeroBeat
running on Linux kernel 5.15.148
compiled by gcc 12.3.0
compiled with Linux API headers 6.1.75
glibc (__GLIBC_MINOR__) is not defined
enabled REALTIME DISPLAY
enabled GPS support
disabled BPF compiler
root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.148 #0 SMP Sun Feb 4 00:12:51 2024 aarch64 GNU/Linux
root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# airmon-ng 

PHY	Interface	Driver		Chipset

phy0	wlan0		mt798x-wmac	Not pci, usb, or sdio

root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# hcxdumptool -i wlan0 -w capture.pcapng

Requesting physical interface capabilities. This may take some time.
Please be patient...

failed to get interface list
PACKET_STATISTICS failed

1 ERROR(s) during runtime
Possible reasons:
 driver is broken
 driver is busy (misconfigured system, other services access the INTERFACE)
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 no transmitter in range
 frames are filtered out by BPF
 driver is broken
0 SHB written to pcapng dumpfile
0 IDB written to pcapng dumpfile
0 ECB written to pcapng dumpfile
0 EPB written to pcapng dumpfile

root@OpenWrt:~# 
"

ZerBea, 2024-02-04:
`failed to get interface list `
GENERIC NETLINK or NETLINK ROUTE failed.

I'll code a test version. Please give me  some minutes.

ZerBea, 2024-02-04:
Here we go:
[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/14156241/hcxdumptool.c.zip)

replace git version by this one
```
$ make
$ ./hcxdumptool -L
$ ./hcxdumptool -I wlan0
```

kimocoder, 2024-02-04:
> Out of scope: rtl8xxxu
> 
> Please take a look at this: https://bugzilla.kernel.org/show_bug.cgi?id=217205#c77
> 
> and this: https://bugzilla.kernel.org/show_bug.cgi?id=217205#c84
> 
> Could be a fix (driver crash) and an improvement add TP-Link TL-WN8200ND v3) for your driver, too: https://github.com/kimocoder/realtek_rtwifi

Now rtl8xxxu is more developed, so I left mine

kimocoder, 2024-02-04:
> Here we go: [hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/14156241/hcxdumptool.c.zip)
> 
> replace git version by this one
> 
> ```
> $ make
> $ ./hcxdumptool -L
> $ ./hcxdumptool -I wlan0
> ```
> 
> Please do not use airmon-ng or iw to set a VIF (wlan0mon). We need the values of the hardware interface.

Yup, "Failed to get interface list"

ZerBea, 2024-02-04:
That's interesting.
I've added tons of debug messages. Every single call got one.

ZerBea, 2024-02-04:
But I have an idea.

ZerBea, 2024-02-04:
Same procedure:
[hcxdumptool.c.zip](https://github.com/ZerBea/hcxdumptool/files/14156726/hcxdumptool.c.zip)

Output should look like this:
```
$ ./hcxdumptool -l
requesting interface list via NETLINK
NETLINK ifindex 4
NETLINK ifname  wlp22s0f0u9u4
NETLINK wiphy 1
stderr, detected interfaces : 1
  1	  4	503eaa124a27	780b8c5efbac	*	wlp22s0f0u9u4   	mt76x0u	NETLINK
```

ZerBea, 2024-02-04:
airmon-ng is not affected, because it read the interface information directly from "/sys/class/ieee80211/phy1/device/driver" and not via NETLINK.

kimocoder, 2024-02-04:
"
root@OpenWrt:~/hcxdumptool hcxdumptool -i wlan0 -w capture.pcapng

Requesting physical interface capabilities. This may take some time.
Please be patient...

requesting interface list via NETLINK
NETLINK ifindex 13
NETLINK ifname  wlan0
NETLINK wiphy 0
stderr, detected interfaces : 0
failed to get interface list
PACKET_STATISTICS failed

1 ERROR(s) during runtime
Possible reasons:
 driver is broken
 driver is busy (misconfigured system, other services access the INTERFACE)
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 no transmitter in range
 frames are filtered out by BPF
 driver is broken
0 SHB written to pcapng dumpfile
0 IDB written to pcapng dumpfile
0 ECB written to pcapng dumpfile
0 EPB written to pcapng dumpfile

root@OpenWrt:~/hcxdumptool
"

ZerBea, 2024-02-04:
Pushed an update to handle phy0
Please try.

ZerBea, 2024-02-04:
Now we check the interface index.
I'm assuming that it always start at 1.

kimocoder, 2024-02-04:
"
root@OpenWrt:~/wiphy wifi config
Type error: attempt to set property on null value
In wiphy_get_entry(), file /usr/share/hostap/wifi-detect.uc, line 50, byte 2:
  called from function wiphy_detect (/usr/share/hostap/wifi-detect.uc:138:41)
  called from anonymous function (/usr/share/hostap/wifi-detect.uc:144:14)

 `    };`
      ^-- Near here

WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
root@OpenWrt:~/wiphy 
"

kimocoder, 2024-02-04:
OpenWRT wifi configuration is affected too

ZerBea, 2024-02-04:
Looks like only kernel 5.15 is affected, while others are working fine.
The workaround should be fine, as long as ifindex starts at 1.
If not, we got a big problem.

kimocoder, 2024-02-04:
@ZerBea I figured it out.....

kimocoder, 2024-02-04:
Now, hcxdumptool is running at least

ZerBea, 2024-02-04:
Great, glad that I could help.

ZerBea, 2024-02-04:
Right now, I'm refactoring hcxdumptool (hcxlabtool as test version has been pushed),
Than I play around with the rtl8xxxu driver to add  the TP-Link v3 to it.

kimocoder, 2024-02-04:
"
root@OpenWrt:~/hcxtest# ./hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  1  14 c87f54bd31aa c87f54bd31aa * phy1-ap0         mt798x-wmac (NETLINK)
  0  15 ca7f54bd31ab b025aa6dc14e * phy0-ap0         mt798x-wmac (NETLINK)

* active monitor mode available (reported by driver - do not trust it)
+ monitor mode available (reported by driver)
- no monitor mode available
"

Yeah! thanks a lot, a router without hcxdumptool and a 256gb usb3.2 disk, is not a proper router :1st_place_medal:

ZerBea, 2024-02-04:
I wouldn't call this a router...
It's an evil and hungry beast!

ZerBea, 2024-02-06:
I've tested opwnwrt on different Raspberries ant can confirm this behavior:
Raspbian: first deviice = phy1
Arch Linux: first deviice = phy1
Debian: first deviice = phy1

OpenWRT: first device = phy0 (what a surprise)

On all distributions: first ifindex is always 1 (increase on every new connected or reconnected device)

ZerBea, 2024-02-08:
Reopened, due to a huge problem:
```
# iw list
Wiphy phy1
	wiphy index: 1
...
Wiphy phy0
	wiphy index: 0
```

```
# iw dev
phy#1
	Interface wlan0
		ifindex 5
```

```
# ip addr
5: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
```

Using getifaddrs() doesn't show wlan1:
```
# ./gi
 ifindex: 1 ifname: lo      
 ifindex: 4 ifname: eth0    
 ifindex: 5 ifname: wlan0   
 ifindex: 6 ifname: br-lan  
```

If have to refactor the entire detection of interfaces.

ZerBea, 2024-02-08:
Looks like OpenWRT is doing something very special here:
3 devices present(2x USB and 1 x Raspberry Pi soc);
```
# iw list | grep phy
Wiphy phy2
	wiphy index: 2
Wiphy phy1
	wiphy index: 1
Wiphy phy0
	wiphy index: 0
```

```
# ls /sys/class/ieee80211/phy0/device/net/
ls: /sys/class/ieee80211/phy0/device/net/: No such file or directory

# ls /sys/class/ieee80211/phy1/device/net/
wlan0

# ls /sys/class/ieee80211/phy2/device/net/
ls: /sys/class/ieee80211/phy2/device/net/: No such file or directory

# iw dev
phy#1
	Interface wlan0
		ifindex 5
		wdev 0x100000001
		addr b8:27:eb:79:5f:d5
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz

Expected, because the Broadcom chip dos not support monitor mode
# iw dev wlan0 set type monitor
command failed: Not supported (-95)

Not expected
# iw dev wlan1 set type monitor
command failed: No such device (-19)

Not expected
# iw dev wlan2 set type monitor
command failed: No such device (-19)
```

Any ideas?

I don't think that's the intention!

ZerBea, 2024-02-08:
I think I got a solution:
```
# ./hcxlabtool
phy: 1 driver: brcmfmac
phy: 0 driver: rtl8xxxu
```
I can now retrieve all the information. But unfortunately, I have ti hijack all unassigned interfaces to find the best one.

kimocoder, 2024-02-08:
bcm has monitor mode and injection via Nexmon

ZerBea, 2024-02-08:
Thanks for the information.
I'm now able to detect unassigned devices running OpenWRT on a Raspberry Pi Zero w.
Next step is to hijack them.

Than I'll move from using IFINDEX:
```
 * @NL80211_CMD_SET_CHANNEL: Set the channel (using %NL80211_ATTR_WIPHY_FREQ
 *	and the attributes determining channel width) the given interface
 *	(identifed by %NL80211_ATTR_IFINDEX) shall operate on.
 *	In case multiple channels are supported by the device, the mechanism
 *	with which it switches channels is implementation-defined.
 *	When a monitor interface is given, it can only switch channel while
 *	no other interfaces are operating to avoid disturbing the operation
 *	of any other interfaces, and other interfaces will again take
 *	precedence when they are used.
```

to WIPHY:
```
 * @NL80211_CMD_SET_WIPHY: set wiphy parameters, needs %NL80211_ATTR_WIPHY or
 *	%NL80211_ATTR_IFINDEX; can be used to set %NL80211_ATTR_WIPHY_NAME,
 *	%NL80211_ATTR_WIPHY_TXQ_PARAMS, %NL80211_ATTR_WIPHY_FREQ,
 *	%NL80211_ATTR_WIPHY_FREQ_OFFSET (and the attributes determining the
 *	channel width; this is used for setting monitor mode channel),
```

Due to the same reasons I'll check if the user tries to run hcxdumptool on a VIF. If that is the case, I'll remove it.

ZerBea, 2024-02-09:
If you take a look at the latest commits on hcxlabtool you'll notice massive changed related to OpenWRT:
https://github.com/ZerBea/wifi_laboratory
And it's not even all the changes and there is still a lot of work to be done. But I think it is worth it, due to OpenWRTs incredible performance and well suited drivers.

Starting in February 2022, support for ARMv5 and ARMv6 architectures will be ended in Arch Linux ARM:
https://archlinuxarm.org/forum/viewtopic.php?f=3&t=15721
OpenWRT is by far the best alternative.

ZerBea, 2024-02-09:
Even OpenWRTs wifi config failed:
```
# wifi config
Type error: attempt to set property on null value
In wiphy_get_entry(), file /usr/share/hostap/wifi-detect.uc, line 50, byte 2:
  called from function wiphy_detect (/usr/share/hostap/wifi-detect.uc:138:41)
  called from anonymous function (/usr/share/hostap/wifi-detect.uc:144:14)

 `    };`
      ^-- Near here

WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
WARNING: Variable 'wlan' does not exist or is not an array/object
```

ZerBea, 2024-02-09:
I'm tired to wrestle with OpenWRT's configs. None of them really works.
I'll hijack the interfaces.

ZerBea, 2024-02-10:
Last but not least I got it. The problem is not related to hcxdumptool and there is no need to hijack devices that are not assigned.
The real cause is related to OpenWRT's wireless configuration in /etc/config/wireless:
By default the devices are configured to a PATH:
`option path 'platform/101c0000.ehci/usb1/1-1/1-1.1/1-1.1:1.0'`
That will not work, if the USB port has been changed.
A better solution e.g. for a Raspberry Pi Zero W is to assign them to the physical address (phy).
```
config wifi-device 'radio0'
        option type 'mac80211'
        option phy 'phy0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'default_radio0'
        optiion ifname 'wlan0'
        option device 'radio0'
        option mode 'managed'

config wifi-device 'radio1'
        option type 'mac80211'
        option phy 'phy1'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'default_radio1'
        option ifname 'wlan1'
        option device 'radio1'
        option mode 'managed'
```
Now the onboard device and an attached USB device are assigned and hcxdumptool is allowed to use them.

Closed, because it issue is related to OpenWRTs configuration.

closed 2024-02-10 by ZerBea