ZerBea/hcxdumptool #433: Android build instructions

gemesa, 2024-03-03:
Do we know why the Android build instructions have been removed in https://github.com/ZerBea/hcxdumptool/commit/23b1586df8aa7702319975bf0632a4f166a80754#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5L129? I know Android is not officially supported in hcxdumptool but then why are the Android build files not removed as well:
- https://github.com/ZerBea/hcxdumptool/tree/master/jni
- https://github.com/ZerBea/hcxdumptool/blob/master/Android.mk
- https://github.com/ZerBea/hcxdumptool/blob/master/AndroidManifest.xml

Is the Android build broken? I am asking because I am planning to try it out in the future.

ZerBea, 2024-03-03:
Android isn't broken.
This instructions have been removed because I assume only experienced users compile hcxdumptool for Android.
An experienced user shouldn't need these instructions.

Next step is to remove Debian instructions, too.

ZerBea, 2024-03-03:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/d187eed1c7caf2ece511c9aa158e4c426b20169e

Debian instructions have been removed, too.

ZerBea, 2024-03-03:
The same applies to hcxtools:
https://github.com/ZerBea/hcxtools/commit/1a80a1b0d1e504db2b41205f4b1959909359ba43

Thanks that you reminded me.

ZerBea, 2024-03-03:
Please let me know if you were successful (compiling hcxdumptool on Android).
If it doesn't work, I'll remove the entire Android build files.

ZerBea, 2024-03-03:
Ok, there are detailed instructions how to use the NDK:
https://developer.android.com/ndk/guides/other_build_systems

I'll remove the entire Android stuff.

closed 2024-03-03 by ZerBea

ZerBea, 2024-03-03:
I've opened a discussion:
https://github.com/ZerBea/hcxdumptool/discussions/434