ZerBea/hcxdumptool #437: prevent spoofed beacons from transmitting

RootLUG, 2024-03-12:
**Version info**

Tested on the latest version from git but I observed this behaviour for several prior versions:
```
root@DietPi:~/blah# hcxdumptool -v
hcxdumptool 6.3.4-14-g3693e77 (C) 2024 ZeroBeat
running on Linux kernel 6.1.21+
running GNU libc version 2.36
compiled by gcc 12.2.0
compiled with Linux API headers 6.6.20
compiled with GNU libc headers 2.36
enabled REALTIME DISPLAY
enabled GPS support
enabled BPF compiler

root@DietPi:~/blah# hcxdumptool -l
  0       2     deadbeef1    deadbeef1    -       wlan0                   brcmfmac        NETLINK
  1       3     deadbeef2    deadbeef2    *       wlan1                   mt76x2u NETLINK
```

**Describe the bug**
I am unable to prevent hcxdumptools from transmitting beacons of SSID networks that are requested by the client.
I have various hidden and offline networks configured on a client and I can see them each time I run hcxdumptools, it's a mix of hidden network and visible networks but that are offline. I have tried to configure `--proberesponsetx=0` but it can't be set to 0. 
The options `--disable_proberequest` and `--disable_beacon` (which I think is jsut for the wildcard SSID?) also does not prevent the beacons from being transmitted.
Is there any other setting that I am missing and should be set, I believe it may be a bug that it's still transmitting the spoofed beacons for these networks.  I am 100% sure these networks are not active because the access points associated with them no longer exists (physical hw).

**To Reproduce**
run hcxdumptool with a client that has hidden network configured or with a network that is not online

**Expected behavior**
I expect to not see spoofed SSID beacons of networks that are not currently active

ZerBea, 2024-03-12:
There are three kinds of attacks, which can be disabled by options and by a Berkeley Packet FIlter (BPF).
1. attacks against an AP
2. attacks against a NETWORK (AP and connected CLIENTs)
3. attacks against a CLIENT

To disable attacks on CLIENTs run:
$ sudo  hcxdumptool -i INTERFACENAME --attemptclientmax=0 --disable_beacon

To monitor outgoing traffic run Wireshark on the same interface as hcxdumptool and use this filter to monitor outgoing packets:
`radiotap.present.dbm_antsignal == False`

or by tshark:
$ tshark -i INTERFACENAME -Y "radiotap.present.dbm_antsignal == False" -o 'gui.column.format:"No.","%m","Time","%t","Source","%s","Destination","%d","Protocol","%p","Length","%L","Info","%i"'
Capturing on INTERFACENAME'

hcxdumptool will not respond to clients. Wireshark should confirm this and you should see this kind of frames (which belongs to attack mode 1 and 2:
```
deauthentication
authentication
associationrequest
reassociationrequest
```

To protect a CLIENT, I recommend to set up a BPF and disable MAC randomization on CLIENT side!

BTW:
At specific time intervals CLIENTs transmit a PROBEREQUEST to a NETWORK SSID stored in their wpa_supplicant.conf
hcxdumptool will show you the entire traffic on a channel. That include this REQUESTs.

RootLUG, 2024-03-13:
Thank you for the great explanation and help, I can confirm that the `--attemptclientmax=0 --disable_beacon` did exactly as described and I no longer see the spoofed beacons to appear.

closed 2024-03-13 by RootLUG

ZerBea, 2024-03-13:
Glad to hear this.
You're welcome.