ZerBea/hcxdumptool #440: hcxdumptool missing options

mohamed1-ae, 2024-03-17:
I had cloned the latest version of the respiratory yet i still can't use options like --enable_status or --filterlist_ap,

it says that the options weren't found, how can i fix this?

v74863, 2024-03-17:
Please close this issue because you are using an old version of hcxdumptool, not the latest.
Starting with the release v6.3.0 (05.05.2023), hcxdumptool uses different options.
In other words, the syntax has changed.
https://github.com/ZerBea/hcxdumptool/discussions/423

By the way, if you are asking for help, please provide more details.
Which Linux distro you are using, which exactly version of hcxdumptool, etc.

To get the latest version please read here
https://github.com/ZerBea/hcxdumptool/blob/master/README.md

ZerBea, 2024-03-17:
Additional to the FAQ (discussion section), more information can be found in changelog.
https://github.com/ZerBea/hcxdumptool/blob/3693e7721ac61b893d948add620de76d90749ed6/changelog#L177

And hcxdumptool's help menu:
$ hcxdumptool -h

Closed this report, because this is an issue tracker. If you have further going questions please use the discussion section.

closed 2024-03-17 by ZerBea