ZerBea/hcxdumptool #449: Aetheros chipset

Aggelos11, 2024-05-01:
I tried to use the tool with Alfa awus036nha which uses ether is chipset but I got the following error: 
driver is busy failed to transmit internal beacon

ZerBea, 2024-05-01:
The ALFA AWUSH036NHA is working fine:
https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7550862

You are running an old version of hcxdumptool that uses outdated WIRELESS EXTENSIONS instead of NETLINK.
The ERROR message you have mentioned above has been removed since hcxdumptool moved to version 6.3.0 (full NETLINK support). Make sure that you are running the latest version of hcxdumptool (at least 6.3.4).

Next time if you report an issue, please add all information as requested by the template:
```
**GitHub is for bugs and features - not for support**
Questions must be posted in the discussion board https://github.com/ZerBea/hcxdumptool/discussions

**Check the FAQ**
Please review the discussions

**Comment output of**
1. $ sudo hcxdumptool -v
2. $ sudo hcxdumptool -l
Bug reports will only be accepted against the current tip of the master git branch. Anything else will be rejected.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Comment the entire command line

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
```

Also make sure all services that take access to the device have been stopped.

Closed due to outdated version of hcxdumptool.

closed 2024-05-01 by ZerBea

ZerBea, 2024-05-01:
I do not recommend to use tools that need WIRELESS EXTENSIONS (WEXT), because they are deprecated:
https://wireless.wiki.kernel.org/wext-statement
Since WiFi 7 you'll run into problems as mentioned above.

It looks like the good old days of WIRELESS EXTENSIONS (WEXT) are finally over. I really appreciated WEXT - until I have updated to Linux kernel 6.3. It looks like the kernel hacker team decided to mark WEXT as deprecated in favor of NL80211.
From now on, dmesg show a warning if the use of WEXT is detected:
`[ 5718.909421] warning: xxxxxxx uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
`
As mentioned here:
https://bugzilla.kernel.org/show_bug.cgi?id=217205#c62
some more information is here:
https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/?id=4ca69027691a

I was hit by this warning, too, and decided to remove entire WEXT code:
https://github.com/ZerBea/hcxdumptool/commit/7c353a7a63d6f95f0e85522a63eb373f282b1da4

Aggelos11, 2024-05-01:
So what do you suggest that I must do in order to be able to use effectively this adaptor?

Aggelos11, 2024-05-01:
If I just use the latest version of the tool won’t it be okay?

ZerBea, 2024-05-01:
Latest release will be fine (hcxdumptool 6.3.4) or latest git head:
```
$ hcxdumptool -v
hcxdumptool 6.3.4-14-g3693e77 (C) 2024 ZeroBeat
running on Linux kernel 6.8.8-arch1-1
running GNU libc version 2.39
compiled by gcc 13.2.1
compiled with Linux API headers 6.7.0
compiled with GNU libc headers 2.39
enabled REALTIME DISPLAY
enabled GPS support
enabled BPF compiler
```
 
Additional I suggest to test hcxlabtool (some command line args have been changed) which will replace hcxdumptool in future times (possible 7.0.0).
https://github.com/ZerBea/wifi_laboratory
```
$ ./hcxlabtool -v
hcxlabtool 7.0 (C) 2024 ZeroBeat
running on Linux kernel 6.8.8-arch1-1
running GNU libc version 2.39
compiled by gcc 13.2.1
compiled with Linux API headers 6.7.0
compiled with GNU libc headers 2.39
enabled BPF compiler
```

A mainline, stable or long term Linux kernel >= 5.15.x as listed here:
https://www.kernel.org/
and mentioned in README.md:
`Operating system: Linux (recommended: kernel >= 6.6, mandatory: kernel >= 5.15)`

Aggelos11, 2024-05-01:
How about the version that is available on Kali repo ( I think it’s version 6.3.1) . Will it work all right?

Aggelos11, 2024-05-01:
Hey I downloaded the latest version and no error messages appears when running but when I cancel it it says 7 error due to broken drivers . 
[image]

ZerBea, 2024-05-02:
> How about the version that is available on Kali repo ( I think it’s version 6.3.1) . Will it work all right?

All versions >= 6.3.0 use the NETLINK (NL80211 and RTNETLINK). Version 6.3.1 should be fine
The changelog give an overview about improvements since 6.3.1.

Aggelos11, 2024-05-02:
Look at the errors I send above? Why are there ? How can I fix them ?

ZerBea, 2024-05-02:
Errors during runtime are never good.
hcxdumptool showed two possible reasons:
`driver is broken (this happens when the driver haven't received updates)`
Please comment output of
$ uname -r

driver is busy (this happens if the operating system prevent access to the device

By default KALI is not well configured and overloaded by tools which are normally nod needed
https://github.com/ZerBea/hcxdumptool/discussions/418
You have to identify system  services that take access to the device and stop them.

Aggelos11, 2024-05-02:
Well is kernel 6.9 amd64. If these error exist it means that the tool does not work properly ?

ZerBea, 2024-05-02:
I ran a small test. Absolutels no ERRORs on kernel 6.8.8

```
0 ERROR(s) during runtime
2560 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
87 EPB written to pcapng dumpfile

exit on sigterm
```

The test target has been attacked successfully:
```
RSN PMKID (best).........................: 1
RSN PMKID ROGUE..........................: 1
RSN PMKID written to 22000 hash file.....: 1
```

Aggelos11, 2024-05-02:
Hey I use the latest Kali and it’s 6.6.9 amd64. I did not compile it my self , it was pre installed

ZerBea, 2024-05-02:
Kernel 6.6.9 (latest version is 6.6.29) is a long term kernel and it should be fine. We can exclude driver problems.
Next step is to take a look at the running services to make sure that there is no service that take access to the device while hcxdumptool is running:
`$ systemctl --type=service --state=running`

Aggelos11, 2024-05-02:
These are the services running 
[image]

ZerBea, 2024-05-02:
Two services that take access to the device and make it busy:
NetworkManager.service
wpa_supplicant.service

BTW:
As I mantioned above, KALI is not well configured by default and serveral unwanted services are running.
If you don't have a 2G/3G/4G/5G modem, you should disable ModemManger.service.
If you don't want to run firmware updates by automatic, you should disable fwupd.service.

If you're running KALI inside a VM, make sure the host does not have access to the device while hcxdumptool is running.

Aggelos11, 2024-05-02:
Well I disabled wpa supplicant and network manager and it gave me no error. I have another question though . How will I know if I got any useful data ( I mean that can be cracked )? Can you explain me the functionality of the + symbol in each column?

ZerBea, 2024-05-02:
Columns P, 3, and 2 show this information:
```
                              P = + AP display     : got PMKID hashcat / JtR can work on
                              3 = + AP display     : got EAPOL M1M2M3 or EAPOL M1M2M3M4 (AUTHORIZATION) hashcat / JtR can work on
                              2 = + CLIENT display : got EAPOL M1M2 (ROGUE CHALLENGE) hashcat  / JtR can work  on
```

Aggelos11, 2024-05-02:
So if there is a + in Columns p and 3 it means I captured data that I can try to crack ?

ZerBea, 2024-05-02:
If a + is in column 3 you got a valid EAPOL handshake hashcat can work on. The recovered PSK belongs to the NETWORK.
If a + is in column P you got a valid PMKID hashcat can work on. The recovered PSK belongs to the NETWORK.
If a + is in column 2 you got a valid EAPOL handshake hashcat can work on. The recovered PSK must not belong to the NETWORK.