ZerBea/hcxdumptool #456: How to filter (skip) certain access points?

mickhaorex, 2024-06-05:
I run hcxdumptool with the parameters:
`sudo hcxdumptool -w $(date +“%F_%H-%M”).pcapng -i wlan1 -F`

Hcxdumptool scans all access points nearby.
How can I exclude certain access point MAC addresses from scanning?
I thought it was possible to run in previous versions:
`sudo hcxdumptool -w $(date +“%F_%H-%M”).pcapng -i wlan1 -F --filtermode=1 --filterlist_ap=filter.txt`
where `filter.txt` is a list of MAC addresses to skip.  (I may be wrong).

Now these parameters (`--filtermode=1 --filterlist_ap=filter.txt`) are not working. 
So how do I add a list of excluded MAC addresses ?

closed 2024-06-05 by ZerBea

dart8888, 2025-12-27:
ИСПРАВЬ 

$ hcxdumptool --bpfc="not wlan addr3 112233445566" > filter.bpf
кое как разобрался почему у меня игнор лист не работает)))

ZerBea, 2025-12-27:
@dart8888 , Your solution will work as expected, but it is very, very restrictive.
 If you convert the pcapng file to a hash file, hcxpcapngtool will display a warning that some useful frames types (in this case undirected proberequest frames) are filtered out:
```
Information: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK. It always happens if the capture file was cleaned or it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.
https://github.com/ZerBea/hcxdumptool/discussions/526
```

dart8888, 2025-12-27:
> [@dart8888](https://github.com/dart8888)Ваше решение будет работать как ожидалось, но оно очень, очень ограничительное. Если вы преобразуете файл pcapng в хеш-файл, hcxpcapngtool выдаст предупреждение о том, что некоторые полезные типы кадров (в данном случае кадры undirected proberequest) отфильтровываются:
> 
> ```
> Information: missing frames!
> This dump file does not contain undirected proberequest frames.
> An undirected proberequest may contain information about the PSK. It always happens if the capture file was cleaned or it could happen if filter options are used during capturing.
> That makes it hard to recover the PSK.
> https://github.com/ZerBea/hcxdumptool/discussions/526
> ```

я тебе говорю исправь перепиши добавь "NOT" в строку фильтра, иначе ничего не работает, всю ночь думал об этом.. 
вот как ты написал:

$ hcxdumptool --bpfc="wlan addr3 112233445566" > filter.bpf
$ sudo hcxdumptool -i wlp5s0f4u2 --rds=1 --bpf=filter.bpf -w test.pcapng

а надо:

$ hcxdumptool --bpfc="not wlan addr3 112233445566" > filter.bpf
$ sudo hcxdumptool -i wlp5s0f4u2 --rds=1 --bpf=filter.bpf -w test.pcapng

чтобы фильтр работал.

ZerBea, 2025-12-27:
@dart8888 , ok, my fault. I have missed the "not". My Cyrillic is a little rusty.
`hcxdumptool --bpfc="not wlan addr3 112233445566" > filter.bpf`
is a protection filter that prevent an attack this AP (MAC: 112233445566)

`hcxdumptool --bpfc="wlan addr3 112233445566" > filter.bpf`
is a very restrictive filter to attack exactly this AP (MAC 112233445566)