ZerBea/hcxdumptool #458: Issues using hcxdumptool in conjuction with ALFA AWUS036ACH

Jake-Grafton, 2024-06-10:
**GitHub is for bugs and features - not for support**
Questions must be posted in the discussion board https://github.com/ZerBea/hcxdumptool/discussions

**Check the FAQ**
Please review the discussions

**Comment output of:**
1. $ sudo hcxdumptool -v
```
hcxdumptool 6.3.4-17-g904a2a8 (C) 2024 ZeroBeat
running on Linux kernel 6.6.31+rpt-rpi-v8
running GNU libc version 2.36
compiled by gcc 12.2.0
compiled with Linux API headers 6.6.31
compiled with GNU libc headers 2.36
enabled REALTIME DISPLAY
enabled GPS support
enabled BPF compiler
```
2. $ sudo hcxdumptool -l
```
  0	  3	e45f0166cd50	e45f0166cd50	-	wlan0           	brcmfmac	NETLINK
  1	  4	00c0cab035be	00c0cab035be	+	wlan1           	rtl88XXau	NETLINK
```

**Describe the bug**
When using hcxdumptool with an [ALFA AWUS036ACH](https://www.amazon.com/ALFA-AWUS036ACH-%E3%80%90Type-C%E3%80%91-Long-Range-Dual-Band/dp/B08SJC78FH) adapter running the [aircrack-ng rtl8812au driver](https://github.com/aircrack-ng/rtl8812au), hcxdumptool launches without any warnings, and starts running. When running, hcxdumptool doesn't populate the RDS with any access points or clients. Upon termination, I'm hit with the following output:
```
0 ERROR(s) during runtime
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
Warning: too less packets received (monitor mode may not work as expected)
Possible reasons:
 no transmitter in range
 frames are filtered out by BPF
 driver is broken
Warning: no PROBERESPONSES received (frame injection may not work as expected)
Possible reasons:
 no AP in range
 frames are filtered out by BPF
 driver is broken
 driver does not support frame injection

exit on watchdog
```
All of these errors point to a driver error but I've seen others have success with the aircrack-ng rtl8812au driver.

**To Reproduce**
`sudo hcxdumptool -i wlan1 -F --rcascan=a`

**Expected behavior**
I expected hcxdumptool to startup without errors and begin scanning for surrounding access points and clients.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Note:** Instead of individual screenshots, I recorded this video showcasing the bug.

https://github.com/ZerBea/hcxdumptool/assets/91102919/f49d07fc-206d-4642-a390-bb51e17f8100

**Additional context**
Add any other context about the problem here.

All of the evidence points towards a driver malfunction/error but hcxdumptool does behave in different ways when different drivers are used. For example, [this](https://github.com/astsam/rtl8812au) driver by **astsam** is old and causes hcxdumptool to terminate before it starts scanning.

ZerBea, 2024-06-11:
Your guess is correct. This are driver issues and exactly that is the reason why I do not recommend to use chipsets that depend on third party drivers.
```
README.md
No support for a third party driver which is not part of the official Linux kernel (https://www.kernel.org/)
Report related issues to the site, from which you downloaded the driver.
```
Each driver coming from git exhibits a different behavior (especially in combination with different kernel versions). Nearly all of them depend on the old hal modules and absolutely none of them will find its way into the Linux kernel tree.
Please read this:
https://github.com/morrownr/USB-WiFi/issues/314
and this:
https://github.com/morrownr/USB-WiFi/issues/449

Luckily there is work in progress (new rtw88 stack) to add RTL8812au support to the Linux kernel:
https://github.com/morrownr/8821au-20210708/issues/133#issuecomment-2159223160

I'm monitoring this rtw88 driver too, as well as the maintainer of the aircrack-ng driver (kimocoder)
I've tested the driver in combination with a RTL8811 device and it is working as expected
I recommend to test this driver and to report possible problems there.

closed 2024-06-11 by Jake-Grafton

morrownr, 2024-06-24:
@Jake-Grafton @ZerBea 

The new 8812au driver is ready for initial testing. Read the first message at the following link for details:

https://github.com/morrownr/8812au-20210820/issues/27

 We really need testers to beat up monitor mode.

ZerBea, 2024-06-24:
First of all, a very sad news:

https://lore.kernel.org/linux-wireless/ds6wc3svkyre4p2rwg4f76o67xndazaeoyjwblhuzichscyxoz@5ttdvbymxr55/T/#mfd2f4928e0e013c10375ca766035c1385e18f8e0

RiP Larry and thanks for your great work. The community will remember you!

ZerBea, 2024-06-24:
@Jake-Grafton 
At the moment I only can test 8811au. Maybe I can put my hands on a cheap 8812au device, soon.

Jake-Grafton, 2024-06-24:
@ZerBea
Understood. I just tested my 8812au based device with the new driver and it worked just fine! (see attached screenshot for hcxdumptool output)
[image]

ZerBea, 2024-06-24:
Glad to hear that it is working.
Latest Linux kernels need latest driver. There is no place for the ancient "hal" ones.

BTW:
rt88/rtw99 will be continued.

ZerBea, 2024-06-24:
An additional information:

None of my mt6 based combined WiFi / BT devices is really working as expected on USB3 (AMD mainboards).
For me it is mandatory to prevent that btusb will be loaded:
`echo "install btusb /bin/true" >> /etc/modprobe.d/blacklist.conf`

Otherwise they do not work:
```
[11321.446538] usb 5-2.4: new high-speed USB device number 5 using xhci_hcd
[11321.541550] usb 5-2.4: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[11321.541557] usb 5-2.4: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[11321.541560] usb 5-2.4: Product: Wireless_Device
[11321.541561] usb 5-2.4: Manufacturer: MediaTek Inc.
[11321.541563] usb 5-2.4: SerialNumber: 000000000
[11321.770836] Bluetooth: Core ver 2.22
[11321.770866] NET: Registered PF_BLUETOOTH protocol family
[11321.770867] Bluetooth: HCI device and connection manager initialized
[11321.770872] Bluetooth: HCI socket layer initialized
[11321.770874] Bluetooth: L2CAP socket layer initialized
[11321.770878] Bluetooth: SCO socket layer initialized
[11321.807855] usbcore: registered new interface driver btusb
[11321.810801] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[11324.591334] Bluetooth: hci0: Device setup in 2718184 usecs
[11324.591341] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[11326.646123] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[11328.779479] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[11330.913067] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[11331.002514] usb 5-2.4: reset high-speed USB device number 5 using xhci_hcd
```
The faulty BT module blocks the USB port. As a result neither WiFi nor BT is working!

morrownr, 2024-06-24:
@ZerBea 

>None of my mt76 based combined WiFi / BT devices is really working as expected on USB3 (AMD mainboards).

I have seen reports of this but had not been able to duplicate it. You added something to the conversation that I had not seen before because most users probably did not consider it to be relevant: AMD mainboards

I very recently got a new Mini PC with an AMD CPU. So I decided to test my AXML. I was able to duplicate the problem. So as to not derail this thread, I'll take this email.