ZerBea/hcxdumptool #464: hcxdumptool exits on unknown error

Potato758, 2024-07-17:
when i run hcxcumptool it runs for like 1 or 2 minutes and exits on an error that i dont even know without any captured eapol packets and of course when i try to use hcxpcapngtool to convert to hc22000 it throws an error due to lack of eapol packets

also some columns and the lower table are completely empty

my command: `sudo hcxdumptool -i wlan0 -w dump.pcapng -F`

output:
[image]

[image]

output of `sudo hcxdumptool -v` :

```
hcxdumptool 6.3.4-32-ge434780 (C) 2024 ZeroBeat
running on Linux kernel 6.8.11-amd64
running GNU libc version 2.38
compiled by gcc 13.2.0
compiled with Linux API headers 6.8.11
compiled with GNU libc headers 2.38
enabled REALTIME DISPLAY
enabled GPS support
enabled BPF compiler
``` 

output of `sudo hcxdumptool -l` :
0       3     5c6199449d8d    9c93e4452ae5    *       wlan0                   rtw_8822ce      NETLINK

ZerBea, 2024-07-17:
I suspect that packet injection is broken on that driver because the status display doesn't show a response to hcxdumptool. The R column is completely empty.

By latest commit I've added the TX ERROR COUNT to the exit status display.
Please do a git pull and enable expanded debug log:
uncomment DEFS line in Makefile:
```
# uncomment to enable DEBUG log
#DEFS		+= -DHCXDEBUG
```
to
```
# uncomment to enable DEBUG log
DEFS		+= -DHCXDEBUG
```
Compile hcxdumptool, run it again and attach hcxdumptool log.

BTW:
The driver is not a Linux stock kernel driver. 
Is the driver from here?
https://github.com/lwfinger/rtw88

ZerBea, 2024-07-17:
I can't reproduce the ERROR running a rtw_8821cd device.
Neither on kernel 6.6.40 nor on kernel 6.10 and neither on the Linux stock kernel driver nor on the third party driver mentioned above:
```
$ hcxdumptool -l
  0	  3	70665514e31a	e80410d97874	+	wlp4s0          	rtw_8821ce	NETLINK

sudo ./hcxdumptool --rcascan=active  -i wlp4s0

116 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
21 PROBERESPONSE(s) captured

exit on sigterm
```

Please notice that you are running an outdated kernel:
"running on Linux kernel 6.8.11-amd64"
Which is not longer maintained:
https://www.kernel.org/

ZerBea, 2024-07-17:
BTW:
This is the easiest way to test that injection is working:
$ sudo hcxdumptool --rcascan=active  -i INTERFACENAME

When hcxdumptool has been terminated the exit status will show you if packet injection is working as expected or not.

Potato758, 2024-07-18:
> Compile hcxdumptool, run it again and attach hcxdumptool log file (hcxdumptool.log).

here is the log:  [hcxdumptool.log](https://github.com/user-attachments/files/16284200/hcxdumptool.log)

and this is the output:

[image]

> BTW: Is this the Linux stcok kernel driver or the driver from here: Is the driver from here? https://github.com/lwfinger/rtw88

its the stock kernel driver, i didnt suspect it was a driver problem because every other tool i tried was working fine

> Please notice that you are running an outdated kernel: "running on Linux kernel 6.8.11-amd64" Which is not longer maintained: https://www.kernel.org/
> 
> There are several driver fixes on 6.9.9 which are backported on 6.40. This fixes never reached 6.8.11.

I didnt know it was outdated. im currently building kernel 6.10 but its taking a while. when im done i will let you know if that fixes the problem

> BTW: This is the easiest way to test that injection is working: $ sudo hcxdumptool --rcascan=active -i INTERFACENAME
> 
> When hcxdumptool has been terminated the exit status will show you if packet injection is working as expected or not.

it seems to be not working

[image]

also the log file after running `sudo hcxdumptool --rcascan=active -i wlan0` was filled with this line `send_80211_proberequest_undirected failed: Resource temporarily unavailable`
although i ran `sudo systemctl stop NetworkManager.service && sudo systemctl stop wpa_supplicant.service` before running hcxdumptool so i really dont understand what might be causing this

also i tested frame injection using aireplay-ng and it was working,i dont know if this helps but thought i should let you know

ZerBea, 2024-07-18:
Thanks for the log file.

Let me explain this output.
`send_80211_beacon failed: Resource temporarily unavailable`

send_80211_beacon failed 
This is an ERROR message generated by hcxdumptool to see what kind of frame has not been transmitted
hcxdumptool wants to transmit a BEACON frame but it failed

Resource temporarily unavailable
This is the ERROR message from the operating system. It means that the frame can't be transmitted because the driver blocked it. This is a driver ERROR  and frames (1196) not transmitted by the device.

I've tested the Linux stock kernel driver:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtw88?h=v6.9.9
and the new rtw88 driver from here:
https://github.com/lwfinger/rtw88
Sooner or later (if everything has been fixed, this driver will be committed to the Linux stock kernel. You can watch the progress and report problems here:
https://github.com/morrownr/8821au-20210708/issues/133

Both are working flawless on my PCIe card:
`04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter`

There are various driver fixes between 6.8.11 and 6.9.9
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/net/wireless/realtek/rtw88?h=v6.8.11
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/net/wireless/realtek/rtw88?h=v6.9.9

That applies to the mac stack, too:
https://bugzilla.kernel.org/show_bug.cgi?id=218884
6.8. didn't get the changes/fixes

Please also take a look at this report. It is not the first time that the driver failed:
https://bugzilla.kernel.org/show_bug.cgi?id=218040

BTW:
You can't compare aireplay-ng frame injection test to hcxdumptool because hcxdumptool transmit a lot more different frame types than aireplay-ng and aireplay-ng does not count ERRORs
E.g aireplay-ng does not transmit BEACON frames - according to hcxdumptool this is the first frame that failed to be transmitted.

ZerBea, 2024-07-18:
If I have some time, I'll compile 6.10 and do a few tests (running hcxlabtool).

Potato758, 2024-07-18:
I installed kernel 6.10 and compiled hcxdumptool again and tried the driver from github but looks like its no use

also i notice a very strange pattern, when i only run `sudo hcxdumptool --rcascan=active -i wlan0` it seems to be working fine

[image]

but when i run `sudo hcxdumptool -i wlan0 -w dump.pcapng` and then run `sudo hcxdumptool --rcascan=active -i wlan0` it behaves the same as i showed in my previous reply (this is what i mean) :

> ![Screenshot_2024-07-18_16_37_06](https://private-user-images.githubusercontent.com/159473900/349987697-119411c7-2046-4a97-93ca-fb1b8e91146d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzNDM2MDAsIm5iZiI6MTcyMTM0MzMwMCwicGF0aCI6Ii8xNTk0NzM5MDAvMzQ5OTg3Njk3LTExOTQxMWM3LTIwNDYtNGE5Ny05M2NhLWZiMWI4ZTkxMTQ2ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxOFQyMjU1MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YWYxN2M1MTI0MTgwZmYxMTc2ODQ5YzZmMjI0ZGQ2N2NkZTc3YjhlZGFkODMyYjI1NjJhNjMxM2E2NTcxMzUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9DaTGo9MnyJYKkAdy95n6FWum_dngOP5flN6MDPLTGo)

and it makes it seem like frame injection isnt working and it doesnt work again until i restart NetworkManager and connect to the internet and then run `sudo systemctl stop NetworkManager.service && sudo systemctl stop wpa_supplicant.service`  just then it works fine although the response it gets is from the same ap's everytime but i dont really know if its a problem from the ap's around me, anyway do you know what is causing this behavior or how to fix it?

btw the output of  `sudo hcxdumptool -i wlan0 -w dump.pcapng` stays the same at all states only `sudo hcxdumptool --rcascan=active -i wlan0` differ

ZerBea, 2024-07-19:
Thanks for testing it on kernel 6.10. The new test has shown that it is a rtw8822ce driver bug.
The driver has been changed, hcxdumptool is the same, but the result is different.
It looks like the updated driver is a little bit better, but it is still not flawless, beacause hcxdumptool is partly working now.
But some frames are still not (correct) handled by the driver and cause it to block: _Resource temporarily unavailable_
There is nothing I can do, because the problem is inside the driver.

You can open an issue report here:
https://github.com/morrownr/USB-WiFi/issues

Additional you can try hcxlabtool, but I don't expect that it is working as expected until the rtw8822ce driver received a fix.
https://github.com/ZerBea/wifi_laboratory
The command line options are nearly the same as hcxdumptool.

BTW
That's right, I use a rtw8821ce which is working fine.

BTW: 2
The screenshot link shows an empty page.

ZerBea, 2024-07-19:
Does dmesg show something related to the rtw88 driver?

ZerBea, 2024-07-19:
Let me explain how hcxdumptool communicates with a driver:
```
RTNETLINK is used to get general information about status and mode, to set mode and to set interface up
NL80211 is used to get WiFi related information and to set monitor mode, active monitor mode, channel and to turn power-save off
A RAW packet socket (PF_PACKET) is used to send the frames to the driver
A second RAW packet socked (PF_PACKET) is used to receive frames from the driver
```
All this are standard system calls and the driver must support them flawless (which is not the case on rtw8822ce).
If they are supported, hcxdumptool will work as expected
https://github.com/ZerBea/hcxdumptool/discussions/361
If the driver does not support them (e.g  they are not present or contain a bug) hcxdumptool will not work.

You may have noticed that I changed the head line, because it is a problem of the rtw8822ce driver. 
I can't code a workaround in hcxdumptool. This has to be fixed inside the driver.

ZerBea, 2024-07-19:
Due to several problems in combination with Realtek drivers: Realtek is now on the list of not recommended chipsets!
https://github.com/ZerBea/hcxdumptool/commit/83404dd0de027b5169a361a6efb858a034465c59

morrownr, 2024-07-19:
@ZerBea 

> Due to several problems in combination with Realtek drivers: Realtek is now on the list of not recommended chipsets!

I agree with this. Maybe someday things will be better.

I'm going to say some things that @Potato758 may not want to hear since his system has a rtl8822c card but it is an assessment that I think is reasonable.

The rtl8822c chip came along late in the WiFi 5 product cycle. I've maintained out-of-kernel usb drivers for the more commonly used Realtek chips in an effort to help Linux users. I also work on rtw88 in an effort to get out of maintaining the out-of-kernel drivers (they are not good drivers for monitor mode). I do not have a rtl8822cu adapter. None of the makers I have a relationship with use that chip which means I do not have good access to recent drivers for the chip. I made a decision a few years ago to not support the rtl8822cu and I have only had 2 requests in the following years. The rtl8812au and rtl88x2bu get hundreds of downloads per day. My point being is that you have to have users to have drivers maintained or improved and I just never see that happening with the rtl8822c chip. Why Realtek releases so many chips that basically do the same thing but require different drivers is ridiculous to me. Mediatek releases far fewer chips but all have good Linux support done in accordance with Linux Wireless Standards. Sure, a bug or two shows up in Mediatek drivers at times but on the whole, Mediatek's drivers are very good.

Sometimes you just have to step back and look at the big picture. Should I put a lot of energy into something that is likely to cause me problems for years or just spend $20-$30 and get something that will likely be a pleasure to use for many years.

I recently got myself a new little Mini PC. I knew the one that I ordered had an Intel wifi card, which is fine if all you do is managed mode, so I ordered a new mt7925 chip based M.2 card to replace it. The mt7925 is a tri-band WiFi 7 chip. I installed Ubuntu 24.04 and poof, wifi worked. Mediatek has had the driver in the Linux kernel since kernel 6.7. Mediatek is basically to the point that they are getting their drivers in the Linux kernel months before products are released much like what Intel and AMD do. Realtek is still confused about Linux wifi support.

In managed mode, I have seen no issues at all. Are their bugs in the mt7925 driver? Sure I just haven't ran into any yet. I don't have time right now but over the next 2 weeks or so, if you are interested, I can make time to run some hcxdumptool commands and provide you with the output. You just tell me what commands to run to give us an idea what is working well in the mt7925 driver and I will make it happen.

@morrownr

Potato758, 2024-07-19:
>It looks like the updated driver is a little bit better, but it is still not flawless, because hcxdumptool is partly working now.
But some frames are still not (correct) handled by the driver and cause it to block: Resource temporarily unavailable

I don't think this is the case, I just didn't notice this behavior until i ran kernel 6.10 . To make sure i ran kernel 6.8.10 and it behaves the same but I was the one who didn't notice this from the start.

> BTW: 2
The screenshot link shows an empty page.

Sorry for that i meant to quote the screenshot where it shows that frame injection isn't working, i edited it and should work now.

> Additional you can try hcxlabtool, but I don't expect that it is working as expected until the rtw8822ce driver received a fix.

I don't know what is going on but it's working almost perfectly, I don't know what exactly should be the output but it just keeps displaying `starting...`, aside from that the output file has BEACON frames and it can be converted to 22000 with hcxpcapngtool without any errors, so everything is working fine. 

```
$ sudo hcxlabtool -i wlan0 -w labtest.pcapng

This is a highly experimental penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!
Misuse within a network, without specific authorization, may cause
irreparable damage and result in significant consequences!
Not understanding what you were doing is not going to work as an excuse!

BPF is unset! Make sure hcxdumptool is running in a 100% controlled environment!

starting...
```
also when I exit it shows no errors:

```
^C
0 ERROR(s) during runtime
0 TX ERROR(s) during runtime
12847 Packet(s) captured by kernel
1805 Packet(s) dropped by kernel
exit on sigterm
```

I also believe that every time I run `sudo hcxdumptool -i wlan0 -w dump.pcapng` frame injection stops working until i connect to the internet. I'm saying this because when i run hcxlabtool after hcxdumptool it also exits on the same error that hcxdumptool does, additionally of course `--rcascan=active` shows that frame injection isn't working after I run hcxdumptool so that's what I conducted.

And I tested hcxdumptool on a completely different machine with a different wireless adapter and it's working fine so it's definitely a driver problem.

> Does dmesg show something related to the rtw88 driver?

I just shows this:
```
[Jul19 21:17] rtw_8822ce 0000:04:00.0 wlan0: entered promiscuous mode
[Jul19 21:18] rtw_8822ce 0000:04:00.0 wlan0: left promiscuous mode
```

Anyway hcxlabtool is working and this is more than enough for me, Thanks for your help :D

Potato758, 2024-07-19:
@morrownr 
Thanks for your explanation of the situation, unfortunately when I got my pc i wasn't very interested in this kind of stuff so I didn't pay much attention to the wireless adapter it came with.
It's true that the driver of this chip has some bugs, but most of the things I run work fine and those that don't, I can usually find a workaround for them or fix it somehow, and this does the job for me. But I very much appreciate your reply.

closed 2024-07-19 by Potato758

ZerBea, 2024-07-19:
hcxdumptool is designed to run mostly headless. That means by default options there is no status display.
This has to be activated by: --rds=1 or --rds=2
```
--rds=<digit>  : enable real time display
                  default = 0 (off)
                  1 = show APs on current channel, show CLIENTs (M1M2ROGUE)
                  2 = show all APs (M1M2, M1M2M3 or PMKID), show CLIENTs (M1M2ROGUE)
                  3 = show all APs, show CLIENTs (M1M2ROGUE)
                  columns:
                   A = AKM (p)re-shared key
                   1 = received M1
                   2 = received M1M2
                   3 = received M1M2M3
                   P = received PMKID
```

It looks like the driver doesn't transmit BEACON frames. If hcxdumpool send a BEACON, the driver crashes.
Hcxlabtool's attack engine is a little bit different. There is no longer need to transmit BEACON frames.
Aireplay-ng doesn't transmit BEACON frames during injection test which could be a reason why it works.

I fully agree to the comment of @morrownr . Realtek releases a  lot of new chipsets - unfortunately the drivers can't follow.

Looks like

dubhater, 2024-09-29:
Beacon frames are special. In rtw88 (and rtl8xxxu) beacons are uploaded to a part of the chip's memory reserved for frames that the chip needs to transmit independently. In AP mode the chip transmits beacons every 102.4 ms. In station mode the chip transmits null data frames when it goes to sleep/wakes up.

So beacons are not sent to the normal TX path in the chip. If you must inject beacons, I think the drivers can be modified to allow it.