ZerBea/hcxdumptool #466: time travel: csv output

KaliAssistant, 2024-07-28:
```
sudo hcxdumptool -v

hcxdumptool 6.3.1 (C) 2023 ZeroBeat
```
I am using hcxdumptool with bpf, and I get EAPOL M1M2. When i output the csvfile:
`hcxpcapngtool 20240728132719-wlan0.pcapng --csv=csv.csv`
```
2024-07-28	05:07:32	aa:bb:cc:dd:ee:ff	AP	[WPA2]	[CCMP]	[PSK]	00	4	254	0.000000	E	0.000000	S	0.000000	0.000000	0	0	0.000000	0.000000	M
2024-07-28	05:07:34	aa:bb:cc:dd:ee:ff	AP	[WPA2]	[CCMP]	[PSK]	00	4	254	0.000000	E	0.000000	S	0.000000	0.000000	0	0	0.000000	0.000000	M
2024-07-28	04:48:44	aa:bb:cc:dd:ee:ff	AP	[WPA2]	[CCMP]	[PSK]	00	4	252	0.000000	E	0.000000	S	0.000000	0.000000	0	0	0.000000	0.000000	M
2024-07-28	04:48:46	aa:bb:cc:dd:ee:ff	AP	[WPA2]	[CCMP]	[PSK]	00	4	252	0.000000	E	0.000000	S	0.000000	0.000000	0	0	0.000000	0.000000	M

```
The first two lines are the results from the last time.

The time in the csv file is completely different from the time in the pcapng file. Instead, it is set back by more than ten minutes.

And hcxpcapngtool always says that the file has a timestamp problem
```
hcxpcapngtool 20240728132719-wlan0.pcapng --csv=kaliroom.csv
hcxpcapngtool 6.2.7 reading from 20240728132719-wlan0.pcapng...

summary capture file
--------------------
file name................................: 20240728132719-wlan0.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.1.73-v8+
application..............................: hcxdumptool 6.3.1
interface name...........................: wlan0
interface vendor.........................: 3460f9
openSSL version..........................: 1.0
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: (incremented on every new client)
MAC CLIENT...............................:
REPLAYCOUNT..............................: 61689
ANONCE...................................:
SNONCE...................................:
timestamp minimum (GMT)..................: 28.07.2024 12:48:44
timestamp maximum (GMT)..................: 28.07.2024 14:27:04
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianness (capture system)...............: little endian
packets inside...........................: 54
frames with correct FCS..................: 49
packets received on 2.4 GHz..............: 49
ESSID (total unique).....................: 5
BEACON (total)...........................: 1
BEACON on 2.4 GHz channel (from IE_TAG)..: 4
ACTION (total)...........................: 2
ACTION (containing ESSID)................: 2
PROBEREQUEST.............................: 5
PROBERESPONSE (total)....................: 1
AUTHENTICATION (total)...................: 3
AUTHENTICATION (OPEN SYSTEM).............: 3
ASSOCIATIONREQUEST (total)...............: 5
ASSOCIATIONREQUEST (PSK).................: 5
EAPOL messages (total)...................: 37
EAPOL RSN messages.......................: 37
EAPOLTIME gap (measured maximum usec)....: 127420108
EAPOL ANONCE error corrections (NC)......: not detected
EAPOL M1 messages (total)................: 28
EAPOL M2 messages (total)................: 6
EAPOL M3 messages (total)................: 1
EAPOL M4 messages (total)................: 2
EAPOL pairs (total)......................: 18
EAPOL pairs (best).......................: 1
EAPOL ROGUE pairs........................: 1
EAPOL M12E2 (challenge)..................: 1

frequency statistics from radiotap header (frequency: received packets)
-----------------------------------------------------------------------
 2427: 49

Information: no hashes written to hash files

Warning: out of sequence timestamps!
This dump file contains frames with out of sequence timestamps.
That is a bug of the capturing tool.

session summary
---------------
processed pcapng files................: 1
```

ZerBea, 2024-07-28:
You're running outdated versions of hcxdumptool and hcxpcapngtool:
hcxdumptool 6.3.1
hcxpcapngtool 6.2.7

The time stamp handling received a first fix here;
```
10.03.2023
==========
release v6.2.9
hcxpcapngtool moved default timestamp from usec to nsec (timeval to timespec)
```

Please update to latest release 6.3.4
or latest git head:
```
$ hcxdumptool -v
hcxdumptool 6.3.4-37-g0f99877 (C) 2024 ZeroBeat
running on Linux kernel 6.6.42-1-lts
running GNU libc version 2.40
compiled by gcc 14.1.1
compiled with Linux API headers 6.8.0
compiled with GNU libc headers 2.39
enabled REALTIME DISPLAY
enabled GPS support
enabled BPF compiler

$ hcxpcapngtool -v
hcxpcapngtool 6.3.4-35-gb34a619 (C) 2024 ZeroBeat
```

If the problem still exist please attach a pcapng file recorded by latest git head of hcxdumptool 

BTW:
hcxtools version should always match to hcxdumptool version.

KaliAssistant, 2024-07-28:
thanks . it works!

closed 2024-07-28 by KaliAssistant

ZerBea, 2024-07-28:
BTW:
The next important change will be v 7.0.0
The entire GPS stuff will be removed from hcxdumptool because it will take to many CPU cycles.
hcxpcapngtool will read a gpx file and correlate it with the WiFi traffic recorded by hcxdumptool.
Main advantage is that you can use every device (gpsd gpxlogger, smart phone, Garmin gps device, GPS USB device) to record a GPS track

KaliAssistant, 2024-07-28:
> 順便一提： 下一個重要的變化將是 v 7.0.0 整個 GPS 內容將從 hcxdumptool 中刪除，因為它將佔用許多 CPU 週期。 hcxpcapngtool 將讀取 gpx 檔案並將其與 hcxdumptool 記錄的 WiFi 流量關聯起來。 主要優點是您可以使用每個裝置（gpsd gpxlogger、智慧型手機、Gar​​min gps 裝置、GPS USB 裝置）來記錄 GPS 軌跡。

What is the output format of GPS records?

ZerBea, 2024-07-28:
GPX format:
https://en.wikipedia.org/wiki/GPS_Exchange_Format

GPX is supported by GPSBABEL
https://www.gpsbabel.org/
The entire conversion table is here:
https://www.gpsbabel.org/capabilities.html
Format specification is here:
https://www.gpsbabel.org/htmldoc-development/fmt_gpx.html

ZerBea, 2024-07-28:
I'm testing it with a Garmin eTrex 30, a navilock USB device, GPS Babel and  Viking in combination with OpenStreetMaps.

KaliAssistant, 2024-07-28:
Which gps device do you recommend?

ZerBea, 2024-07-28:
I use:
```
NAVILOCK NL-701US in combination with GPSD GPXLOGGER
Garmin eTrex 30
Garmin Drive 55
```

Main advantage of all Garmin devices is that you can use (free and daily updated) OpenStreetMap maps on them:
https://wiki.openstreetmap.org/wiki/OSM_Map_On_Garmin/Download

Attached an example track, recorded by an eTrex 30 and converted by gpsbabel (should be installable via KALI package manager) to GPX format,
This track can be directly shown by Viking (should be installable via KALI package manager)
[Offterschwang.gpx.zip](https://github.com/user-attachments/files/16402472/Offterschwang.gpx.zip)

ZerBea, 2024-07-28:
But every device which is in the conversion list of gpsbabel should be fine.

KaliAssistant, 2024-07-28:
Okay, I'll buy it and test it .

ZerBea, 2024-07-28:
I recommend to wait.
There is a big problem, I have to find a solution for:
The clock of the device running hcxdumptool must be in absolute sync with the gps device.
At the moment this is only the case if the gps device is connected directly to the system on which hcxdumptool is running.

KaliAssistant, 2024-07-28:
ok

ZerBea, 2024-07-28:
Unfortunately I have seen gps devices too on which the internal clock is not in sync with the gps time.
As a result we have three different timestamps:
time stamp of the device hcxdumptool is runnig
time stamp of the clock of the gps device
time stamp of the GPS signal

ZerBea, 2024-07-28:
But you can play around with Viking, GPS babel, GPSD GPXLOGGER in combination with your current gps device and OpenStreetMap to see how it works at all.

KaliAssistant, 2024-07-31:
@ZerBea Could I have your contact information, please?

ZerBea, 2024-07-31:
Sent via hashcat forum mail.

KaliAssistant, 2024-07-31:
ok

ZerBea, 2024-07-31:
Came back: deny recept!

I've enabled private mails. Please send your email address via hashcat forum to which I can reply.

ZerBea, 2024-07-31:
You should have received an email.

KaliAssistant, 2024-07-31:
yes