ZerBea/hcxdumptool #486: pcap/pcap.h: No such file or directory

ButtBundy, 2024-12-19:
build 6.3.5 wants to reference to pcap.h but it could it be pcapng.h?

```
fatal: detected dubious ownership in repository at '/opt/hcxdumptool'
To add an exception for this directory, call:

	git config --global --add safe.directory /opt/hcxdumptool
cc -O3 -Wall -Wextra -Wpedantic -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.5\" -DVERSION_YEAR=\"2024\" -DHCXSTATUSOUT -DHCXNMEAOUT -DHCXWANTLIBPCAP -lpcap
hcxdumptool.c:41:10: fatal error: pcap/pcap.h: No such file or directory
   41 | #include <pcap/pcap.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:75: hcxdumptool] Error 1
```

ZerBea, 2024-12-19:
Requirement is not satisfied. You need to install "libpcap-dev" as mentioned in requirements section of README.md
```
### Requirements
...
- libpcap and libpcap-dev (If internal BPF compiler has been enabled.)
...
```
Than everything compiles fine:
```
$ make
cc -O3 -Wall -Wextra -Wpedantic -std=gnu99   -o hcxdumptool hcxdumptool.c -DVERSION_TAG=\"6.3.5-3-g3725338\" -DVERSION_YEAR=\"2024\" -DHCXSTATUSOUT -DHCXNMEAOUT -DHCXWANTLIBPCAP -lpcap
```
Installs fine:
```
$ sudo make install
install -D -m 0755 hcxdumptool /usr/bin/hcxdumptool
```
and works as expected:
```
$ hcxdumptool -v
hcxdumptool 6.3.5-3-g3725338 (C) 2024 ZeroBeat
running on Linux kernel 6.6.65-1-lts
running GNU libc version 2.40
compiled by gcc 14.2.1
compiled with Linux API headers 6.10.0
compiled with GNU libc headers 2.40
enabled REALTIME DISPLAY
enabled GPS support
enabled BPF compiler
```

Closed, because it is not a hcxdumptool issue.

closed 2024-12-19 by ZerBea

ButtBundy, 2024-12-19:
works like a charm my appologies for not thinking about installing the dependencies